armbian/build #3421: Problem found in extra-buildpkgs/htop

Salamandar, 2022-01-21:
I just upgraded to bullseye and htop doesn't work anymore :
```
admin@nas ~ ❯❯❯ htop
malloc(): invalid size (unsorted)
[1]    76593 abort      htop
```

After a bit of search, I found out the 3.1.0 deb package for htop actually provides 2.2.0 o.O

The package in question is : https://imola.armbian.com/apt/pool/bullseye-utils/h/htop/htop_3.1.0-0~armbian20.08.2%2B1_arm64.deb

igorpecovnik, 2022-01-21:
Those are our hacky builds, but we are not going to make them anymore since most of our functionalities were build upstream. But Debian is a bit behind ... Check backports.

Salamandar, 2022-01-23:
Alright. For now I manually built htop, that makes the trick :)

hzyitc, 2023-01-02:
@igorpecovnik Consider to remove htop in our apt repo manually?

In our build script, it specify 2.2.0 but in apt repo is 3.1. This make confused.

Since we don't need to modify it, I will remove it in our build script.

https://github.com/armbian/build/blob/b8b160ad2f08caebce54b2e6b9b654309a6fba26/packages/extras-buildpkgs/90-htop.conf#L1-L9

igorpecovnik, 2023-01-02:
> Since we don't need to modify it, I will remove it in our build script.

Yes, we can remove it, also htop.