armbian/build #3443: sunxi: Add_frame_inversion_to_correct_audio_channels.patch not applied properly

matthijskooijman, 2022-01-26:
**Describe the bug**
The patch https://github.com/armbian/build/blob/master/patch/kernel/archive/sunxi-5.15/Add_frame_inversion_to_correct_audio_channels.patch is not an actual patch the kernel, but it is a patch that adds a `projects/Allwinner/devices/A64/patches/linux/04-Add-frame-inversion-to-correct-multi-channel-audio.patch ` file in the kernel source directory, which I believe is otherwise unused:

https://github.com/armbian/build/blob/10017519839916f35f851df20d6f0f17452ef4a6/patch/kernel/archive/sunxi-5.15/Add_frame_inversion_to_correct_audio_channels.patch#L27-L28

Tracing through history, it seems this patch originates from LibreELEC and was suggested by @jernejsk in https://github.com/armbian/build/pull/1383#issuecomment-573190891 and commited by @igorpecovnik in c962682. It seems the patch was taken verbatim from LibreElec, where I suspect this `projects/Allwinner` directory is used as a source of kernel patches maybe.

It was originally added for 5.10, but copied verbatim up to including 5.15 (and 5.16 seems to be in progress, no patches in there at all yet).

**Is this still needed?**

I'm not entirely sure if this patch is really needed. The goal is to add `simple-audio-card,frame-inversion;` to the `sound_hdmi` (H3/H4) or `hdmi_sound` (A64) section, but it seems that when it was first added in 5.10, that was already present in the "megous" kernel used, see [here for H3/H4](https://github.com/megous/linux/blob/5aa571dfc47b2fa92615ea5774475fa3650e8eb3/arch/arm/boot/dts/sunxi-h3-h5.dtsi#L965) and [here for A64](https://github.com/megous/linux/blob/5aa571dfc47b2fa92615ea5774475fa3650e8eb3/arch/arm64/boot/dts/allwinner/sun50i-a64.dtsi#L116). Later kernel versions seem to no longer use the "simple-audio-card" driver at all anymore, but a specific allwinner HDMI sound card driver (identified through the compatible value), see e.g. [here for H3/H4](https://github.com/megous/linux/blob/98029b03d3bba7d49ae89c9ab6d806422a0e5f80/arch/arm/boot/dts/sunxi-h3-h5.dtsi#L102-L114), A64 has the same thing.

So I think this patch can just be dropped from all kernel patch sets?

The-going, 2022-02-01:
@igorpecovnik I also think that this patch can be safely removed from the list of applied\used.

@matthijskooijman Thanks for the feedback.

igorpecovnik, 2022-02-01:
ok

closed 2022-02-10 by igorpecovnik