armbian/build #3472: drivers/video/fbdev/core/fbcon.c:2998:16: error: ‘struct fbcon_display’ has no member named ‘scrollmode’; did you mean ‘yscroll’?

Myron-S, 2022-02-10:
**Describe the bug**
`	== kernel ==

drivers/video/fbdev/core/fbcon.c: In function ‘fbcon_scrolldelta’:
drivers/video/fbdev/core/fbcon.c:2998:16: error: ‘struct fbcon_display’ has no member named ‘scrollmode’; did you mean ‘yscroll’?
  switch (disp->scrollmode) {
                ^~~~~~~~~~
                yscroll
make[4]: *** [scripts/Makefile.build:280: drivers/video/fbdev/core/fbcon.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [scripts/Makefile.build:497: drivers/video/fbdev/core] Error 2
make[2]: *** [scripts/Makefile.build:497: drivers/video/fbdev] Error 2
make[1]: *** [scripts/Makefile.build:497: drivers/video] Error 2
make: *** [Makefile:1824: drivers] Error 2
`

**To Reproduce**
Try and build U-boot and Kernel using ./compile.sh keeping default kernel options.

**Provide logs:**
[debug.zip](https://github.com/armbian/build/files/8044220/debug.zip)

igorpecovnik, 2022-02-10:
There has been a lot of changes today. Update build script and it should work.

CI automation didn't find any issues:
https://github.com/armbian/build/actions/runs/1826137975

Myron-S, 2022-02-11:
Ok. I shall clone he repo again and try again,  Will report back with my findings.

closed 2022-02-11 by Myron-S

MS1987, 2022-03-02:
I use the lastest commit of Master to compile for NanoPi NEO3, and have the same problem 
[compilation.log](https://github.com/armbian/build/files/8166266/compilation.log)
.

cc336625, 2022-03-02:
I have the same problem
```
	== kernel ==
drivers/video/fbdev/core/fbcon.c: In function ‘fbcon_scrolldelta’:
drivers/video/fbdev/core/fbcon.c:2969:14: error: ‘struct fbcon_display’ has no member named ‘scrollmode’
 2969 |  switch (disp->scrollmode) {
      |              ^~
make[4]: *** [scripts/Makefile.build:277：drivers/video/fbdev/core/fbcon.o] 错误 1
make[3]: *** [scripts/Makefile.build:540：drivers/video/fbdev/core] 错误 2
make[2]: *** [scripts/Makefile.build:540：drivers/video/fbdev] 错误 2
make[1]: *** [scripts/Makefile.build:540：drivers/video] 错误 2
make: *** [Makefile:1868：drivers] 错误 2
make: *** 正在等待未完成的任务....
```

igorpecovnik, 2022-03-02:
> I have the same problem

No problems in CI builds:
https://github.com/armbian/build/runs/5383450629

Build commands?

Custom config?
Custom sources?
In this case you need to take care of the source code troubles.