armbian/build #3506: Building img failed - unmount_on_exit() called! main.sh:1 -> image-helpers.sh:0 err

Myron-S, 2022-02-22:
Simply tried to build a SD card writable image using the build framework for BananaPiPro.  Included all desktop packages.

Logs ./output/debug generated by ./compile.sh are here:
https://drive.google.com/file/d/1-IO-o6mj70oGff9tWZ9qgGUSmvZmQtSZ/view?usp=sharing

Last few lines of output.log . . .
```
Displaying message: ADDING ADDITIONAL APT SOURCES
Displaying message: Adding APT Source ppa:oibaf/graphics-drivers
Displaying message: Return code : 0
Displaying message: Adding APT Source ppa:saiarcot895/chromium-dev
Displaying message: Return code : 0
Displaying message: Adding APT Source deb http://ryanfortner.github.io/box64-debs/debian ./
Displaying message: Return code : 0
Displaying message: Adding GPG Key /Armbian/build/config/desktop/focal/appgroups/emulators/sources/apt/box64.source.gpg
Displaying message: Adding APT Source deb http://paulcarroty.gitlab.io/vscodium-deb-rpm-repo/debs/ vscodium main
Displaying message: Return code : 0
Displaying message: Adding GPG Key /Armbian/build/config/desktop/focal/appgroups/programming/sources/apt/vscodium.source.gpg
Displaying message: Installing the desktop packages for Armbian info
Displaying message: ERROR in function create_rootfs_cache main.sh:578 -> main.sh:539 -> debootstrap.sh:58 -> debootstrap.sh:339 -> general.sh:0 err
Displaying message: Installation of Armbian desktop packages for current bananapipro focal 3dsupport browsers chat desktop_tools editors email emulators internet multimedia office programming remote_desktop xfce no failed  err
Displaying message: Process terminated  info
Displaying message: unmount_on_exit() called! main.sh:1 -> image-helpers.sh:0 err
Displaying message: Unmounting /Armbian/build/.tmp/rootfs-c1efdc09-151f-4107-a4a2-85d4a0a5385a/ info
Displaying message: ERROR in function unmount_on_exit main.sh:1 -> image-helpers.sh:94 -> general.sh:0 err
Displaying message: debootstrap-ng was interrupted  err
Displaying message: Process terminated  info

```

matthijskooijman, 2022-03-03:
The error in the issue subject line is, I think, only a message from the cleanup process after the error. I believe the actual error is:

> Displaying message: ERROR in function create_rootfs_cache main.sh:578 -> main.sh:539 -> debootstrap.sh:58 -> debootstrap.sh:339 -> general.sh:0 err

I'm not sure what the problem is exactly (haven't looked closely), but you might find a hint in debootstrap.sh line 339.

igorpecovnik, 2022-03-03:
One option is https://github.com/armbian/build/issues/3519 In general - check inside output/debug folder.

Myron-S, 2022-03-03:
I successfully built a server image, but building desktop image fails. @igorpecovnik I shall have a read of thread #3519 and see if that gives me any clues.

Myron-S, 2022-03-09:
> One option is #3519 In general - check inside output/debug folder.

I didn't use `--depth 1` when cloning the git. My build environment is Ubuntu impish operating under a Hyper-V VM.  Thought of using Docker but it confuses the hell out of me. A native build environment is simpler.

The out-of-the-box configuration of Ubuntu Impish VM is only used for the Armbian build framework.

The-going, 2022-03-10:
@Myron-S   There are two main reasons for this failure.
1. The free space has run out.
2. Incorrect spelling of the package name

In your log file: debootstrap-list.log
```
PACKAGE_LIST_DESKTOP =:
anacron                        apport-gtk                     blueman                        bluez                         
bluez-cups                     bluez-tools                    brltty                         brltty-x11                    
caffeine                       cifs-utils                     cups                           cups-bsd                      
cups-client                    cups-filters                   dbus-x11                       dictionaries-common           
dmz-cursor-theme               doc-base                       evince                         evince-common                 
fontconfig                     fontconfig-config              fonts-arphic-ukai              fonts-arphic-uming            
fonts-dejavu-core              fonts-freefont-ttf             fonts-guru                     fonts-guru-extra              
fonts-kacst                    fonts-kacst-one                fonts-khmeros-core             fonts-liberation              
fonts-nanum                    fonts-opensymbol               fonts-stix                     fonts-symbola                 
fonts-ubuntu-font-family-console foomatic-db-compressed-ppds    gdebi                          ghostscript-x                 
gir1.2-appindicator3-0.1       gnome-font-viewer              gnome-screenshot               gnome-user-docs-de            
gnome-user-docs-es             gnome-user-docs-it             gnome-user-docs-pt             gnome-user-docs-ru            
gnome-user-docs-sl             gnome-system-monitor           gstreamer1.0-packagekit        gstreamer1.0-plugins-base-apps
gstreamer1.0-pulseaudio        gtk2-engines                   gtk2-engines-murrine           gtk2-engines-pixbuf           
gvfs-backends                  hplip                          hunspell-en-us                 indicator-printers            
inputattach                    kerneloops                     keyutils                       language-pack-gnome-de        
language-pack-gnome-en         language-pack-gnome-es         language-pack-gnome-fr         language-pack-gnome-it        
language-pack-gnome-pt         language-pack-gnome-ru         language-pack-gnome-sl         laptop-detect                 
libappindicator3-1             libatk-adaptor                 libfont-afm-perl               libfontconfig1                
libfontembed1                  libfontenc1                    libgail-common                 libgl1-mesa-dri               
libgsettings-qt1               libgtk2.0-bin                  libnotify-bin                  libpam-gnome-keyring          
libproxy1-plugin-gsettings     libproxy1-plugin-networkmanager libu2f-udev                    libwmf0.2-7-gtk               
libxcursor1                    lightdm                        lightdm-settings               lxtask                        
mesa-utils                     mousepad                       mousetweaks                    network-manager-gnome         
network-manager-l2tp           network-manager-openconnect    network-manager-openvpn        network-manager-pptp          
network-manager-ssh            network-manager-vpnc           numix-gtk-theme                numix-icon-theme              
numix-icon-theme-circle        openprinting-ppds              orca                           p7zip-full                    
pamix                          pasystray                      pavucontrol                    pavucontrol-qt                
pavumeter                      pinta                          policykit-1                    policykit-desktop-privileges  
printer-driver-all             profile-sync-daemon            pulseaudio-module-bluetooth    redshift                      
slick-greeter                  smbclient                      software-properties-gtk        spice-vdagent                 
system-config-printer          system-config-printer-common   terminator                     thunar-volman                 
ttf-ubuntu-font-family         ubuntu-drivers-common          update-inetd                   update-manager                
update-manager-core            update-notifier                update-notifier-common         viewnior                      
wamerican                      wbrazilian                     wbritish                       wfrench                       
witalian                       wportuguese                    wspanish                       wswiss                        
x11-apps                       x11-xserver-utils              xarchiver                      xbacklight                    
xcursor-themes                 xdg-user-dirs                  xdg-user-dirs-gtk              xfce4                         
xfce4-notifyd                  xfce4-screenshooter            xfce4-terminal                 xfonts-100dpi                 
xfonts-75dpi                   xfonts-base                    xfonts-encodings               xfonts-scalable               
xfonts-utils                   xinit                          xorg-docs-core                 xscreensaver                  
xserver-xorg                   xserver-xorg-video-fbdev       xterm                          xtermcontrol                  
[... cut ...]
 				${OUTPUT_VERYSILENT:+' >/dev/null 2>/dev/null'} ';EVALPIPE=(${PIPESTATUS[@]})'
 
+			echo "stage after install PACKAGE_LIST_DESKTOP" >> $DEST/${LOG_SUBPATH}/err-debootstrap.log
+			(LC_ALL=C df -h) >> $DEST/${LOG_SUBPATH}/err-debootstrap.log
+
 			[[ ${EVALPIPE[0]} -ne 0 ]] && exit_with_error "Installation of Armbian desktop packages for ${BRANCH} ${BOARD} ${RELEASE} ${DESKTOP_APPGROUPS_SELECTED} ${DESKTOP_ENVIRONMENT} ${BUILD_MINIMAL} failed"
 		fi
 
```
try building the image again and publish it here `err-debootstrap.log`

closed 2022-11-25 by igorpecovnik