armbian/build #3519: image-helpers.sh segfault

sej7278, 2022-03-01:
Using these flags I built for orangepi r1+ lts.....

```
./compile.sh  BOARD=orangepi-r1plus-lts BRANCH=current RELEASE=bullseye \
    BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no \
    COMPRESS_OUTPUTIMAGE=sha,gpg,img
```

Whilst it resulted in an image at the end, there's a disturbing segfault that i can reproduce 100%

```
[ .... ] Installing [ armbian-bsp-cli-orangepi-r1plus-lts_22.05.0-trunk_arm64.deb ]
/home/simon/build/lib/image-helpers.sh: line 204: 571258 Segmentation fault      (core dumped) chroot "${SDCARD}" /bin/bash -c "DEBIAN_FRONTEND=noninteractive apt-get -yqq $apt_extra --no-install-recommends install $name" >> "${DEST}"/${LOG_SUBPATH}/install.log 2>&1
[ error ] ERROR in function install_deb_chroot [ main.sh:589 -> main.sh:550 -> debootstrap.sh:69 -> distributions.sh:334 -> image-helpers.sh:225 -> general.sh:0 ]
[ error ] Installation of /root/armbian-bsp-cli-orangepi-r1plus-lts_22.05.0-trunk_arm64.deb failed [ orangepi-r1plus-lts bullseye no rockchip64 ]
[ o.k. ] Process terminated 
[ .... ] Installing [ armbian-firmware_22.05.0-trunk_all.deb ]

```

i'm running this from within qemu-kvm, so wondered if it was something to do with running qemu aarch64 inside qemu x86_64?

igorpecovnik, 2022-03-01:
Try to use Docker.

sej7278, 2022-03-01:
i'll try, but not keen on installing docker on this box.

i managed to build the minimal image ok just now

igorpecovnik, 2022-03-01:
I think its one of the qemu bugs ... If you need bullseye/sid or Jammy, you need x64 Ubuntu Hirsute host. Or if you build natively on arm64.

sej7278, 2022-03-01:
exactly the same error on a debian bullseye vm, lets see where docker gets me

sej7278, 2022-03-02:
I might try building a different board as I noticed the r1+ only got a Sid build not bullseye on the website

sej7278, 2022-03-02:
i managed to build for bananpi on my hirsute vm (no docker):

```./compile.sh  BOARD=bananapi BRANCH=current RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img```

so maybe the problem is orangepi specific

sej7278, 2022-03-02:
ah now this is interesting, i managed to build a nanopi-r2s which is basically the same board:

```./compile.sh  BOARD=nanopi-r2s BRANCH=current RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img```

igorpecovnik, 2022-03-02:
Indeed its something wrong but it looks like with bullseye only. Nothing we can do about that. Use any other - buster works, focal works, ... others i didn't try.

igorpecovnik, 2022-03-02:
> i noticed the website has Sid, how do i select that, in compile.sh there doesn't seem to be an option?

Hidden out by purpose:
[Thi](https://docs.armbian.com/Developer-Guide_Build-Options/#hidden-options-for-advanced-users-default-values-are-marked-bold)

> i wonder is the build infra smart enough to notice the error

Current build infra has some issues regarding error detection, so this can't be ruled out ... but we fixed version, which is not yet on user level, will be merged probably in a couple of weeks.

sej7278, 2022-03-02:
ok got it now thanks, going to try this:

```./compile.sh  BOARD=orangepi-r1plus-lts BRANCH=edge RELEASE=sid BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img```

sej7278, 2022-03-04:
i'm really confused now, as this works:

```./compile.sh  BOARD=orangepi-r1plus-lts BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img```

so its the combination of a full bullseye and the 5.15 kernel that seems to be the problem?! that's gibberish. what's in the non-minimal build that conflicts with the 5.15 kernel?

sej7278, 2022-03-05:
ok i'm closing this as i can now build bullseye with the current kernel no problem, i can only assume it was the `--depth 1` checkout was missing some key files, or maybe something wasn't downloading properly as builds are repeatedly taking 24mins not an hour now.

this works: `./compile.sh  BOARD=orangepi-r1plus-lts BRANCH=current RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no COMPRESS_OUTPUTIMAGE=sha,gpg,img `

closed 2022-03-05 by sej7278