armbian/build #3522: Problem found in building 22.02 minimal orangepipc jammy/edge, /boot too small

matthijskooijman, 2022-03-03:
**Describe the bug**
When building an image for the orangepi pc, using docker on a 21.08 Impish host, initramfs generation fails because (I think) the boot partition is too small.
AFAICS the configuration below should, at first glance, match the official image roughly (based on the download page, I have not investigated which config is used for those exactly), which does have enough space in /boot, so either there is something in the config that is different (I'm using prebuilt as much as possible, maybe it's that?), or there is something in my environment (though the docker build should minimize that...).

I'll try with `BOOTSIZE=384` (up from 256) next to confirm that his is indeed the problematic partition, and then see if I can compare /boot contents between my image and the official build.

**To Reproduce**
```
$ cat userpatches/config-default.conf 
KERNEL_ONLY="no"
KERNEL_CONFIGURE="prebuilt"
BOARD="orangepipc"
# Jammy has gstreamer 1.2 with v4l2 codec support
RELEASE="jammy"
# The official jammy build also has the edge (5.16) kernel, so also use
# that here (maybe also works with current, not tested yet)
BRANCH="edge"
BUILD_MINIMAL="yes"
BUILD_DESKTOP="no"

CLEAN_LEVEL="make,debs,oldcache"        # comma-separated list of clean targets: "make" = make clean for selected kernel and u-boot,
                                        # "debs" = delete packages in "./output/debs" for current branch and family,
                                        # "alldebs" = delete all packages in "./output/debs", "images" = delete "./output/images",
                                        # "cache" = delete "./output/cache", "sources" = delete "./sources"
                                        # "oldcache" = remove old cached rootfs except for the newest 8 files

REPOSITORY_INSTALL=""                   # comma-separated list of core modules which will be installed from repository
                                        # (instead of to be built or installed from local cache).
                                        # "u-boot", "kernel", "bsp", "armbian-config", "armbian-firmware"
                                        # when empty and KERNEL_CONFIGURE="prebuilt", all are installed from repo

DEST_LANG="en_US.UTF-8"                 # sl_SI.UTF-8, en_US.UTF-8

# advanced
EXTERNAL_NEW="prebuilt"                 # compile and install or install prebuilt additional packages
INSTALL_HEADERS=""                      # install kernel headers package
LIB_TAG="master"                        # change to "branchname" to use any branch currently available.
USE_TORRENT="yes"                       # use torrent network for faster toolchain and cache download
DOWNLOAD_MIRROR=""                      # set to "china" to use mirrors.tuna.tsinghua.edu.cn
CARD_DEVICE=""                          # device name /dev/sdx of your SD card to burn directly to the card when done

$ ./compile.sh docker default
```

**Provide logs:**
Full logs here: https://gist.github.com/matthijskooijman/830616d3e8d198f37b9b23688c98b280

Tail of build output:
```
[ o.k. ] Preparing image file for rootfs [ orangepipc jammy ]
[ o.k. ] Current rootfs size [ 541 MiB ]
[ o.k. ] Creating blank image for rootfs [ 684 MiB ]
[ .... ] dd:  684MiB [75.2MiB/s] [======================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
[ .... ] Creating device node [ /dev/loop22p1 ]
[ .... ] Creating rootfs [ ext4 on /dev/loop22p1 ]
[ .... ] Copying files to [ / ]
[ .... ] Copying files to [ /boot ]

sent 18.08M bytes  received 4.74K bytes  12.05M bytes/sec
total size is 18.06M  speedup is 1.00
[ .... ] Updating initramfs... [ update-initramfs -uv -k 5.15.18-sunxi ]
[ error ] Updating initramfs FAILED, see: [ /root/armbian/output/debug/install.log ]
[ error ] unmount_on_exit() called! [ main.sh:1 -> image-helpers.sh:0 ]
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-3dccc9b7-7e8e-4c4e-856f-a7cc31d01aae/ ]
[ error ] ERROR in function unmount_on_exit [ main.sh:1 -> image-helpers.sh:94 -> general.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 

```

Tail of install.log:
```
Adding library /lib/arm-linux-gnueabihf/libnss_dns.so.2
dns: libresolv: /lib/arm-linux-gnueabihf/libresolv.so.2
Adding library /lib/arm-linux-gnueabihf/libresolv.so.2
Building cpio /boot/initrd.img-5.15.18-sunxi.new initramfs

gzip: stdout: No space left on device
E: mkinitramfs failure gzip 1
update-initramfs: failed for /boot/initrd.img-5.15.18-sunxi with 1.
```

igorpecovnik, 2022-03-03:
Not supported user land and not supported kernel - high possibility for troubles. We can't afford to support you here, sorry.

The-going, 2022-03-03:
> KERNEL_CONFIGURE="prebuilt"

Can take the values `yes` or `no`

> [ .... ] Creating device node [ /dev/loop22p1 ]
> [ .... ] Creating rootfs [ ext4 on /dev/loop22p1 ]

A very large device number `(loop22)`. It looks very much like there have already been failures in the build system and some devices have remained mounted. As a result, you simply don't have enough space at this stage.

This is my assumption based on my own experience. After two failed builds, I reboot the VM and start over.

matthijskooijman, 2022-03-03:
> Not supported unfinished user land and not supported kernel - high possibility for troubles. We can't afford to support you here, sorry.

Understood.

I can probably figure this out myself, without needing any holdholding (though pointers are always welcome, I am still getting familiar with the Armbian build). However, if it is ok, I would still share my findings here for the benefit of others and to track this publicly (and maybe suggest ways how things can be improved to improve the experience)?

> A very large device number (loop22). It looks very much like there have already been failures in the build system and some devices have remained mounted. As a result, you simply don't have enough space at this stage.

Nope, the large loop count is because of snapd, which seems to use about three loop devices for each snap I have installed :-S

> I'll try with BOOTSIZE=384 (up from 256) next to confirm that his is indeed the problematic partition, and then see if I can compare /boot contents between my image and the official build.

I tried this, which did not work, because the default configuration is to not have a separate /boot partition, so BOOTSIZE is forced to be 0. So the problem was not a full /boot, but a full rootfs, so it looks like the 25% extra space reserved here is not enough for the initrd: https://github.com/armbian/build/blob/ed589b248aef675cb5e845fcb55c12dfbcc5e925/lib/debootstrap.sh#L559

Adding `EXTRA_ROOTFS_MIB_SIZE="128"` to my config allowed the build to complete. 

I have yet to look more closely at the generated image, but maybe the difference between my build and the official build is that I'm using minimal (25% extra space) and the official build (Armbian_22.02.1_Orangepipc_jammy_edge_5.16.11.img.xz) is a desktop build (30% extra space)?

Looking at this, I wondered why the process is as it is: Naively I would think generating initrd first and only then deciding on how big the rootfs (or bootfs) should be would make more sense. However, I found #1584, and the fix #1723, which shows it used to be like I suggested, but this broke cryptsetup.

So it seems this "25%/30%" extra space is really space for initrd, plus any empty space to end up in the image. I wonder if:
 - It would not make sense to specify the space for initrd in absolute terms (based on the expected initrd size, which does not seem directly related to the rootfs size)
 - It would make sense to create an even bigger image first, and then shrinking it afterwards (though I'm not sure if all filesystems support this in a portable way...)

matthijskooijman, 2022-03-03:
W00ps, forgot to reply to:

> >     KERNEL_CONFIGURE="prebuilt"

> Can take the values yes or no

Apparently it als takes `prebuilt'`:

https://github.com/armbian/build/blob/ed589b248aef675cb5e845fcb55c12dfbcc5e925/lib/main.sh#L172-L175

igorpecovnik, 2022-03-03:
```
gzip: stdout: No space left on device
E: mkinitramfs failure gzip 1

```
> pointers are always welcome

Problem is ramdisk size, not boot or rootfilesystem size.

matthijskooijman, 2022-03-07:
TL;DR: There seems to be some (to me) weirdness with the size of the image, I'm adding debug output to figure out what happens exactly, will report more later.

> Check @150balbes commits. I think he hit and fix this problem somewhere. (he is offline on vacations and can't help)

Hm, I had a look at their armbian-tv branch, but it's dozens of commits just saying "fix", so I could not find anything useful in there.

> Problem is ramdisk size, not boot or rootfilesystem size.

But the ramdisk is generated as a file in /boot (which is part of the rootfs by default), so it looks like free space in the rootfs image is the problem.

Looking at the generated images, I see that the official (Jammy desktop) and my (Jammy minimal) builds both have near identical /boot sizes (42M vs 41M). The official build has 1041MB used, 103MB free, my build has 575MB used, 185MB free.

My build was done with `EXTRA_ROOTFS_MIB_SIZE="128"`, so even without that it would still have (185-128)=57MB free, so it *is* large enough to contain the initrd. This makes me suspect the problem is that the initrd *generation* process needs some more empty space (it probably collects files in /tmp or so before creating a compressed archive).

Also, the amount of free space is significantly less than the 25/30% that is added during the build process (e.g. (1041-42)=999MB before initrd generation, 30% of that would be 300MB, not 103+42MB), so that suggests that in addition to create the initrd, there are more changes to the rootfs contents that happen *after* the image size is decided. However, looking at deboostrap.sh, I cannot really find these (except for some fstab and bootloader config changes in `prepare_partitions`, which should not be so big). From `output/debug/debootstrap.log` I find that this extra size is already used up [directly after running update-initramfs](https://github.com/armbian/build/blob/b4dfaaf286727786641c89abb65649e000a2eb85/lib/debootstrap.sh#L863), so I did not look beyond that.

What could be the case is that this extra disk usage is due to differences in filesystem block/cluster/whatever sizes or filesystem type, where the same files on my host filesystem take up less space than on the final image filesystem. So the files actually take up more space than expected in the image, reducing the free space accordingly. I'll add some debug output to confirm this is indeed the case.

matthijskooijman, 2022-03-07:
TL;DR: space difference is explained by fileystem overhead, update-initramfs tempspace used seems like it should have fit even in a minimal image (but by a large margin).

It turns out the "missing" space I was looking for is just FS overhead. On ext4, by default of space is 5% is reserved, and about 1.6% of space is used for inode tables [[source](https://serverfault.com/questions/282317/disk-space-overhead-in-ext4)]. I tried building a desktop image now, which reports 2295M of files, adding 30% gives an image size of 2992M. Directly after mkfs, `df` reports total space of 2796, which is pretty much exactly that 6.6%. After adding in rootfs files, only 2341MB is used (so even a little less than the 2295 reported, maybe because some files like `/tmp', '/run', etc. are excluded). Then generating initramfs adds 24M, which is exactly the size of the generated initrd (two files in /boot: regular and uboot version).

IOW: the disk space I was missing, is completely explained by filesystem overhead that I had forgotten to take into account. Since this filesystem overhead scales linearly with size, compensating for this using the 25/30% of extra size seems a reasonable approach.

As for the initrd temp space, on the desktop build, it seems to need about 65M of space to generate the image, of which 24M stays in the image. This does not explain the failure with the minimal image yet, though, since I deduced that without `EXTRA_ROOTFS_MIB_SIZE` that would have 57MB free (probably even a little more accounting for FS overhead) on top of the 24M used for the initrd, so together that's more than 65M. But maybe the minimal initrd is bigger somehow, or there is some other difference?

In any case, it seems that on a minimal build, the extra free size margin (the 25/30%) is simply not enough to do initrd generation (since the base image size is smaller *and* the percentage is smaller, that results in a lot less free space). One simple fix for this could be to make `/var/tmp` (which is where update-initramfs does its work) inside the mounted image a tmpfs, so that there is no need to have free space for the initrd generation, just the result? Or, the margin could just be increased (there is already a different margin for desktop and minimal images). Of course the free space is not only needed for the initrd generation, but probably also some free space is needed for the first boot (at least up to the moment where the SD root partition is resized).

matthijskooijman, 2022-03-07:
TL;DR: Available space varies a bit with factors, but it seems that a minimal build just has a bit too few free space available for the initramfs build. I'll try a orangepizero focal current build to compare, since that is an officially built minimal image.

Trying a minimal build again, it seems that the FS overhead on the minimal build is a little higher (584M image results in 632M space in `du`, so just under 8% overhead), suggesting that at least a part of it is fixed-size rather than scaling with the filesystem size. Also, after copying in the rootfs files, the disk usage is slightly bigger than expected (543M of files take up 553M of space, unlike the desktop build while this was slightly smaller instead). Also, it seems that the "5% reserved for root" is not actually subtracted from the total space in `du`, but from the available space (so Used + Available < Total, except that the difference is not quite exactly 5% but 4.something).

Together, this means that the available space (according to `du`) before generating the initrd is only 49M. Even though some of the "reserved for root" space seems to be usable for initrd (but maybe not all), this is just not enough to allow generating an initrd, so the build fails.

So in summary, it seems that for a minimal image, there is just too little margin in the free space department for the initrd generation. I suspect that on focal and/or current images, either the free space is just a little bigger, or the initrd just a little smaller, allowing the generation to succeed. To confirm, I'm building orangepizero focal current minimal now, which is listed in targets.conf (so I'm assuming that one should work).

The-going, 2022-03-08:
A very interesting investigation.
I'm only collecting a minimal image. And I have never encountered such a problem.
Please replace this line:
https://github.com/armbian/build/blob/d511679092f07fb9e09df180698ee7d6288cb36c/lib/debootstrap.sh#L863
```
        # DEBUG: print free space
        local freespace=$(LC_ALL=C df -h)
-       echo $freespace >> $DEST/${LOG_SUBPATH}/debootstrap.log
+       echo "======= stage create_image =======" >> $DEST/${LOG_SUBPATH}/debootstrap.log
+       (LC_ALL=C df -h) >> $DEST/${LOG_SUBPATH}/debootstrap.log
        display_alert "Free SD cache" "$(echo -e "$freespace" | grep $SDCARD | awk '{print $5}')" "info"
        display_alert "Mount point" "$(echo -e "$freespace" | grep $MOUNT | head -1 | awk '{print $5}')" "info"
```
My log without error looks like this:
```
======= stage create_image =======
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  8.9G  79% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-e2560863-c41a-450d-8dfe-90e15010b698
/dev/loop7p1    960M  837M   88M  91% /home/leo/build/.tmp/mount-e2560863-c41a-450d-8dfe-90e15010b698
```

matthijskooijman, 2022-03-08:
@The-going What is your build config (in particular BOARD, RELEASE and BRANCH)? It seems your image is a lot bigger (960M) that mine (632M), so also has 50% more free space probably, so no issue.

I cannot quickly run with your suggested patch now (in the train), but from my  [https://github.com/armbian/build/issues/3522#issuecomment-1060859374](previous notes) I can derive that the output would be something like: 632M size, 553M used, and IIRC something like 47M of free space, but this is before update-initramfs (which would fail subsequently in my build, so your patch output would never be reached).

matthijskooijman, 2022-03-08:
I did another minimal build for the zeropi, focal, current (which is one of the official builds), which did complete succesfully. Looking at the the disk space, I see:

```
rootfs_size: 587 MiB
imagesize: 591 MiB
sdsize: 740 MiB
After mount
Filesystem                      1M-blocks   Used Available Use% Mounted on
/dev/loop24p1                         688      1       658   1% /root/armbian/.tmp/mount-8b53cdca-506a-4205-9c24-7245eb5b9534

After rsync /
Filesystem                      1M-blocks   Used Available Use% Mounted on
/dev/loop24p1                         688    580        80  88% /root/armbian/.tmp/mount-8b53cdca-506a-4205-9c24-7245eb5b9534

After rsync /boot               
Filesystem                      1M-blocks   Used Available Use% Mounted on
/dev/loop24p1                         688    597        62  91% /root/armbian/.tmp/mount-8b53cdca-506a-4205-9c24-7245eb5b9534

After update_initramfs
Filesystem                      1M-blocks   Used Available Use% Mounted on
/dev/loop24p1                         688    617        42  94% /root/armbian/.tmp/mount-8b53cdca-506a-4205-9c24-7245eb5b9534
```

So that image is indeed just a little bigger than the orangepipc, jammy, edge build I did before (rootfs_size 543MiB, sdsize 684MiB), so this focal build I just did probably has pretty much exactly enough free space to complete the build.

The-going, 2022-03-09:
Thanks for the feedback. I think this information is enough.
I have added the following debugging code.
```
diff --git a/lib/debootstrap.sh b/lib/debootstrap.sh
index bdf011ca3..32a725c45 100644
--- a/lib/debootstrap.sh
+++ b/lib/debootstrap.sh
@@ -17,6 +17,12 @@
 # update_initramfs
 # create_image
 
+# DEBUG: print_free_space
+print_free_space () {
+       local localcall=$@
+       echo "== stage $localcall ==" >> $DEST/${LOG_SUBPATH}/debootstrap.log
+       (LC_ALL=C df -h) >> $DEST/${LOG_SUBPATH}/debootstrap.log
+}
 
 
 
@@ -847,6 +865,8 @@ update_initramfs()
        }
        display_alert "Updated initramfs." "for details see: $DEST/${LOG_SUBPATH}/install.log" "info"
 
+       chroot $chroot_target /bin/bash -c "ls -alh /boot/{initrd.img-*,uInitrd-*}" >> $DEST/${LOG_SUBPATH}/debootstrap.log
+
        display_alert "Re-enabling" "initramfs-tools hook for kernel"
        chroot $chroot_target /bin/bash -c "chmod -v +x /etc/kernel/postinst.d/initramfs-tools" >> "${DEST}"/${LOG_SUBPATH}/install.log 2>&1
  
 ```
It is called around the string https://github.com/armbian/build/blob/11f1223ce27e4447b59733e165b79a8140da0773/lib/debootstrap.sh#L858
in this way:
```
	print_free_space "before update_initramfs: $(basename "${BASH_SOURCE[1]}"): ${FUNCNAME[1]}: ${BASH_LINENO[0]}"

	# stage: create final initramfs
	[[ -n $KERNELSOURCE ]] && {
		update_initramfs $MOUNT
	}

	# DEBUG: print free space
	local freespace=$(LC_ALL=C df -h)

	print_free_space "after update_initramfs: $(basename "${BASH_SOURCE[1]}"): ${FUNCNAME[1]}: ${BASH_LINENO[0]}"

	display_alert "Free SD cache" "$(echo -e "$freespace" | grep $SDCARD | awk '{print $5}')" "info"
	display_alert "Mount point" "$(echo -e "$freespace" | grep $MOUNT | head -1 | awk '{print $5}')" "info"
```
My configuration file contains the following variables:
```
KERNEL_ONLY="no" 
KERNEL_CONFIGURE="no" 
KERNEL_KEEP_CONFIG="yes"
BUILD_KSRC=no
BUILD_MINIMAL=yes
SELECTED_CONFIGURATION="cli_minimal"

EXTRAWIFI="no"
EXPERT=yes

BOARD=orangepipc2
BRANCH=edge
RELEASE=hirsute
```
The debootstrap.log file now contains:
```
== stage debootstrap.sh: debootstrap_ng: 121 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  9.7G  77% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-ae4fa57d-eeee-434c-b09c-78a753842eca
/dev/loop7p1    960M  1.3M  923M   1% /home/leo/build/.tmp/mount-ae4fa57d-eeee-434c-b09c-78a753842eca
== stage before update_initramfs: debootstrap.sh: debootstrap_ng: 122 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  9.0G  79% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-ae4fa57d-eeee-434c-b09c-78a753842eca
/dev/loop7p1    960M  815M  109M  89% /home/leo/build/.tmp/mount-ae4fa57d-eeee-434c-b09c-78a753842eca
-rw-r--r-- 1 root root 11M Mar  9 09:00 /boot/initrd.img-5.16.11.2-sunxi64
-rw-r--r-- 1 root root 11M Mar  9 09:00 /boot/uInitrd-5.16.11.2-sunxi64
== stage after update_initramfs: debootstrap.sh: debootstrap_ng: 122 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  8.9G  79% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-ae4fa57d-eeee-434c-b09c-78a753842eca
/dev/loop7p1    960M  837M   88M  91% /home/leo/build/.tmp/mount-ae4fa57d-eeee-434c-b09c-78a753842eca
```

The-going, 2022-03-09:
Next step:
```diff
diff --git a/lib/debootstrap.sh b/lib/debootstrap.sh
index 32a725c45..a9fbaa956 100644
--- a/lib/debootstrap.sh
+++ b/lib/debootstrap.sh
@@ -624,7 +624,7 @@ PREPARE_IMAGE_SIZE
                if [[ $BUILD_DESKTOP == yes ]]; then
                        local sdsize=$(bc -l <<< "scale=0; ((($imagesize * 1.30) / 1 + 0) / 4 + 1) * 4")
                else
-                       local sdsize=$(bc -l <<< "scale=0; ((($imagesize * 1.25) / 1 + 0) / 4 + 1) * 4")
+                       local sdsize=$(bc -l <<< "scale=0; ((($imagesize * 1.15) / 1 + 0) / 4 + 1) * 4")
                fi
        fi
 
```

```bash
cat output/debug/debootstrap.log                                                          
== stage debootstrap.sh: debootstrap_ng: 121 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  9.7G  77% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-fbb6b9c2-e7a1-48d8-b4cd-4c3effcb52a7
/dev/loop7p1    880M  1.2M  845M   1% /home/leo/build/.tmp/mount-fbb6b9c2-e7a1-48d8-b4cd-4c3effcb52a7
== stage before update_initramfs: debootstrap.sh: debootstrap_ng: 122 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  9.0G  79% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-fbb6b9c2-e7a1-48d8-b4cd-4c3effcb52a7
/dev/loop7p1    880M  815M   31M  97% /home/leo/build/.tmp/mount-fbb6b9c2-e7a1-48d8-b4cd-4c3effcb52a7
-rw-r--r-- 1 root root 11M Mar  9 11:33 /boot/initrd.img-5.16.11.2-sunxi64
-rw-r--r-- 1 root root 11M Mar  9 11:33 /boot/uInitrd-5.16.11.2-sunxi64
== stage after update_initramfs: debootstrap.sh: debootstrap_ng: 122 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  8.9G  79% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-fbb6b9c2-e7a1-48d8-b4cd-4c3effcb52a7
/dev/loop7p1    880M  837M  9.2M  99% /home/leo/build/.tmp/mount-fbb6b9c2-e7a1-48d8-b4cd-4c3effcb52a7
```

The-going, 2022-03-09:
I have adjusted the debugging code to print the calculated and actual partition sizes.
The results are printed:
```
== imagesize=808 ==
== *1.15 sdsize=932 ==

== stage after prepare partitions:  debootstrap.sh: prepare_partitions: 833 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   33G  8.9G  79% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-a25db9ae-df93-44cf-9127-83c7c36bab17
/dev/loop7p1    880M  1.2M  845M   1% /home/leo/build/.tmp/mount-a25db9ae-df93-44cf-9127-83c7c36bab17

== stage before update initramfs:  debootstrap.sh: create_image: 943 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   34G  8.1G  81% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-a25db9ae-df93-44cf-9127-83c7c36bab17
/dev/loop7p1    880M  815M   31M  97% /home/leo/build/.tmp/mount-a25db9ae-df93-44cf-9127-83c7c36bab17

-rw-r--r-- 1 root root 11M Mar  9 14:05 /boot/initrd.img-5.16.11.2-sunxi64
-rw-r--r-- 1 root root 11M Mar  9 14:05 /boot/uInitrd-5.16.11.2-sunxi64

== stage after update initramfs:  debootstrap.sh: create_image: 953 ==
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           1.2G  1.2M  1.2G   1% /run
/dev/vda2        45G   34G  8.0G  81% /
tmpfs           5.9G     0  5.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           4.0M     0  4.0M   0% /sys/fs/cgroup
tmpfs           1.2G  4.0K  1.2G   1% /run/user/1000
tmpfs            14G  804M   14G   6% /home/leo/build/.tmp/rootfs-a25db9ae-df93-44cf-9127-83c7c36bab17
/dev/loop7p1    880M  837M  9.2M  99% /home/leo/build/.tmp/mount-a25db9ae-df93-44cf-9127-83c7c36bab17
```

The `update-initramfs' command creates a temporary folder in '/var/tmp/mkinitramfs_XXXXXX`
and copies the necessary files there. If there is not enough space, then we will receive similar messages:
```
...
Adding module /lib/modules/5.16.11.2-sunxi64/kernel/net/ceph/libceph.ko
cp: error writing '/var/tmp/mkinitramfs_m5j3IP//usr/lib/modules/5.16.11.2-sunxi64/kernel/net/ceph/libceph.ko': No space left on device
...
```

@igorpecovnik Do you think it makes sense to add this debugging code to the build system? Or is it enough to adjust the logic to exclude such errors?

igorpecovnik, 2022-03-09:
> Do you think it makes sense to add this debugging code to the build system? Or is it enough to adjust the logic to exclude such errors?

Perhaps we set a threshold to throw warning to console if loop free space falls below 5% and verbose error if cp error is detected?

The-going, 2022-03-09:
Good. I'll think about it.

matthijskooijman, 2022-03-09:
Thanks for the extra details. I'm a bit surprised that your build with only 15% extra space (https://github.com/armbian/build/issues/3522#issuecomment-1062841087) actually builds correctly, seeing there is only 9.2M of free space after the initrd is built (I would expect that this not enough tempspace to build the initrd). However, your image is a lot bigger than mine, so I guess that also means that the 5% "reserved for root" space (which is *on top of* the free space reported by df) is bigger, and that space can be used by update-initramfs.

In any case, my conclusion of all this is that the official minimal builds have *just* enough free space to allow building. Builds that produce smaller images or bigger initrds are then likely to overflow this free space and break. Even though these builds are currently not supported (i.e. jammy builds), I would expect these builds to become supported later (i.e. when jammy is released), so it would be good to make changes needed to make them work. And more generally, maybe make changes to make this work a bit more reliably.

From my investigation, there are four things for which the "25/30%" of "extra" image size are needed:
 1. Filesystem overhead (i.e. the available disk space is less than the size of the .raw file created). This overhead seems to be partially fixed, and partially scales with the image size.
 2. Space to store the initrd. This is a fixed amount (in the sense that it does not scale with image size), though it probably does depend on the specifics of the image (distribution, kernel, installed packages). I've seen 20-24M, so very big.
 3. Temporary space to build the initrd. In my testing, this was about 50M, on top of the previous two points.
 4. A bit of empty space after flashing the image to use during boot (not sure how much is needed, I think only up to the point where the partition is resized).

To make the build process more stable, I can imagine a few fixes:
 1. Just increase the 25% overhead (e.g. to 30% for desktop and non-desktop builds, or 30% for minimal and desktop builds and keep 25% for cli builds). This is by far the least amount of work, but is also the most imprecise, prone to these problems resurfacing later, and results in bigger images even when not strictly needed.
 2. Make `/var/tmp` a `tmpfs` while running `update-initramfs`. I suspect this is fairly easy and removes point 3 above (and the resulting dependency on the image contents).
 3. Add some fixed amount of space to the image size, in addition to the percentage added now. This ensures that smaller images do not have (as much) less free space as bigger images (and as needed). Big question then is of course how to determine how much you need. It might be helpful to simply assign numbers to each of the above points, so actually figure out how much absolute and relative filesystem overhead there is (probably filesystem dependent, though, so that adds complexity), how much (building the) initrd needs typically and how much free space you want/need. Naming these explicitly in the build script might be helpful to help address this issue in the future. Also note that point 3 and 4 above can overlap (i.e. any temp space you allocate will end up as free space in the image), so if you would e.g. want 5% free space and need 60M temp space, you could just add 5% or 60M, whichever is bigger.
 4. Create an image that is significantly bigger and then shrink it. But this does add overhead, and I'm not sure if all filesystems have good tools to shrink them... Also, too much complexity I think.

> Perhaps we set a threshold to throw warning to console if loop free space falls below 5% and verbose error if cp error is detected?

Adding some warnings (in addition to making some actual changes) is probably a good idea too. If applying point 3 above, a warning could be added to check whether the predicted filesystem overhead is (somewhat) correct maybe.

igorpecovnik, 2022-03-09:
Historically we didn't need to pay attention much on initrd, but it seems this starts to represent a serious impact to the image size. 

> Just increase 

This is quick and easy but yes, its not a perfect fix. Its adjusting the existing logic. If we are talking about 10-15% increase in (uncompressed) image size, this is IMO not an issue. 
+ warning 

More is up to you guys.

The-going, 2022-03-09:
> 2\. Make `/var/tmp` a `tmpfs` while running `update-initramfs`. I suspect this is fairly easy and removes point 3 above (and the resulting dependency on the image contents).

At the current time, this should be enough for trouble-free operation. In this case, we will not be interested in the size of the additional space for creating the initramfs archive, but only the estimated size for writing the finished initrd.img, uInitrd.

rpardini, 2022-03-17:
I've hit similar problem with some very small cloud images. I added `EXTRA_ROOTFS_MIB_SIZE` back then as a dirty hack/workaround... so thanks @matthijskooijman for deep analysis here

I really like the idea of using tmpfs for `/var/tmp` during initrd creation (also, set TMPDIR in chroot maybe).

I still feel bad about having to run that under qemu since it's very, very slow, both module heuristics and then compressing the cpio could be done natively eventually.

The-going, 2022-03-17:
> I added `EXTRA_ROOTFS_MIB_SIZE` back then as a dirty hack/workaround..

You did everything right. We can't calculate all the possible options. This should be done by the user for his specific case.

> (also, set TMPDIR in chroot maybe)

Please show a line of code to correctly understand what it is about.

matthijskooijman, 2022-03-17:
> I still feel bad about having to run that under qemu since it's very, very slow, both module heuristics and then compressing the cpio could be done natively eventually.

Yeah, I've noticed that update-initramfs is indeed really slow... I wonder if you could maybe make e.g. `/usr/bin/cpio` a (readonly) bindmount of the host system's `cpio` binary (probably also requires bindmounting `/lib/x86_64-linux-gnu` and `/usr/lib/x86_64-linux-gnu` to work) to do this transparently and improve build performance? A bit off-topic for this issue, though...

rpardini, 2022-03-17:
> Yeah, I've noticed that update-initramfs is indeed really slow... I wonder if you could maybe make e.g. `/usr/bin/cpio` a (readonly) bindmount of the host system's `cpio` binary (probably also [...]

GMTA. We'd have to inject gzip and probably bash as well, it could work, but it could be that initramfs gets confused and adds the host's binaries/libs to the initrd. Using pigz could also speedup a lot, even if emulated. 

Another approach would be to introduce some initrd caching (hash kernel version + family + board + patches hash; look for matching initrd cache; if not found, produce one and store in cache).

The-going, 2022-03-17:
> Another approach would be to introduce some initrd caching

This is a bad idea.

> I really like the idea of using tmpfs for `/var/tmp` during initrd creation (also, set TMPDIR in chroot maybe).

Why did I ask about the lines of code?
Here we need to mount the tmpfs folder of the host to `"${target}"/var/tmp`:
https://github.com/armbian/build/blob/ed85576979ddc58ecdce88713380051fb2a70fa4/lib/image-helpers.sh#L32-L41

The update-initramfs script creates a temporary folder in the path /var/tmp. Copies the necessary files to it, and then creates an archive from it.

matthijskooijman, 2022-03-20:
> > Another approach would be to introduce some initrd caching

> This is a bad idea.

I'm also afraid that it would be quite tricky to reliably determine which cached version to use and/or invalidate the cache when needed. Using the wrong version probably has significant impact (such as unbootable systems maybe)?

>  Here we need to mount the tmpfs folder of the host to "${target}"/var/tmp:

I'm not sure this is really the best place. AFAIU this code is also used during initial rootfs generation, when the rootfs itself is already a tmpfs. I do not think there would be much point in making `/var/tmp` a tmpfs when `/` is already tmpfs, it is only needed in the last stage when rootfs is loopback-mounted image file.

Having said that, this probably is the easiest way to do this, and the double tmpfs is not really needed, but also wouldn't hurt much I guess.

igorpecovnik, 2022-09-14:
Following MRs, that has recently been merged, should improve this situation:
https://github.com/armbian/build/pull/4185
https://github.com/armbian/build/pull/4181

Closing as I think problem from the topic should be resolved. If not, reopen.

closed 2022-09-14 by igorpecovnik

matthijskooijman, 2022-09-14:
Looks good, thanks for following up!