armbian/build #3531: Possible cleanups / improved documentation about rootfs cache

matthijskooijman, 2022-03-11:
When reading the build scripts to figure out how rootfs caching works, I came across some things that puzzled me and might be problems, or at least complications that can be simplified. Let me just list them.

1. Redundent cycles setting
    Before checking the cache, this code decides whether to loop once or twice: https://github.com/armbian/build/blob/5fe0f36aa832ac64d1db05adc4eb09a4df4b5509/lib/debootstrap.sh#L122-L126
    However, inside the loop, there is this line: https://github.com/armbian/build/blob/5fe0f36aa832ac64d1db05adc4eb09a4df4b5509/lib/debootstrap.sh#L141 so the loop will run only once regardless of the value of `$cycles`. This line was added in 84d7079c5dfd190dfeb8b0bb48ac064d0a5a7dad, so more recently than the cycles-setting code in dd50c432f812e24deaa2ae67b931fd7f51a865e5, but now it seems dd50c432f812e24deaa2ae67b931fd7f51a865e5 can just be reverted.
2. Incorrect "previous cache hash" calculation
    This line calculates the rootfs hash: https://github.com/armbian/build/blob/5fe0f36aa832ac64d1db05adc4eb09a4df4b5509/lib/debootstrap.sh#L132
    It was added in #2257. Before that PR, only the (manually bumped) `$ROOTFSCACHE_VERSION` was used in the hash, trying the current value and the previous value (`$ROOTFSCACHE_VERSION - $n` where n is 0 or 1), to ensure that after a version bump, clients would just use the previous cache version, rather than all building from scratch in the time that new downloads are being prepared. With this PR, the current month is also included in the hash, which means that even without manual version bumps, new caches are created at least once per month.
    I can see a few potential issues (though they might also be lack of understanding at my end):
     1. There is a `$D` in that line that does not seem to be set anywhere. If it is empty, this will just use the current date (`+ $n month`), but if it is set to some non-date string, `date` will error out. Maybe this is for artificially setting a different build date on the build servers somehow? If so, it is a fairly short variable that is prone to being set accidentally and not quite self-explanatory.
     2. This tries the current month and the *next* month, which seems weird. Shouldn't this be the *previous* month (so `- $n month`)?
     3. If b. is fixed, this falls back to the previous cache version on month changes, preventing the AR-166 issue. However, on manual ROOTFSCACHE_VERSION bumps, I think the issue still (or rather, again) exists, since then no downloadable files for the new version will exist (with neither the current or previous month). To really properly fix this, I guess there should be two nested for loops, one to also try the previous month, and one to also try the previous version (which will also try the previous version and previous month, which would be needed when the manual version bump happens near a month change).
 3. Incorrect cache hash for locally generated rootfs
    [The loop that tries the current and previous cache hashes](https://github.com/armbian/build/blob/5fe0f36aa832ac64d1db05adc4eb09a4df4b5509/lib/debootstrap.sh#L129), is also responsible for setting the `$cache_fname` used for a locally-generated cache file. If, for some reason, no cache files can be downloaded, the build falls back to local generation, but uses the filename set by the *second* iteration of the loop, which is the hash of the *previous* entry, not the current one (assuming 2.ii above is fixed, currently it is actually going to be the hash of a *future* entry from next month, unless this is actually by design like this?).
 4. Unclear what ROOT_FS_CREATE_ONLY does. The [docs](https://docs.armbian.com/Developer-Guide_Build-Options/) say "ROOT_FS_CREATE_ONLY ( FORCE ): set to skip rootfs download and create locally", suggesting that it only has one possible value ("FORCE") and that should only skip remote downloads, but otherwise build normally. Looking at the code, though:
    1. Setting any non-empty value causes the build to check for a valid cache entry, or download or build one if not present (but not extract it), and then exit the build. It seems that this case is intended for internal use on the download servers, to just generate a rootfs and not do a full build.
    2. Setting `ROOT_FS_CREATE_ONLY=force` (note the lowercase, unlike the uppercase from docs) does the same, but additionally tries only a single cache filename (bypassing problem 3 above), but also bypassing the partial download and archive integrity checks here: https://github.com/armbian/build/blob/5fe0f36aa832ac64d1db05adc4eb09a4df4b5509/lib/debootstrap.sh#L143-L161
    3. I guess you could run with `ROOT_FS_CREATE_ONLY="force"` once to skip the download process and generate a local rootfs (which only works if you did not previously download one), and then run a second time without it to use the previously downloaded one. There does not seem to be an option to just do a normal build, but just never download a rootfs (generating a new one every month instead).
 5. In general, it seems that the control flow of this code has evolved over time, and is not as simple as it could be anymore. Especially the handling of `ROOT_FS_CREATE_ONLY` is intertwined in the rest of the logic, and contains some duplicate code that I think could be simplified.

Taking a step back and looking at what the code should do, I think there's roughly two usecases:
 1. Doing a normal build. This should check for, in order: an existing up-to-date local cache, a downloadable up-to-date cache, a 1-version-older local cache, and a 1-version-older remote cache (where 1-version older can refer to the version, month, or both). The first of these caches is unpacked and used, if it is corrupt, the build is aborted (or the next version could be tried, but that's not done now for simplicity). If none of these caches is available, a new rootfs is created (saved to local cache using the current version hash). In any case, the build continues.
 2. Doing a download server build. This should check for an existing up-to-date local cache, and if it does not exist or is corrupt, a new rootfs should be created and saved to the local cache. In any case, the build aborts.

I can imagine a "never download" option for usecase 1. to skip the downloads and just check the up-to-date local cache and fall back to building a new rootfs if it is not available (Without also checking for a 10version-older local cache, if you're not downloading, there is not really a point in trying an older version too).

Essentially, usecase 2. is then just usecase 1., but with the "never download" option, verifying cache files rather than unpacking them, deleting the rootfs after creating it, and not continuing with the build.

If these usecases are correct, maybe I can make some suggestion for an improved code structure to handle just these?

The-going, 2022-03-11:
@matthijskooijman I think the easiest way to fix this is to make an algorithm for this function and write the code again. This will be a faster and better option.

matthijskooijman, 2022-03-11:
I noticed one more issue with this line: https://github.com/armbian/build/blob/5fe0f36aa832ac64d1db05adc4eb09a4df4b5509/lib/debootstrap.sh#L131

AFAICS that is the only time the `$n` variable is used, and it looks like the idea is that `FORCED_MONTH_OFFSET` (which is used in the cache hash) becomes 0 and then 1, to try two different cache versions, unless `FORCED_MONTH_OFFSET` is already set externally, then just use that one version. However, on the first run of the loop, this sets `FORCED_MONTH_OFFSET`, so on the second run it is already set and thus not changed. Effectively I think this means the loop just does exactly the same thing twice...

> @matthijskooijman I think the easiest way to fix this is to make an algorithm for this function and write the code again. This will be a faster and better option.

Yeah, that was where I was going with the last part of my post, trying to establish the usecases that should be implemented by this code and then designing it again...

The-going, 2022-03-12:
> Yeah, that was where I was going with the last part of my post, trying to establish the usecases that should be implemented by this code and then designing it again...

Let's then set sail and go full speed ahead.
Maybe you will try to write an algorithm, as you see it, without the intervention of any of the other participants / developers, in order for the alternative point of view to sound fully. This can be a description in words of a sequence of actions or, as taught at school, in the form of squares, rhombuses, rectangles and ovals connected by arrows.

igorpecovnik, 2022-03-12:
tl;dr;

Let me add few bits why caching is done as it is done. Cache creation was always meant as a internal function and we provide cache via our servers. In case cash is not present, it builds from packages. Which is slow process.

Cache hash is rebuild if packages are changed, if version is changed, if there is no cache for current or previous month. This latter is needed to cover corner case - few days are needed that cache - once rebuild - gets to CDN. FORCE functionality is here to create a cache with this script and OFFSET to make it for next month without changing internal cache version number.

matthijskooijman, 2022-03-15:
@igorpecovnik Thanks for the additonal clarification. Am I understanding correctly that the online cache generation uses OFFSET to generate the next month's cache in advance, so it is available once clients start looking for it? If so, doesn't that mean that there is no need for clients to try the previous month cache version at all? Or is that intended as a fail-safe, in case the cache generation fails for some reason?

Also, am I correct in that https://github.com/armbian/scripts/blob/master/cacherebuild.sh is the script that drives cache generation, and it is ran inside github actions using https://github.com/armbian/scripts/blob/master/.github/workflows/build-rootfs-cache.yml ?

rpardini, 2022-03-17:
Nice discussion. I've left most of that code untouched, if moved-around, in armbian-next so it's very noisy logging-wise now due to all the `curl` invocations/failures. It needs a rewrite anyway considering `set -e` and such. I'd also like to add rootfs-like caching for shallow git bundles, so it is an area I am also likely to investigate somewhat soon. 
I am placing my bets on using `ghcr.io` as storage backend for both in a near future.

closed 2022-07-31 by igorpecovnik