armbian/build #3564: [discussion] ClockworkPi DevTerm A06

littlecxm, 2022-03-24:
Hi,I bought DTA06(DevTerm A06) recently and it use RK3399 chip
official dist is [announced](http://dl.clockworkpi.com/), however it only works on **[hardcoded](https://github.com/clockworkpi/DevTerm/blob/0d92f26384119bc15815ec7bca572b9f39c054e6/Code/patch/armbian_build_a06/patch/README#L7)** specific armbian version
I look the dist repo in here and I found it similar as armbian exists supported board OrangePi4
so it just like
```
SoC: RK3399
PMU: RK808
Codec: ES8323
Screen: CWD686 (MIPI Port)
```
and it with a little thermal printer
so could I try to (or help to) merge the [official repo](https://github.com/clockworkpi/DevTerm/tree/main/Code/patch/armbian_build_a06/patch) to mainline?

Tonymac32, 2022-03-24:
Right.  I also think it would be awesome if the clockwork pi team opened a PR here, then at least there would be some vendor involvement, it looks like a small series of patches all in all.

littlecxm, 2022-03-24:
> You can join as board maintainer https://forum.armbian.com/topic/19172-do-you-like-to-see-your-favorite-board-supported/#comment-130294

:D yeah I've posted there
based on community contribute rule i choose talk at here before coding

littlecxm, 2022-05-07:
Close by: https://github.com/armbian/build/pull/3768

closed 2022-05-07 by littlecxm