armbian/build #3647: Problem found in ...

kylongmu, 2022-04-07:
**Describe the bug**
[ o.k. ] Adding [ Wireless drivers for Realtek 8723DU chipsets branch:v5.13.4 ]
[ o.k. ] Checking git sources [ rtl8723du v5.13.4 ]
[ .... ] Creating local copy 
[ .... ] Fetching updates 
fatal: couldn't find remote ref v5.13.4
fatal: couldn't find remote ref v5.13.4
[ .... ] Checking out 
error: pathspec 'FETCH_HEAD' did not match any file(s) known to git

[ o.k. ] Checking git sources [ rtl88x2cs tune_for_jethub ]
[ .... ] Fetching updates 
fatal: couldn't find remote ref tune_for_jethub
fatal: couldn't find remote ref tune_for_jethub
[ .... ] Checking out 
error: pathspec 'FETCH_HEAD' did not match any file(s) known to git

[ o.k. ] Adding [ Wireless drivers for Realtek 8822BS chipsets  ]
[ o.k. ] Checking git sources [ rtl8822bs local_rtl8822bs ]
[ .... ] Fetching updates 
fatal: couldn't find remote ref local_rtl8822bs
fatal: couldn't find remote ref local_rtl8822bs
[ .... ] Checking out 
error: pathspec 'FETCH_HEAD' did not match any file(s) known to git

**To Reproduce**
./compile
select "UBoot and.."  ->"Do not.." ->"khadas-vim3l"

**Provide logs:**
Drop logs to pastebin service of your choice and supply logs from output/debug
[output.log](https://github.com/armbian/build/files/8439976/output.log)

adeepn, 2022-04-07:
This is your local problem. Clean armbian cache, check github avail and try again.
```
[ o.k. ] Adding [ Wireless drivers for Realtek 88x2cs chipsets branch:tune_for_jethub ]
[ o.k. ] Checking git sources [ rtl88x2cs tune_for_jethub ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Enumerating objects: 790, done.
remote: Counting objects: 100% (790/790), done.
remote: Compressing objects: 100% (515/515), done.
remote: Total 790 (delta 279), reused 779 (delta 272), pack-reused 0
Receiving objects: 100% (790/790), 4.23 MiB | 2.74 MiB/s, done.
Resolving deltas: 100% (279/279), done.
From https://github.com/jethome-ru/rtl88x2cs
 * branch            tune_for_jethub -> FETCH_HEAD
 * [new branch]      tune_for_jethub -> origin/tune_for_jethub
[ .... ] Checking out
[ o.k. ] Adding [ Wireless drivers for Realtek 8723DU chipsets branch:v5.13.4 ]
[ o.k. ] Checking git sources [ rtl8723du v5.13.4 ]
[ .... ] Creating local copy
[ .... ] Fetching updates
remote: Enumerating objects: 5018, done.
remote: Counting objects: 100% (1138/1138), done.
remote: Compressing objects: 100% (628/628), done.
remote: Total 5018 (delta 641), reused 916 (delta 506), pack-reused 3880
Receiving objects: 100% (5018/5018), 6.40 MiB | 7.37 MiB/s, done.
Resolving deltas: 100% (3883/3883), done.
From https://github.com/lwfinger/rtl8723du
 * branch            v5.13.4    -> FETCH_HEAD
 * [new branch]      v5.13.4    -> origin/v5.13.4
[ .... ] Checking out
[ o.k. ] * applying wireless-rtl8723du.patch
```

kylongmu, 2022-04-07:
> This is your local problem. Clean armbian cache, check github avail and try again.
>
> ```
> [ o.k. ] Adding [ Wireless drivers for Realtek 88x2cs chipsets branch:tune_for_jethub ]
> [ o.k. ] Checking git [...]

Thanks for your help, I tried with your suggest again, but the problem is same.  Even use the option REGIONAL_MIRROR=china .

kylongmu, 2022-04-07:
Modify the lib/configuration.sh GITHUB_SOURCE='https://gitclone.com/github.com' to other accelerator site can solve this problem. 
Build with option: RIGIONAL_MIRROR=china  
So I guess maybe the gitclone.com cached some wrong copy.
If the option can set to customer's favorite site with RIGIONAL_MIRROR="https://abc.com" will be better
`
case $GITHUB_MIRROR in
	fastgit)
		GITHUB_SOURCE='https://hub.fastgit.xyz'
		;;
	gitclone)
		GITHUB_SOURCE='https://gitclone.com/github.com'
		;;
	*)
		GITHUB_SOURCE='https://github.com'
		;;
esac

`

closed 2022-05-02 by igorpecovnik