armbian/build #3762: Debian’s dpkg package doesn’t support zstd compression ?

Haraade, 2022-05-05:
Build, from source.
sudo dpkg -i linux-image-current-odroidxu4_22.05.0-trunk_armhf.deb

dpkg-deb: error: archive 'linux-image-current-odroidxu4_22.05.0-trunk_armhf.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive linux-image-current-odroidxu4_22.05.0-trunk_armhf.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 linux-image-current-odroidxu4_22.05.0-trunk_armhf.deb

Haraade, 2022-05-05:
Bullseye

igorpecovnik, 2022-05-05:
@The-going  any clues?

The-going, 2022-05-05:
> @The-going any clues?

I don't know yet. I do not test and do not build debian.
But maybe the question is in the configuration?

@Haraade Show the difference between the default configuration and the configuration in the kernel directory.

igorpecovnik, 2022-05-05:
https://unix.stackexchange.com/questions/669004/zst-compression-not-supported-by-apt-dpkg

Haraade, 2022-05-05:
> > @The-going any clues?
> 
> I don't know yet. I do not test and do not build debian. But maybe the question is in the configuration?
> 
> @Haraade Forgive my inattention. Tell me which operating system the package was built on.

Build in docker. Host Os, Ubuntu jammy.

pyavitz, 2022-05-05:
either add to the packaging patch or mod the builddeb to force xz.

```sh
-	dpkg --build "$pdir" ..
+	dpkg-deb -Zxz --build "$pdir" ..```

The-going, 2022-05-05:
Or, someday we will start collecting packages for a specific operating system inside this native OS.
Even using cross compilation.

igorpecovnik, 2022-05-05:
> inside this native OS.

Yes, we are preparing that. https://github.com/armbian/build/actions/runs/2276249931 but this is yet another subsystem that would need to be maintained in order to work without (other) issues.

The-going, 2022-05-05:
@igorpecovnik Don't mind my grumbling. I am slowly moving in this direction, but very slowly.

The-going, 2022-05-05:
I think we can set this compression by default if not set.
```diff
diff --git a/lib/compilation.sh b/lib/compilation.sh
index 0edaad161..25292fbf6 100644
--- a/lib/compilation.sh
+++ b/lib/compilation.sh
@@ -521,7 +521,7 @@ CUSTOM_KERNEL_CONFIG
        eval CCACHE_BASEDIR="$(pwd)" env PATH="${toolchain}:${PATH}" \
                'make $CTHREADS $kernel_packing \
                KDEB_PKGVERSION=$REVISION \
-               KDEB_COMPRESS=${DEB_COMPRESS} \
+               KDEB_COMPRESS=${DEB_COMPRESS:-xz} \
                BRANCH=$BRANCH \
                LOCALVERSION="-${LINUXFAMILY}" \
                KBUILD_DEBARCH=$ARCH \
```
This needs to be tested.

igorpecovnik, 2022-05-05:
> Don't mind my grumbling. I am slowly moving in this direction, but very slowly.

Me too ;) This is something for future, low priority. I have added to CI just to monitor how fragile this is.

The-going, 2022-05-05:
> This is something for future, low priority.

This is a serious question for me. Today I have to limit myself to only one operating system, the one that is currently in the VM.

igorpecovnik, 2022-05-05:
> This is a serious question for me
> to limit

Of course.

Primary focus remains on Ubuntu LTS packages base, Debian stable 2nd. "Slowly" i meant we would like to make Debian based images with Debian host, Ubuntu with Ubuntu. Regardless of importance, we can only proceed there in best effort way.

belegdol, 2022-05-18:
I just tried the following build command:
```
./compile.sh BOARD=odroidxu4 BRANCH=current KERNEL_ONLY=yes KERNEL_CONFIGURE=no USE_CCACHE=no KEEP_KERNEL_CONFIG=no DEB_COMPRESS=xz
```
I am still getting the error:
```
dpkg-deb: error: archive '/home/julas/linux-dtb-current-odroidxu4_22.02.1.5_armhf.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /home/julas/linux-dtb-current-odroidxu4_22.02.1.5_armhf.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
dpkg-deb: error: archive '/home/julas/linux-image-current-odroidxu4_22.02.1.5_armhf.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /home/julas/linux-image-current-odroidxu4_22.02.1.5_armhf.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
```

The-going, 2022-05-19:
> BOARD=odroidxu4 BRANCH=current ..... DEB_COMPRESS=xz

https://github.com/armbian/build/blob/2c0a0a5e2382e1f9ad29139cecf49b99af739551/config/sources/families/odroidxu4.conf#L18-L29
The following part of the code will be executed for the `BRANCH=current` parameter.
https://github.com/armbian/build/blob/2c0a0a5e2382e1f9ad29139cecf49b99af739551/lib/compilation-prepare.sh#L54-L58
My recommendation to apply the parameter 'DEB_COMPRESS=xz' is valid only for the kernel version greater than 5.10
You will have to fix this patch yourself. Use the advice:
https://github.com/armbian/build/issues/3762#issuecomment-1118956198

https://github.com/armbian/build/blob/2c0a0a5e2382e1f9ad29139cecf49b99af739551/patch/misc/general-packaging-5.3.y.patch#L73

igorpecovnik, 2022-05-19:
FYI perhaps related https://armbian.atlassian.net/browse/AR-1194

The-going, 2022-05-19:
Igor, good evening. I understood what needs to be done and how to do it.
I just don't have enough time for this job. That's why I'm giving advice here.

The problem needs to be fixed using an algorithm similar to what we use for the chroot environment.
First we create a system environment, and then we run package build scripts in it.

igorpecovnik, 2022-05-19:
> I just don't have enough time

OK, understand. The same here :( 

Anyway we are moving best effort / when possible ...

The-going, 2022-05-19:
`REPOSITORY_UPDATE="update" ` I'll look at it tomorrow.

igorpecovnik, 2022-05-20:
More hints - also rockchip64 and rk3399 legacy share the same problem. Only source package is added, but not the rest.

The-going, 2022-05-20:
Okay, I'll take a look.

belegdol, 2022-05-20:
Both rockchip64 and rk3399 use the plain `dpkg --build "$pdir" ..` call, not respecting `DEB_COMPRESS=xz`, so this might indeed be related. What is the preferred way of updating the packaging patches?

belegdol, 2022-05-20:
https://github.com/aptly-dev/aptly/pull/1050

The-going, 2022-05-20:
@belegdol you said:

> I am building on jammy for bullseye.

please publish:
```dpkg --info output/debs/linux-image-current-*```

belegdol, 2022-05-20:
With my PR merged and built with `DEB_COMPRESS=xz`
```
$ dpkg --info output/debs/linux-image-current-odroidxu4_22.02.1.6_armhf.deb 
 new Debian package, version 2.0.
 size 29767148 bytes: control archive=48892 bytes.
     414 bytes,    12 lines      control              
  193016 bytes,  1895 lines      md5sums              
     423 bytes,    15 lines   *  postinst             #!/bin/bash
     303 bytes,    12 lines   *  postrm               #!/bin/bash
     936 bytes,    34 lines   *  preinst              #!/bin/bash
     301 bytes,    12 lines   *  prerm                #!/bin/bash
 Package: linux-image-current-odroidxu4
 Source: linux-5.4.195-odroidxu4
 Version: 22.02.1.6
 Architecture: armhf
 Maintainer: Igor Pecovnik <igor.pecovnik@****l.com>
 Installed-Size: 96425
 Section: kernel
 Priority: optional
 Homepage: http://www.kernel.org/
 Description: Linux kernel, version 5.4.195-odroidxu4
  This package contains the Linux kernel, modules and corresponding other
  files, version: 5.4.195-odroidxu4.
```

PR #3795 (2022-05-20): Honor $DEB_COMPRESS on kernels 5.3+
# Description

This partially fixes #3762 if `DEB_COMPRESS=xz` is used.

Jira reference number [AR-1194]

# How Has This Been Tested?

- [x] Installed created packages on an Odroid HC1 running Debian bullseye
- [x] Ran `./compile REPOSITORY_UPDATE="update"`

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [ ] I have made corresponding changes to the documentation
- [ ] My changes generate no new warnings
- [  [... cut]
commit 247c4c45 (2022-05-20): Honor $DEB_COMPRESS on kernels 5.3+ (#3795)

fix: add the KDEB_COMPRESS variable to the debuild script of the kernel build version 5.3+

This partially fixes #3762 if DEB_COMPRESS=xz is used.
Jira reference number AR-1194
closed 2022-05-20 by The-going