armbian/build #3796: Build kernel config Rock 3A edge / No boot emmc or access mmc whit sd card boot

J108560, 2022-05-21:
Since edge kernel config file for rock 3a move to config linux-rockchip64-edge.config all EDGE image do not boot from mmc (sdcard work). Also, when boot from SD Card the MMC is not accessible. In my opinion you need to edit the line # CONFIG_MMC_SDHCI_OF_DWCMSHC to **m** or **y**. This driver is necessary for rock 3a. My own build image whit this line edit work as expected.

Best regards Sven

EvilOlaf, 2022-05-21:
Feel free to provide a pull request with the mentioned changes.

Take note that in this repository we only track issues with _building_ Armbian, not running. For latter please refer to forums.

closed 2022-05-21 by EvilOlaf

J108560, 2022-05-21:
dann eben nicht.....

igorpecovnik, 2022-05-21:
> dann eben nicht.....

There are hundreds of hints and fixes reported on forum, but no matter how good our organisation is and how many hours [A few of us](https://github.com/armbian/build/graphs/contributors?from=2021-10-20&to=2022-05-21&type=c) will never be able to integrate all of them. In fact, our 30 hours investment into public code can only cover a few. And for those hours they get nothing but complains and more complains afterwards since it is impossible to fix everything.