armbian/build #3820: Patch 5.4.196 for odroidxu4-current fails

belegdol, 2022-05-26:
I tried updating odroidxu4-current to 5.4.196. Unfortunately, the patch 5.4.195 to 5.4.195 conflicts with aufs5-standalone.patch from `debian/patches/features/all/aufs5/` folder. I tried adding a patch removing the problematic section from the aufs5 patch to `patch/kernel/archive/odroidxu4-5.4`, but the problem is that the debian patches are already being applied before. What would be the preferred way of fixing this? According to http://aufs.sourceforge.net/, aufs5 is not supported with kernel-5.4 since last September anyway. Can we just drop the patches altoghether?

igorpecovnik, 2022-05-26:
>  Can we just drop the patches altoghether?

Yes, you can drop AUFS support on this kernel.

igorpecovnik, 2022-05-26:
IMO move to 5.10.y and also remove this line https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh#L185 and I think we are good to go.

PR #3822 (2022-05-26): Drop AUFS from kernels below 5.10 as these are not supported by upstream
# Description

AUFS patches are incompatible with Linux commit f00432063db1a0db484e85193eccc6845435b80e, which has been backported to stable branches. As upstream AUFS does not support anything below kernel 5.10, drop the patch on these older kernels. Fixes #3820.

# How Has This Been Tested?

Please describe the tests that you ran to verify your changes. Please also note any relevant details for your test configuration.

- [x] Built kernel 5.4.196 and saw no patch errors

# Checklist:

- [x] My code follows the style gui [... cut]
closed 2022-05-26 by igorpecovnik