armbian/build #3850: During build of rootfs there are lot of  messages "cannot change locale"

iav, 2022-06-04:
On rootfs creation step I see a lot of messages like 
```
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "en_US:en",
        LC_ALL = "en_US.UTF-8",
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
```
```
[ o.k. ] Applying distribution specific tweaks for [ jammy ]
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[ o.k. ] Applying common tweaks
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[ .... ] Cleaning [ package lists ]
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
```
in my config set `DEST_LANG="en_GB.UTF-8"`

iav, 2022-06-05:
Now I see fix works only for ubuntu, but not debian.
Working on second version.

iav, 2022-06-05:
Now `$DEST_LANG` config var set default codepage in rootfs cache image. Is it correct? Maybe, it should set when new image builds on top of generic cached rootfs?
@The-going ?

The-going, 2022-06-05:
> Now `$DEST_LANG` config var set default codepage in rootfs cache image. Is it correct? Maybe, it should set when new image builds on top of generic cached rootfs?

The encoding en_US.UTF-8 must be set at the very beginning, at the first stage of the debutstrap process
in an unconditional order. This is due to the fact that it is basic and contains not only letters but also icons -
lines, corners, double lines, what frames and tables are drawn in the console tools (ncurses library).
For example, in the mc file manager.

The-going, 2022-06-05:
Вопрос. Какая цель поставлена?
От этого сильно зависит алгоритм реализации.

Посмотри внимательно на https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L25
Этого файла не касалась рука дизайнера.
С точки зрения общего алгоритма, последовательность действий здесь выглядит более оптимально.
Сначала создаётся архив базовой системы:
https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L177-L188
Который состоит из строго определённой последовательности действий:
 - 1 https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L61-L65
 - 2 https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L75
 - 3 https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L78
 - 4 https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L85-L87
 - 5 https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L103
 Это всё! Можно заворачивать в архив.
Если есть желание опртимизировать алгоритм, сообщи. Вместе реализовать получится быстрее.

iav, 2022-06-05:
Однако вот выставил я в конфиге, что хочу дефолтную британскую локаль — и вижу именно её в запакованном rootfs cache. И делается это во втором месте, которое я тронул: debootstrap.sh:253:create_rootfs_cache()

Так вот как минимум при сборке BOARD=helios64 BRANCH=edge RELEASE=bullseye
код debootstrap.sh:253:create_rootfs_cache() работает, а тот, что ты процитировал — нет.

iav, 2022-06-05:
Да, цель — всё должно работать _правильно_ в широком смысле этого слова. Сейчас оно валит на экран ошибки — значит, что-то там не так, и если результат удовлетворительный, то это случайность. Надо, чтобы было _нормально._

The-going, 2022-06-05:
Хорошо. Я цитировал из файла chroot-buidpackages.sh
https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/chroot-buildpackages.sh#L85-L87
здесь сначала редактируется файл /etc/locale.gen.
Могу предположить, что это необходимо для дебиан или старых дистибутивов.  Но я не проверял.
В новых Ubuntu этот шаг не требуется.

Мы разговарим про файл debootstrap.sh и здесь:
https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/debootstrap.sh#L258
ты устанавливаеш американскую локаль в безусловном порядке

The-going, 2022-06-05:
Раньше, когда этого небыло и я создавал корневую систему локально, получалась ерунда.
Создавалась только русская локаль и рамка в `mc` рисовалась символами `@`

iav, 2022-06-05:
а ты кеши rootfs сам создаёшь, или готовые качаешь?
У меня почему-то они всегда создаются локально — не знаю уж, чем я это спровоцировал. Но мне кажется, что если выбранная локаль отрабатывается не при создании конечного образа, а при создании рутфс (как я вижу сейчас) — то и поменять локаль в процессе созданеия образа _нормально_ не получится. И кажется мне, что я сейчас нашёл ошибку в дизайне процесса. Но я в этом пока не уверен полностью.

The-going, 2022-06-05:
Корневую систему я создаю локально, когда мне требуется что то отлчающееся от заданного по умолчанию.
https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/debootstrap.sh#L122
Если этот параметр установлен, будет сделана только корневая файловая система и процесс завершится.

> если выбранная локаль отрабатывается не при создании конечного образа, а при создании рутфс (как я вижу сейчас) — то и поменять локаль в процессе созданеия образа _нормально_ не получится.

Совершенно верно.

iav, 2022-06-05:
> здесь сначала редактируется файл /etc/locale.gen.
> Могу предположить, что это необходимо для дебиан или старых дистибутивов.

именно так. только ubuntu это понимает. 
И всё же больше всего меня смущает то, что я не вижу, чтобы так тщательно тобой отрецензированный мне тут код из `create_chroot() ` у меня вообще вызывался.

iav, 2022-06-05:
I tried it.
First I made rootfs with `DEST_LANG="en_US.UTF-8"`, then when constructing the image I set `DEST_LANG="zu_ZA.UTF-8"`.
Result image in which en_US.UTF-8, zu_ZA.UTF-8 will not allowed anywhere. @igorpecovnik, is it "as planned", or should locale set be moved from the preparation of rootfs to the image preparation?

igorpecovnik, 2022-06-05:
IIRC it was set to rootfs to not have errors there, but it should probably be moved to image creation, yes.

The-going, 2022-06-05:
Please excuse me, I'm slow to answer.

I want everyone to understand me correctly. I am trying to draw an analogy between two algorithms for creating a root file system and the actions that follow after that.
`chroot-buildpackages.sh <-> debootstrap.sh`

In the create_rootfs_cache function, it is necessary to generate the locale en_US.UTF-8 unconditionally.
This line stays in place: https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/debootstrap.sh#L258

These lines: https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/debootstrap.sh#L259-L260
yes, need to be moved to the function https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/distributions.sh#L622
There is already a logic for choosing an OS and you can apply various options for Debian and Ubuntu, if necessary.

Why is this line necessary? I don't understand. https://github.com/armbian/build/blob/4b05414c19d40a94b794922e67907176e70073d7/lib/debootstrap.sh#L257

igorpecovnik, 2022-06-05:
> yes, need to be moved to the function
> There is already a logic for choosing an OS and you can apply various options for Debian and Ubuntu, if necessary.

Yeah, agree.

The-going, 2022-06-05:
I keep talking about the algorithm.
We can make the function of creating a root file system in a general way. A function as a separate executable file that accepts command-line arguments: OS, architecture, variant....
We will call this function to create packages in a chroot environment. We will call this function to create the root file system of the future image. We will call this function to create a target file system for the build system itself and run it in docker.
I think the idea is clear.

iav, 2022-06-05:
then, please, review #3856
I shall to run for focal|jammy|buster|sid|bullseye, it will take a time.

The-going, 2022-06-07:
@iav which OS do you usually create and use ubuntu or debian?

iav, 2022-06-07:
Usually I create `jammy` and `bullseye`, run on jammy with and without docker.
U did tests with all currently enabled os.

The-going, 2022-06-07:
I am testing debian buster in a chroot environment (chroot-buildpackages.sh ).
For Debian, I found some problems.

The-going, 2022-06-07:
Посмотри, существует ли в твоём комплектном дебиан каталог /usr/lib/locale?

iav, 2022-06-07:
Что значит «в комплектном дебиан»? Это где именно? в кеше рутфс, в итоговом образе, ещё где-то? 
Это плохо?

The-going, 2022-06-07:
Прошу прощения. У меня был сбой и я не ншёл эту папку.

The-going, 2022-06-07:
I see for debian:
```
I: Configuring the base system...
....
I: Configuring locales...
....
[ o.k. ] cmdline before: [ en_US.UTF-8 en_US.utf-8 C.UTF-8 ru_RU.UTF-8
**                                                                                                                                                                             
***                                                                                                                                                                            
C                                                                                                                                                                              
POSIX                                                                                                                                                                          
****                                                                                                                                                                           
LANG=C                                                                                                                                                                         
LANGUAGE=en_US:en                                                                                                                                                              
LC_CTYPE="C"                                                                                                                                                                   
LC_NUMERIC="C"                                                                                                                                                                 
LC_TIME="C"                                                                                                                                                                    
LC_COLLATE="C"                                                                                                                                                                 
LC_MONETARY="C"                                                                                                                                                                
LC_MESSAGES="C"                                                                                                                                                                
LC_PAPER="C"                                                                                                                                                                   
LC_NAME="C"                                                                                                                                                                    
LC_ADDRESS="C"                                                                                                                                                                 
LC_TELEPHONE="C"                                                                                                                                                               
LC_MEASUREMENT="C"                                                                                                                                                             
LC_IDENTIFICATION="C"                                                                                                                                                          
LC_ALL=C                                                                                                                                                                       
*****                                                                                                                                                                          
#  File generated by update-locale                                                                                                                                             
******                                                                                                                                                                         
C.UTF-8 ]
[ o.k. ] File exist [ /etc/locale.gen ]
[ o.k. ] invalid locale for file /etc/locale.gen [ en_US.utf-8 ]
[ o.k. ] invalid locale for file /etc/locale.gen [ C.UTF-8 ]
[ o.k. ] cmd: locale-gen [ Generating locales (this might take a while)...
  en_US.UTF-8... done                                                                                                                                                          
  ru_RU.UTF-8... done                                                                                                                                                          
Generation complete. ]
[ o.k. ] cmdline after: [ 
en_US.utf8                                                                                                                                                                     
ru_RU.utf8                                                                                                                                                                     
**                                                                                                                                                                             
[... cut ...]
LC_ALL=                                                                                                                                                                        
******                                                                                                                                                                         
C.UTF-8                                                                                                                                                                        
locale-archive ]
```

The-going, 2022-06-07:
@iav I'm asking stupid questions. My main OS is openSUSE.
I don't know much about Debian.

iav, 2022-06-07:
Насколько я понимаю, локаль `C`  — особый-специальный случай, и для неё не нужна какая-то специальная генерация. Собственно, в файле с доступными локалями она есть только в свежих убунтах, в дебиане её там и нет, и генерация для неё не проходит. По этому-то я её из дебиановской версии и убрал. Хотя сначала, пока ещё не знал, что у убунты и дебиана это по разному работает, вставил.

iav, 2022-06-07:
Однако, я пока так и не понял, что именно у вас _не так._ 
Как тестить? Что запускать, должно получаться, что получается?

iav, 2022-06-07:
конструкция редактирования sed -i '/ C.UTF-8/s/^# //g' /etc/locale.gen не имеет смысла в дебиане, но имеет в убунте. Потому что в убунте эта строка есть. Соответсвенно, в дебиане она просто ничего не делает.

iav, 2022-06-07:
потому мне _кажется,_ что в дебиане просто нет локали C.UTF-8 в явном виде, так, чтобы в списках мелькать.
ну, нет и нет.

The-going, 2022-06-07:
Хорошо понял.
Ты сборку образов проверял?

closed 2022-06-07 by igorpecovnik

iav, 2022-06-07:
> Хорошо понял. Ты сборку образов проверял?

Да. Собрал для Helios64 с русской локалью образ всех ~~пяти семей Нью-Йорка~~ поддерживаемых дебианов и убунт, и убедился, что оно на реальном железе грузится, и mc в сессии ssh говорит мне по русски и рамочки при этом рисует.

The-going, 2022-06-08:
Вот за это от всех пользователей и меня лично большое спасибо!

PR #3856 (2022-06-07): Missed locale warning #3850 
Fix #3850 second try.
The first was ubuntu-only. Hope this will work for debian and ubuntu both.
additional tests needed.