armbian/build #3872: Helios64 build errors - current kernel - fullos - bullseye - SPI Error

gr3k, 2022-06-09:
Hello, 
due to missing helios64 support / new builds I wanted to build image myself.
Unfortunately I get some errors ... maybe You can help me in some way ? 

  LD [M]  net/vmw_vsock/vmw_vsock_virtio_transport_common.o
  CC [M]  net/openvswitch/vport-gre.o
  LD [M]  net/openvswitch/openvswitch.o
scripts/sign-file.c: In function ‘display_openssl_errors’:
scripts/sign-file.c:89:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   89 |         while ((e = ERR_get_error_line(&file, &line))) {
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/sign-file.c: In function ‘drain_openssl_errors’:
scripts/sign-file.c:102:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  102 |         while (ERR_get_error_line(&file, &line)) {}
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/extract-cert.c: In function ‘display_openssl_errors’:
scripts/extract-cert.c:46:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   46 |         while ((e = ERR_get_error_line(&file, &line))) {
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/extract-cert.c: In function ‘drain_openssl_errors’:
scripts/extract-cert.c:59:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   59 |         while (ERR_get_error_line(&file, &line)) {}
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:22.11-27.6: Warning (spi_bus_reg): /fragment@1/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:36.11-41.6: Warning (spi_bus_reg): /fragment@2/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:50.11-55.6: Warning (spi_bus_reg): /fragment@3/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:64.11-69.6: Warning (spi_bus_reg): /fragment@4/__overlay__/spidev: SPI bus unit address format error, expected "0"
arch/arm64/boot/dts/rockchip/rk3399-tinker-2.dts:401.23-422.4: Warning (i2c_bus_reg): /i2c@ff3c0000/vdd_cpu_b: I2C bus unit address format error, expected "60"
arch/arm64/boot/dts/rockchip/rk3399-tinker-2.dts:497.19-518.4: Warning (i2c_bus_reg): /i2c@ff3d0000/vdd_gpu: I2C bus unit address format error, expected "60"
drivers/spi/spi-rockchip.c:767:12: error: redefinition of ‘rockchip_spi_setup’
make[2]: *** [scripts/Makefile.build:288: drivers/spi/spi-rockchip.o] Error 1
make[1]: *** [scripts/Makefile.build:551: drivers/spi] Error 2
make: *** [Makefile:1868: drivers] Error 2
[ error ] ERROR in function compile_kernel [ main.sh:588 -> main.sh:489 -> compilation.sh:508 -> general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

closed 2022-06-09 by igorpecovnik

iav, 2022-06-09:
Provide build parameters and confirm you use ubuntu jammy as build host.
I got bullseye image for helios64 without error.

gr3k, 2022-06-09:
@iav , i just do a : ./compile.sh docker after git clone armbian build repo ,so yes it was ubuntu jammy.
I choice full os, do not change kernel cfg, helios64 from csc , current, bullseye, image with console , standard image 

I changed to edge instead of current, and it was successfully builded. Now im going to test it ...

iav, 2022-06-09:
Try to build `edge` or `legacy` instead of `current`

iav, 2022-06-10:
or add into `userpatches/lib.config` line `KERNELBRANCH='tag:v5.15.45'` to use earllier kernel version

gr3k, 2022-06-10:
@iav I can confirm, 
Today I done one more test. **I'm able** to compile **current** ,but only when I choice 'Use precompiled packages from Armbian repository'.
Choicing 'Do not change the kernel configuration' option at kernel configuration screen generates errors.

I ran helios64 from yesterday edge build .  It was ran for more than 10h : Armbian 22.08.0-trunk Bullseye . 5.18.3-rockchip64
now ,i need to compile ZFS due to missing zfs-dkms kernel modules....

igorpecovnik, 2022-06-10:
This is end user problem from our perspective and we don't like to see that among build system issues. There are thousands of amateurs [on forum](https://forum.armbian.com/) **that have time** and know-how to help. I am pretty sure answers can be obtained from searching the forum topics and I remember already giving guidance on this topic.