armbian/build #3883: [Question] How to support Kprobes in armbian linux kernel?

diyism, 2022-06-11:
When I try to build anbox-module, it complains:

```
$ apt install linux-headers-current-sunxi64
$ git clone https://github.com/diyism/anbox-modules.git
$ cd anbox-modules
$ ./INSTALL.sh
DKMS make.log for anbox-ashmem-1 for kernel 5.15.43-sunxi64 (aarch64)
Sat 11 Jun 2022 08:56:51 PM HKT
make -C /lib/modules/5.15.43-sunxi64/build V=0 M=$PWD
make[1]: Entering directory '/usr/src/linux-headers-5.15.43-sunxi64'
  CC [M]  /var/lib/dkms/anbox-ashmem/1/build/deps.o
  CC [M]  /var/lib/dkms/anbox-ashmem/1/build/ashmem.o
/var/lib/dkms/anbox-ashmem/1/build/deps.c:9:3: error: #error "Your kernel does not support KProbes, but this is required to compile ashmem as a kernel module on kernel 5.7 and later"
    9 | # error "Your kernel does not support KProbes, but this is required to compile ashmem as a kernel module on kernel 5.7 and later"
```

How to support Kprobes in armbian linux kernel?

EvilOlaf, 2022-06-11:
I suggest to build your own kernel package using the build framework and enable `CONFIG_KPROBES` during the process.
https://docs.armbian.com/Developer-Guide_Build-Preparation/

diyism, 2022-06-11:
@EvilOlaf  Thanks, I'll give a try.

igorpecovnik, 2022-06-11:
You can also send a PR with enabling this everywhere so this will work next time you will need it. And for others. Here:
https://github.com/armbian/build/tree/master/config/kernel
Some hase it, some don't.

diyism, 2022-06-11:
No luck, I modified  "build/config/kernel/linux-sunxi64-current.config" to enable "CONFIG_KPROBES=y"
and then exec:
```

./compile.sh BOARD=pine64 BRANCH=current KERNEL_ONLY=yes RELEASE=focal
...
  CC      spl/lib/elf.o
  AR      spl/lib/built-in.o
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  SYM     spl/u-boot-spl.sym
  COPY    spl/u-boot-spl.bin
  MKIMAGE spl/sunxi-spl.bin
  BINMAN  all
[ o.k. ] Building deb [ linux-u-boot-current-pine64_22.08.0-trunk_arm64.deb ]
[ error ] ERROR in function compile_uboot [ main.sh:588 -> main.sh:481 -> compilation.sh:321 -> general.sh:0 ]
[ error ] Building u-boot package failed 
[ o.k. ] Process terminated
```

igorpecovnik, 2022-06-11:
It must be some host dependency problem if you can't pass u-boot compilation. Use Docker.

iav, 2022-06-11:
And use `KERNEL_CONFIGURE=yes` option to invoke kernel config. then you get kernel config in `output` subtree, and copy it into `userpatches` directory for use in future.

closed 2022-06-19 by igorpecovnik