armbian/build #3897: 30-armbian-sysinfo doesn't show network usage anymore

sej7278, 2022-06-14:
This line is never going to work, its basically comparing the string `vnstat.db` with the string `eth0`:

https://github.com/armbian/build/blob/master/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo#L159

surely it should be grep'ing /var/lib/vnstat/vnstat.db for eth0 or similar?

looks like https://github.com/armbian/build/commit/5082eb9d4e03e0cdd211708424df3928ac63f67e broke it - seems to want to suppress an error message on firstboot, but ends up suppressing all output forever! comment out that line and it works again.

AristoChen, 2022-07-16:
Hi @sej7278, 

Thanks for reporting this, I will check this next week.

AristoChen, 2022-07-19:
Hi @sej7278,

Looks like it might be `vnstat` that breaks the function. Previously, all the available internet interfaces will be listed under `/var/lib/vnstat`, so this [line](https://github.com/armbian/build/blob/master/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo#L159) will be able to check if `PRIMARY_INTERFACE` exists in vnstat or not. However, only `vnstat.db` exists in `/var/lib/vnstat` now, which makes this function broken.

Here is the patch, please let me know whether it works for you, thanks!
```patch
diff --git a/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo b/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo
index 417a6f941..447a37219 100755
--- a/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo
+++ b/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo
@@ -155,11 +155,8 @@ display "storage temp" "$storage_temp" $HDD_TEMP_LIMIT "0" "°C" "" ; a=$((a+$?)
 display "Battery" "$battery_percent" "20" "1" "%" "$status_battery_text" ; a=$((a+$?))
 (( $a > 0 )) && echo "" # new line only if some value is displayed

-# Check whether PRIMARY_INTERFACE exist in /var/lib/vnstat/
-PRIMARY_INTERFACE=$(comm -12 <(ls -1 /var/lib/vnstat/ 2> /dev/null) <(echo "$PRIMARY_INTERFACE" | sed 's/+/\n/g') | sed -n -e 'H;${x;s/\n/+/g;s/^+//;p;}')
-
 line=0
-if [[ $(command -v vnstat) && -n $PRIMARY_INTERFACE ]]; then
+if [[ $(command -v vnstat) && $(vnstat --iflist | grep -c $(echo "$PRIMARY_INTERFACE" | sed 's/+/\n/g')) -eq 1 ]]; then
        traffic=$(vnstat -i $PRIMARY_INTERFACE --oneline | cut -d";" -f4,5)
        traffic_rx=$(echo $traffic | cut -d";" -f1,1 | sed -r 's/([0-9]+\.[0-9]{1})[0-9]*/\1/')
        traffic_tx=$(echo $traffic | cut -d";" -f2,2 | sed -r 's/([0-9]+\.[0-9]{1})[0-9]*/\1/')

```

---
Here is the test result on Khadas VIM1

- Before the patch
``` shell
root@khadas-vim1:~# /etc/update-motd.d/30-armbian-sysinfo
System load:   2%               Up time:       1:39
Memory usage:  11% of 1.89G     IP:            192.168.0.102
CPU temp:      46°C             Usage of /:    36% of 15G
```

- After the patch
```shell
root@khadas-vim1:~# /etc/update-motd.d/30-armbian-sysinfo
System load:   2%               Up time:       1:40
Memory usage:  11% of 1.89G     IP:            192.168.0.102
CPU temp:      46°C             Usage of /:    36% of 15G
RX today:      0 B
```

neheb, 2022-07-19:
```
[ $(command -v vnstat) && $(vnstat --iflist | grep -c $(echo "$PRIMARY_INTERFACE" | sed 's/+/\n/g')) -eq 1 ]
  ^------------------^ SC2046: Quote this to prevent word splitting.
                       ^-- SC2107: Instead of [ a && b ], use [ a ] && [ b ].
                          ^-- SC2046: Quote this to prevent word splitting.
                                                      ^-- SC2046: Quote this to prevent word splitting.
                                                        ^-- SC2001: See if you can use ${variable//search/replace} instead.

```

AristoChen, 2022-07-19:
Hi @neheb, 

Thanks for the checking, may I know what version of shellcheck are you using? cuz the message generated [here](https://www.shellcheck.net/) is different from yours.

I have updated the code, which generate no message by using shellcheck online.
```patch
diff --git a/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo b/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo
index 417a6f941..70610079c 100755
--- a/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo
+++ b/packages/bsp/common/etc/update-motd.d/30-armbian-sysinfo
@@ -155,11 +155,8 @@ display "storage temp" "$storage_temp" $HDD_TEMP_LIMIT "0" "°C" "" ; a=$((a+$?)
 display "Battery" "$battery_percent" "20" "1" "%" "$status_battery_text" ; a=$((a+$?))
 (( $a > 0 )) && echo "" # new line only if some value is displayed

-# Check whether PRIMARY_INTERFACE exist in /var/lib/vnstat/
-PRIMARY_INTERFACE=$(comm -12 <(ls -1 /var/lib/vnstat/ 2> /dev/null) <(echo "$PRIMARY_INTERFACE" | sed 's/+/\n/g') | sed -n -e 'H;${x;s/\n/+/g;s/^+//;p;}')
-
 line=0
-if [[ $(command -v vnstat) && -n $PRIMARY_INTERFACE ]]; then
+if [[ "$(command -v vnstat)" ]] && [[ $(vnstat --iflist | grep -c "$PRIMARY_INTERFACE") -eq 1 ]]; then
        traffic=$(vnstat -i $PRIMARY_INTERFACE --oneline | cut -d";" -f4,5)
        traffic_rx=$(echo $traffic | cut -d";" -f1,1 | sed -r 's/([0-9]+\.[0-9]{1})[0-9]*/\1/')
        traffic_tx=$(echo $traffic | cut -d";" -f2,2 | sed -r 's/([0-9]+\.[0-9]{1})[0-9]*/\1/')

```

neheb, 2022-07-19:
@AristoChen if you noticed, I removed double [[]].

AristoChen, 2022-07-19:
@neheb Oh I see, so is it prefered to use `[]` instead of `[[]]`? cuz `[[]]` seems widely used in this project.

neheb, 2022-07-19:
probably not. [[]] is bash specific. [] shows more warnings in shellcheck. only reason why I used it.

closed 2022-07-31 by igorpecovnik