armbian/build #3902: Build attempt failing with: Failure trying to run: /sbin/ldconfig

cheetah2003, 2022-06-16:
Trying to build the bullseye minimal image.

Building machine is: i7-4770K, 32GB RAM, Debian 10 (buster)

Target machine: Rockchip RK3329 (FriendlyElec NanoPi R2S)

Relevant output from ./compile.sh:
> I: Extracting util-linux...
> I: Extracting libxxhash0...
> I: Extracting liblzma5...
> I: Extracting zlib1g...
> [ o.k. ] Installing base system [ Stage 2/2 ]
> W: Failure trying to run:  /sbin/ldconfig
> W: See //debootstrap/debootstrap.log for details
> [ error ] ERROR in function create_rootfs_cache [ main.sh:588 -> main.sh:549 -> debootstrap.sh:58 -> debootstrap.sh:238 -> general.sh:0 ]
> [ error ] Debootstrap base system for current nanopi-r2s bullseye   yes second stage failed 
> [ o.k. ] Process terminated 
> [ error ] unmount_on_exit() called! [ main.sh:1 -> image-helpers.sh:0 ]
> [ o.k. ] Unmounting [ /home/cheetah/src/armbian/build/.tmp/rootfs-ab08a0e0-9c4e-44fc-a7b2-bf2088912fa4/ ]
> [ error ] ERROR in function unmount_on_exit [ main.sh:1 -> image-helpers.sh:94 -> general.sh:0 ]
> [ error ] debootstrap-ng was interrupted 
> [ o.k. ] Process terminated 

I tried to locate this 'debootstrap.log', but it doesn't exist anywhere within the project's folders.  I used 'find' to show me all .log files, and it's just not there anywhere.   There is a 'debootstrap-list.log' in ./output/debug -- but that just seems to be a list of packages.

I tried running as normal user and as root, but still get the same error.  Sudo is available to the build process.  Any help would be much appreciated!

iav, 2022-06-16:
Builder need to run on ubuntu. If you use debian — run builder with docker (like `./compile.sh docker BOARD=...`
Or use virtual machine with ubuntu.

cheetah2003, 2022-06-17:
That did the trick.  Thanks iav!

closed 2022-06-17 by cheetah2003