armbian/build #3936: OrangePi 4 Legasy with jammy

AndreiCherniaev, 2022-06-25:
I try to build armbian (full OS) for OrangePi with old kernel 4 for jammy.

My log:
_[ .... ] Extension being added [ rkbin-tools :: added by main.sh:381 -> configuration.sh:172 -> config/sources/families/rk3399.conf:1 -> config/sources/families/include/rockchip64_common.inc:1 ]
[ o.k. ] Extension manager [ processed 3 Extension Methods calls and 3 Extension Method implementations ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ jammy ]
[ .... ] Installing build dependencies 
[ o.k. ] Syncing clock [ host ]
[ o.k. ] Checking for external GCC compilers 
[ error ] ERROR in function download_and_verify [ main.sh:588 -> main.sh:433 -> general.sh:1561 -> general.sh:1804 -> general.sh:0 ]
[ error ] verification failed 
[ o.k. ] Process terminated_ 

But the same I get for focal. Is it ok?

My host os is Ubuntu 22 x86-64.

EvilOlaf, 2022-06-26:
Check `output/debug` folder for more clues.

AndreiCherniaev, 2022-06-26:
> Check `output/debug` folder for more clues.

What should I do?

_Displaying message: Checking for external GCC compilers  info
gpg: error reading key: No public key
gpg: keyserver receive failed: Server indicated a failure
gpg: error reading key: No public key
gpg: keyserver receive failed: Server indicated a failure
gpg: assuming signed data in '/home/a/Documents/build/cache/toolchain/gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz'
gpg: Signature made Tue 19 Nov 2013 01:00:03 PM +04
gpg:                using RSA key D47877808F427EAF
gpg: Can't check signature: No public key
Displaying message: ERROR in function download_and_verify main.sh:588 -> main.sh:433 -> general.sh:1561 -> general.sh:1804 -> general.sh:0 err
Displaying message: verification failed  err
[logs-26_06_2022-01_59_08.zip](https://github.com/armbian/build/files/8986118/logs-26_06_2022-01_59_08.zip)
_

igorpecovnik, 2022-06-26:
Try with Docker.

./compile.sh docker

AndreiCherniaev, 2022-06-26:
> Try with Docker.
> 
> ./compile.sh docker

_libssl3 is already the newest version (3.0.2-0ubuntu1.5).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 udev : Depends: libudev1 (= 249.11-0ubuntu3) but 249.11-0ubuntu3.3 is to be installed
E: Unable to correct problems, you have held broken packages.
The command '/bin/sh -c apt-get update     && apt-get -y dist-upgrade     && apt-get install -y --no-install-recommends        $CUSTOM_PACKAGES        acl        aptly        aria2        bc        binfmt-support        binutils        bison        btrfs-progs        build-essential        busybox        ca-certificates        ccache        cpio        cryptsetup        cryptsetup-bin        debian-archive-keyring        debian-keyring        debootstrap        device-tree-compiler        dialog        distcc        dosfstools        dwarves        f2fs-tools        fakeroot        flex        gawk        gcc-aarch64-linux-gnu        gcc-arm-linux-gnueabihf        gcc-arm-linux-gnueabi        gcc-arm-none-eabi        gdisk        git        imagemagick        jq        kmod        libbison-dev        libc6-amd64-cross        libc6-dev-armhf-cross        libfdt-dev        libelf-dev        libfile-fcntllock-perl        libfl-dev        liblz4-tool        libncurses5-dev        libpython2.7-dev        libpython3-dev        libssl-dev        libusb-1.0-0-dev        linux-base        libmpc-dev        locales        lsb-release        lzop        ncurses-base        ncurses-term        nfs-kernel-server        ntpdate        openssh-client        p7zip-full        parted        parallel        patchutils        pigz        pixz        pkg-config        psmisc        pv        python2        python3        python3-dev        python3-distutils        python3-pkg-resources        python3-setuptools        qemu-utils        qemu-user-static        rsync        swig        sudo        tzdata        u-boot-tools        udev        unzip        uuid        uuid-dev        uuid-runtime        wget        whiptail        xfsprogs        xxd        zip        zlib1g-dev     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100

[ error ] Docker container build exited with code:  [ 0 ]_

igorpecovnik, 2022-06-26:
Yes, there is something wrong. Need inspection. 

Try removing udev from userpatches/Dockerfile

AndreiCherniaev, 2022-06-26:
> 

> Try removing udev from userpatches/Dockerfile

_[ o.k. ] Using config file [ /home/a/Documents/build/userpatches/config-docker.conf ]
[ .... ] Docker container not found or out of date 
[ .... ] Building a Docker container 
unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/a/Documents/build/userpatches/Dockerfile: no such file or directory

[ error ] Docker container build exited with code:  [ 0 ]_

igorpecovnik, 2022-06-26:
are you running WSL or on some NFS share?

AndreiCherniaev, 2022-06-26:
> are you running WSL or on some NFS share?

I use Ubuntu 22 with btrfs drive. No WSL, no NFS or other share.

igorpecovnik, 2022-06-26:
Starting from clean clone?

@iav Any ideas?

AndreiCherniaev, 2022-06-26:
> Starting from clean clone?

Sorry, my English is not good. "clean clone" - what is it?

igorpecovnik, 2022-06-26:
Create new folder and download build script again:
https://github.com/armbian/build#simply-start-with-the-build-script

AndreiCherniaev, 2022-06-26:
> Create new folder and download build script again: https://github.com/armbian/build#simply-start-with-the-build-script

I remove armbian folder and start since git clone. But there is the same problem.

But I think I know why it can be... In my system I can't do "sudo apt-get update":

a@dom:~$ sudo apt-get update
Get:1 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [109 kB]
Err:1 http://archive.ubuntu.com/ubuntu jammy InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
Get:3 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [99,8 kB]
Err:2 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
Get:4 http://archive.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Err:3 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
Err:4 http://archive.ubuntu.com/ubuntu jammy-security InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
Reading package lists... Done
W: GPG error: http://archive.ubuntu.com/ubuntu jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://archive.ubuntu.com/ubuntu jammy-updates InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy-updates InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://archive.ubuntu.com/ubuntu jammy-backports InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy-backports InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: http://archive.ubuntu.com/ubuntu jammy-security InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 871920D1991BC93C
E: The repository 'http://archive.ubuntu.com/ubuntu jammy-security InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

igorpecovnik, 2022-06-26:
This is outside Armbian. Its broken Ubuntu install or network issues.

iav, 2022-06-26:
@AndreiCherniaev if you see that errors on `sudo apt-get update` — something base is broken on your computer's Operating System. 
Better solution will be system reinstall.

AndreiCherniaev, 2022-06-26:
> @AndreiCherniaev if you see that errors on `sudo apt-get update` — something base is broken on your computer's Operating System. Better solution will be system reinstall.

Yes, I reinstall Ubuntu 22 and now armbian works.

closed 2022-06-26 by AndreiCherniaev