armbian/build #3949: CONFIG_CPU_FREQ_STAT=y missing in rockchip64 kernel?

ThomasKaiser, 2022-06-30:
FYI: http://ix.io/41hL

Memory performance is crap compared to other RK3399 boot BLOBs and `CONFIG_CPU_FREQ_STAT=y` seems to be missing since throttling happens (of course ASUS ARM division not able to provide a proper cooling solution) but sbc-bench can not display statistics.

With ASUS own OS image [throttling happens as well (reported correctly) and memory performance is as crappy as with Armbian](https://github.com/ThomasKaiser/sbc-bench/issues/44).

Tonymac32, 2022-06-30:
Linux 4.19.193 (tinkerboard-2) 	06/30/22 	_aarch64_	(6 CPU)

Where did you get this kernel?

Tonymac32, 2022-07-01:
http://ix.io/41jZ

My old as dirt 5.10 build, I'll need to compile a new one.

ThomasKaiser, 2022-07-01:
> Where did you get this kernel?

No idea, found the result by parsing http://ix.io/user/ – if this is an inofficial build with an unknown kernel branch (and obviously also boot BLOB) then forget about all of this.

Necrosys, 2022-07-01:
Sorry for this confusion. I'm just experimenting with Armbian.

> Linux 4.19.193 (tinkerboard-2) 06/30/22 _aarch64_ (6 CPU)
> 
> Where did you get this kernel?

I got this kernel source from https://github.com/TinkerBoard2/kernel/tree/linux4.19-rk3399-debian10

> if this is an inofficial build with an unknown kernel branch (and obviously also boot BLOB) then forget about all of this.

Yes, this is my unofficial build for Tinkerboard 2. Sorry.

Necrosys, 2022-07-01:
I can confirm that CONFIG_CPU_FREQ_STAT=y is not missing in official Armbian rockchip64 edge/current/legacy kernels.

https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-edge.config#L572
https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-current.config#L561
https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-legacy.config#L571

You can safely close this issue now.

P.S. I'm not related with ASUS in any way.

closed 2022-07-01 by ThomasKaiser

Tonymac32, 2022-07-01:
I will bring this topic to the team to see what can be done about identifying such images.  Within Armbian I set up this board with mainline ATF/U-boot, I apologize for any issues with the WIP kernels, I made the device tree over several beers because I had no interest in the Rockchip kernel. 

The performance difference is not easy to explain, for RAM I know there is more than one supplier, my tested board is Micron.  The Asus blob brings up the RAM at 856 MHz, where the ATF is limited to 800MHz, so that makes the situation more confusing.

Tonymac32, 2022-07-01:
Something with the RK3399 that I haven't bothered understanding is why DDR3 boards can run the RAM at 933 MHz, and LPDDR4 is stuck at 800 (except in special cases where RK gives a vendor 856)