armbian/build #4118: USERPATCHES_PATH on early stages of compile.sh

pavlot, 2022-08-26:
**Describe the bug**
Not sure it is a bug, more likely enhancement.
As I can see from sources `compile.sh` use `"${SRC}"/userpatches/` to determinate path where configs stored.
Why `${USERPATCHES_PATH}` does not used? 
In this case it will be possible to provide folder external to Armbian sources.

igorpecovnik, 2022-08-26:
@psztoch added this feature, perhaps we missed this?
https://github.com/armbian/build/commit/738996ea331bf5e1fd042f8bd728da246a4a561e
There is another idea on the topic but got stalled:
https://github.com/armbian/build/pull/2915

pavlot, 2022-08-26:
I think this part is missing:

```diff
diff --git a/compile.sh b/compile.sh
index 11b6edf..72e07a2 100755
--- a/compile.sh
+++ b/compile.sh
@@ -208,64 +208,78 @@ if [[ "${1}" == docker && -f /etc/debian_version && -z "$(command -v docker)" ]]
 
 fi
 
+# Script parameters handling
+while [[ "${1}" == *=* ]]; do
+
+	parameter=${1%%=*}
+	value=${1##*=}
+	shift
+	display_alert "Command line: setting $parameter to" "${value:-(empty)}" "info"
+	eval "$parameter=\"$value\""
+
+done
+
+[[ -z "${USERPATCHES_PATH}" ]] && USERPATCHES_PATH="${SRC}"/userpatches
+display_alert "USERPATCHES_PATH=${USERPATCHES_PATH}" "${BASH_SOURCE[0]}" "info"
+
 
 # Create userpatches directory if not exists
-mkdir -p "${SRC}"/userpatches
+mkdir -p "${USERPATCHES_PATH}"
 
 
 # Create example configs if none found in userpatches
-if ! ls "${SRC}"/userpatches/{config-default.conf,config-docker.conf,config-vagrant.conf} 1> /dev/null 2>&1; then
+if ! ls "${USERPATCHES_PATH}"/{config-default.conf,config-docker.conf,config-vagrant.conf} 1> /dev/null 2>&1; then
 
 	# Migrate old configs
 	if ls "${SRC}"/*.conf 1> /dev/null 2>&1; then
 		display_alert "Migrate config files to userpatches directory" "all *.conf" "info"
-                cp "${SRC}"/*.conf "${SRC}"/userpatches  || exit 1
+                cp "${SRC}"/*.conf "${USERPATCHES_PATH}"  || exit 1
 		rm "${SRC}"/*.conf
-		[[ ! -L "${SRC}"/userpatches/config-example.conf ]] && ln -fs config-example.conf "${SRC}"/userpatches/config-default.conf || exit 1
+		[[ ! -L "${USERPATCHES_PATH}"/config-example.conf ]] && ln -fs config-example.conf "${USERPATCHES_PATH}"/config-default.conf || exit 1
 	fi
 
 	display_alert "Create example config file using template" "config-default.conf" "info"
 
 	# Create example config
-	if [[ ! -f "${SRC}"/userpatches/config-example.conf ]]; then
-		cp "${SRC}"/config/templates/config-example.conf "${SRC}"/userpatches/config-example.conf || exit 1
+	if [[ ! -f "${USERPATCHES_PATH}"/config-example.conf ]]; then
+		cp "${SRC}"/config/templates/config-example.conf "${USERPATCHES_PATH}"/config-example.conf || exit 1
 	fi
 
 	# Link default config to example config
-	if [[ ! -f "${SRC}"/userpatches/config-default.conf ]]; then
-                ln -fs config-example.conf "${SRC}"/userpatches/config-default.conf || exit 1
+	if [[ ! -f "${USERPATCHES_PATH}"/config-default.conf ]]; then
+                ln -fs config-example.conf "${USERPATCHES_PATH}"/config-default.conf || exit 1
 	fi
 
 	# Create Docker config
-	if [[ ! -f "${SRC}"/userpatches/config-docker.conf ]]; then
-		cp "${SRC}"/config/templates/config-docker.conf "${SRC}"/userpatches/config-docker.conf || exit 1
+	if [[ ! -f "${USERPATCHES_PATH}"/config-docker.conf ]]; then
+		cp "${SRC}"/config/templates/config-docker.conf "${USERPATCHES_PATH}"/config-docker.conf || exit 1
 	fi
 
 	# Create Docker file
-        if [[ ! -f "${SRC}"/userpatches/Dockerfile ]]; then
-		cp "${SRC}"/config/templates/Dockerfile "${SRC}"/userpatches/Dockerfile || exit 1
+        if [[ ! -f "${USERPATCHES_PATH}"/Dockerfile ]]; then
+		cp "${SRC}"/config/templates/Dockerfile "${USERPATCHES_PATH}"/Dockerfile || exit 1
         fi
 
 	# Create Vagrant config
-	if [[ ! -f "${SRC}"/userpatches/config-vagrant.conf ]]; then
-	        cp "${SRC}"/config/templates/config-vagrant.conf "${SRC}"/userpatches/config-vagrant.conf || exit 1
+	if [[ ! -f "${USERPATCHES_PATH}"/config-vagrant.conf ]]; then
+	        cp "${SRC}"/config/templates/config-vagrant.conf "${USERPATCHES_PATH}"/config-vagrant.conf || exit 1
 	fi
 
 	# Create Vagrant file
-	if [[ ! -f "${SRC}"/userpatches/Vagrantfile ]]; then
-		cp "${SRC}"/config/templates/Vagrantfile "${SRC}"/userpatches/Vagrantfile || exit 1
+	if [[ ! -f "${USERPATCHES_PATH}"/Vagrantfile ]]; then
+		cp "${SRC}"/config/templates/Vagrantfile "${USERPATCHES_PATH}"/Vagrantfile || exit 1
 	fi
 
 fi
 
-if [[ -z "${CONFIG}" && -n "$1" && -f "${SRC}/userpatches/config-$1.conf" ]]; then
-	CONFIG="userpatches/config-$1.conf"
+if [[ -z "${CONFIG}" && -n "$1" && -f "${USERPATCHES_PATH}/config-$1.conf" ]]; then
+	CONFIG="${USERPATCHES_PATH}/config-$1.conf"
 	shift
 fi
 
 # usind default if custom not found
-if [[ -z "${CONFIG}" && -f "${SRC}/userpatches/config-default.conf" ]]; then
-	CONFIG="userpatches/config-default.conf"
+if [[ -z "${CONFIG}" && -f "${USERPATCHES_PATH}/config-default.conf" ]]; then
+	CONFIG="${USERPATCHES_PATH}/config-default.conf"
 fi
 
 # source build configuration file
@@ -289,18 +303,5 @@ pushd "${CONFIG_PATH}" > /dev/null || exit
 source "${CONFIG_FILE}"
 popd > /dev/null || exit
 
-[[ -z "${USERPATCHES_PATH}" ]] && USERPATCHES_PATH="${CONFIG_PATH}"
-
-# Script parameters handling
-while [[ "${1}" == *=* ]]; do
-
-	parameter=${1%%=*}
-	value=${1##*=}
-	shift
-	display_alert "Command line: setting $parameter to" "${value:-(empty)}" "info"
-	eval "$parameter=\"$value\""
-
-done
-
 # shellcheck source=lib/main.sh
 source "${SRC}"/lib/main.sh
```

pavlot, 2022-08-26:
I can prepare PR if you think it is make sense :)

igorpecovnik, 2022-09-14:
@The-going What do you think about this?

The-going, 2022-09-17:
> What do you think about this?

This is a proper little improvement. But incomplete.
```
armbian-build> grep -nr '/userpatches/' config/* extensions/* lib/* .github/*
extensions/gen-sample-extension-docs.sh:11:     mkdir -p "${SRC}/userpatches/extensions"
extensions/gen-sample-extension-docs.sh:12:     generate_sample_extension_to_stdout >"${SRC}/userpatches/extensions/sample-extension.sh"
lib/extensions.sh:344:# will look for it in /userpatches/extensions first.
lib/extensions.sh:377:  for extension_base_path in "${SRC}/userpatches/extensions" "${SRC}/extensions"; do
```
P.S.
Since some time I have not been using the `userpatches` folder.
Everything that the script copied there remains in its original form.
Only config-def.conf and config-test.conf are added. Any other possibilities,
hooks (take configuration or patch from `userpatches/.....` if it exists) I find it very inconvenient.

pavlot, 2022-09-18:
Thank you for pointing to missed code :) I will check it.

Regarding convenience - my usecase is to split my project patches/scripts and Armbian. 

I come to this idea after I decide: oh it is time to have fresh clean copy of Armbian to repeat all steps from scratch, I have set of my scripts which will customize distribution. And have deleted Armbian folder. I think you already understand that all my scripts were inside Armbian folder :)

So now idea is that userpatches it is my separate repository (even more than one), so they have their own flow: scrips added branches created etc and it is does not intersect with Armbian code

pavlot, 2022-09-19:
@psztoch 
> Then `[[ -z "${USERPATCHES_PATH}" ]] && USERPATCHES_PATH="${CONFIG_PATH}"` is very important for me.

In my patch it was moved to 
```
[[ -z "${USERPATCHES_PATH}" ]] && USERPATCHES_PATH="${SRC}"/userpatches
```
So I propose to set `USERPATCHES_PATH` as root of all user scripts, would it suits you?

psztoch, 2022-09-19:
In your case it is necessary to set USERPATCHES_PATH.
If you have several repositories with configurations, USERPATCHES_PATH should automatically switch to any path compatible with the given CONFIG file.

I would force it at all: `USERPATCHES_PATH="${CONFIG_PATH}"` - But not everyone may like it.

pavlot, 2022-09-19:
I've change it because of chain of resolving:
https://github.com/armbian/build/blob/master/compile.sh#L267
` if [[ -z "${CONFIG}" && -f "${SRC}/userpatches/config-default.conf" ]];`
And after this `CONFIG_PATH` will be set to directory where config reside 

After I replace it with user patches - CONFIG is taken from User patches directory. This approach still valid for your use case: you can have separate configs (placed to separate USER_PATCHES)

The-going, 2022-10-02:
> Thank you for pointing to missed code :) I will check it.
>
> Regarding convenience - my usecase is to split my project patches/scripts and Armbian.
>
> I come to this idea after I decide: oh it is time [...]

I think I'm beginning to understand that your desires coincide with mine.
I have GIT storage on a dedicated 2 GB large disk. And I want these repositories to remain always clean.
What do you think if we consider the possibility of an alternative use `compile.sh` a script in the root folder `armbian/build`
of the repository and provide a custom command `~/bin/armbian` to execute in any current directory?

In this case, such a concept as a folder

https://github.com/armbian/build/blob/31ac6383e1ac7ebddd0813abc0f1f9632a9c9c40/compile.sh#L212-L218

of custom patches loses its meaning.

The first step: running the command with the configure parameter as:
```> armbian configure```
will create a user configuration file in the current folder (.)
and the necessary directory infrastructure with the contents necessary for compiling and building the image.
```> armbian build``` - Collect all the necessary packages
```> armbian build kernel image``` - Build (or rebuild) kernel packages and build an image.

The-going, 2022-11-28:
For commit https://github.com/armbian/build/commit/27aae916f35b11704230f9f150a6d866963e63ab
```
armbian-build> grep -nr userpatches lib/* extensions/* | wc -l
15
```
For HEAD
```
armbian-build> grep -nr userpatches lib/* extensions/* | wc -l
40
```

This is a clear regression. I'm just stating a fact.
Are we going to fix this?

clarkmcc, 2023-08-29:
Also wondering why `USERPATCHES_PATH` is not being used. `compile.sh` is not using this script.

The-going, 2023-08-30:
@clarkmcc This issue has aged a lot. It should be closed.

closed 2023-08-30 by rpardini

clarkmcc, 2023-08-30:
@The-going okay, but my question still stands. If it's not related to this issue, should I open a new issue?

The-going, 2023-08-30:
> but my question still stands.
The script in the chain calls a lot of files in the `lib`, `extensions` folders
and a simple check in the source tree says that this variable is used 55 times
```bash
armbian/build> grep -nr USERPATCHES_PATH lib/* extensions/* | wc -l
55
```

clarkmcc, 2023-08-30:
So I'm running
```
./compile.sh ... USERPATCHES_PATH=my-directory/ 
```
and can prove from the logs that it's using the default userpatches directory, instead of the one I specified. I can even see the `Applying cmdlin param` confirms that it accepted my specified userpatches directory, even though it's not using the directory for the customize-image.sh script. 

Is this a bug? If so, I'll open a new issue.

The-going, 2023-08-30:
> Is this a bug?

I don't know. To be honest.

> If so, I'll open a new issue.

This is your right. I wonder what the developers who support the build system today will answer you.