armbian/build #4130: configuration.sh: line 35: jq: command not found

tosiara, 2022-08-29:
`jq` was missing on my Jammy machine (default installation)
Usage: https://github.com/armbian/build/blob/master/lib/configuration.sh#L35

Somehow `hostdeps` did not work?

igorpecovnik, 2022-08-29:
Did you try to run it again? Perhaps a network issue?

tosiara, 2022-08-29:
I executed `apt-get install jq` and that solved the issue. But I will try to reproduce this also on a brand new Jammy VM

tosiara, 2022-08-29:
Yes, always reproducible in a brand new Ubuntu 22. Right after you complete the wizard:

[image]

igorpecovnik, 2022-08-29:
> Add jq into prepare_host_basic

IMO this as variable can be overridden so it falls more into "configuration" part then directly to function ...?

hzyitc, 2022-08-29:
> IMO this as variable can be overridden so it falls more into "configuration" part then directly to function ...?

So let's install `jq` in `prepare_host_basic` ?

`prepare_host_basic` will also be run when building with docker which doesn't need `jq` in the host.

igorpecovnik, 2022-09-01:
I this problem will become obsolete after we merge https://github.com/armbian/build/pull/4080

igorpecovnik, 2022-09-04:
jq is not used anymore.

closed 2022-09-04 by igorpecovnik