armbian/build #4145: Problem found using Armbian with ODROID N2+

ronkpunk, 2022-09-02:
**Describe the bug**
I'm trying to use Armbian on my Odroid N2+.
I tried to burn many image on my SD with no luck.
This are images I've tried (also following different thread and suggestions found):
- Armbian 22.08.1 Bullseye
- Armbian 22.05.4 Bullseye
- Armbian 21.08.1 Bullseye
- Armbian 22.05.4 Jammy 
- Armbian 22.05.4 Jammy 
- Armbian 21.05.2 Hirsute Edge

All tests have same results.
Odroid attached with network, keyboard and hdmi cable. I can see petitboot, after that, hdmi lost signal, I can't find odroid on the network and keyboard has no reaction (I've tried also caps lock and lock num excpecting at least a led on the keyboard)

**To Reproduce**
Steps to reproduce the behavior:
- Burn on SD with BalenaEtcher any of the images listed above
- Put SD on detached Odroid N2+
- Attach Odroid N2+ to HDMI, LAN and Keyboard
- Attach Odroid N2+ to power
- 
**Provide logs:**
Drop logs to pastebin service of your choice and supply logs from output/debug
I can't find any log. I'm using Windows so I can't see what SD contains and I can't see anything after petit boot

igorpecovnik, 2022-09-02:
> I'm trying to use Armbian on my Odroid N2+

This is build system bug tracker. 

Usage support is here:
https://forum.armbian.com/forum/112-odroid-n2n2/

I can assure you that Armbian works well on Odroid N2.

> I tried to burn many image on my SD with no luck.

Try changing dip switch to other position.

closed 2022-09-02 by igorpecovnik

ronkpunk, 2022-09-02:
sorry for mistake. I'm pretty new with odroid, DIP switch is the SPI Boot Select Switch? (SD / eMMC)

igorpecovnik, 2022-09-02:
> I'm pretty new

Once again I kindly ask you to join users community https://forum.armbian.com and make friends.