armbian/build #4153: lepotato docker build fails on MX Linux

ewe2, 2022-09-04:
**Describe the bug**

`./compile.sh docker`  run with no changes to the kernel config and just the image with standard console interface..
The last few code updates leading up to and including  the current state of the build code (git 10aeb2f30), the build fails at the point where it chroots the $SDCARD at `/bin/bash`  at stage 2/2, although the latest code has rewritten the informational messages, I believe the error is still the same.  I'm not sure why this is, I do know that MX Linux symlinks /bin to /usr/bin, why docker should have an issue with that confuses me.

**To Reproduce**
Run `./compile.sh docker` It will reliably fail trying to chroot the $SDCARD  at `/bin/bash` in `debootstrap.sh`

**Provide logs:**
[paste of error(]https://pastebin.com/TJu21Hqz)

igorpecovnik, 2022-09-04:
Try un-commenting 

`#DOCKER_FLAGS+=(--privileged)`

in userpatches/config-docker.conf before running the script.

ewe2, 2022-09-04:
No change to the bug unfortunately, it's not a privileges issue. To be extra certain, i ran the build again with sudo, and the results were identical to the above error.

igorpecovnik, 2022-09-04:
Can you provide logs - everything that is in output/debug folder ... and generating from scratch, delete also Docker images if any. 

Currently have no clues - don't know nothing about MX Linux.

ewe2, 2022-09-04:
Apologies., here are most of the logs. compilation.log was too big to pastebin for me, but in all cases, the compilation isn't the issue.

[output.log](https://pastebin.com/U2JFjeRs)
[debootstrap-list.log](https://pastebin.com/GDxu4buB)
[potential-paths.log](https://pastebin.com/9ttnjwT5)
[install.log](https://pastebin.com/uLBDwm3x)
[hostdeps.log](https://pastebin.com/jS4EKvpD)
[compiler.log](https://pastebin.com/cbh4qhtp)
[patching.log](https://pastebin.com/sdUmR1BD)

igorpecovnik, 2022-09-19:
> looks to just be an issue with the loop device.

This is permission issue. Uncomment #DOCKER_FLAGS+=(--privileged) in the userpatche/config-docker.conf

ewe2, 2022-09-19:
That was uncommented already.

iav, 2022-10-05:
Please, compare `Dockerfile` in `userpatches` and `config/templates`.
Copy it from `config/templates` to `userpatches` if file in templates more recent.

And even if you not have to refresh dockerfile — please, remove builder docker image with command `docker rmi armbian:22.11.0-trunk` and let them be recreated.

ewe2, 2022-10-05:
This is that diff, removing fdisk doesn't seem to be a very good idea if its crashing the build at that point. I'll rebuild and let you know the result.
```
--- config/templates/Dockerfile 2022-09-19 15:50:02.126207499 +1000
+++ userpatches/Dockerfile      2022-09-05 19:42:27.586929150 +1000
@@ -49,7 +49,6 @@
        dwarves \
        f2fs-tools \
        fakeroot \
-       fdisk \
        flex \
        gawk \
        gcc-aarch64-linux-gnu \
```

iav, 2022-10-05:
than means `fdisk` was added, not removed. And then change have to be activated (by docker container recreation).
Hope that should solve problem with missed `sfdisk`

ewe2, 2022-10-05:
OK now its building the temporary image fine, but docker isn't finding that loop minor device despite the configuration being set in the Dockerfile, see the debug output.

[logs-06.10.22.zip](https://github.com/armbian/build/files/9717434/logs-06.10.22.zip)

ewe2, 2022-11-03:
Updated the build, did a full jammy image with XFCE desktop. It's clearly failing at finding the /dev/loop0p1 and there is no difference in the docker configs. 

[logs-03.11.22.zip](https://github.com/armbian/build/files/9926965/logs-03.11.22.zip)

rpardini, 2022-11-05:
@ewe2 you might wanna _try_ the `armbian-next` branch. No promises, but a lot was done trying to get loop devices and binfmt working inside Docker. In the end, it still depends on how the hosts' kernel was built.

ewe2, 2022-11-06:
thanks, if the host kernel is a problem, what config do I need to check and reroll myself?

rpardini, 2022-11-06:
`BINFMT_MISC` and a few others. should be standard.

ewe2, 2022-11-08:
Ok, using `armbin-next` has succeeded with a minimal uboot configuration build but building even a command-line system fails with the same error. My docker has the device-cgroup-rule option so that isn't the problem. My kernel is a pretty-much standard debian one (5.17) using debian patches, its got all the BINFMT_MISC and the BLK_DEV stuff mostly in module form, so I cannot see how these are issues. I'm stumped.

ewe2, 2022-11-22:
It is as I suspected: I rebooted my MX system into systemd init and ran a full desktop build which ran to successful completion. Builds depend on systemd, at least for docker. This should be made clear to users. Consider this a wishlist bug perhaps.

closed 2023-11-14 by igorpecovnik