armbian/build #4154: Build tries to download rootfs from wrong URI

ccbur, 2022-09-04:
**Describe the bug**
I'm new to armbian and try to docker build armbian edge for my ordroidm1. The build fails in checksum verification for the rootfs. The downloaded file _bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst_ in cache is filled with html!? I assume the URI for the download [bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst](https://stpete-mirror.armbian.com/cache/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst) is wrong as the corresponding filename on the server seems to be [arm64-bullseye-cli-ae3a25c8-0064.tar.zst](https://stpete-mirror.armbian.com/cache/rootfs/0064/arm64-bullseye-cli-ae3a25c8-0064.tar.zst). This looks like a bug to me!?

**To Reproduce**
Steps to reproduce the behavior:
`./compile.sh docker BOARD="odroidm1" BRANCH=edge RELEASE=bullseye BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_CONFIGURE=no KERNEL_ONLY=no
`
**Provide logs:**
...
[🌿] Building image [ odroidm1 ]
[🌱] Checking for rootfs cache [ edge odroidm1 bullseye no ]
[🌱] Checking local cache [ bullseye-cli-arm64.1f6...3e1.tar.zst ]
[🌿] searching on servers
[🌿] downloading from cache.armbian.com using http(s) network [ bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst ]
md5sum: /root/armbian/cache/rootfs/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst.asc: no properly formatted MD5 checksum lines found
[💥] error: verification failed [  in download_and_verify_internal() at /root/armbian/lib/functions/general/downloads.sh:187 ]
[💥] Build terminating... wait for cleanups...
[💥]     👇👇👇 Showing logfile below 👇👇👇 [ /root/armbian/.tmp/logs-3a7e51e4-7c1a-4029-ab44-108abf573c8f/021.000.get_or_create_rootfs_cache_chroot_sdcard.log ]
[👉]    --> 🦋  467: 1 - 1 - 1: ehBE: group:  [ <get_or_create_rootfs_cache_chroot_sdcard> ] 
[👉]    --> 🐛  467: 1 - 1 - 1: ehBE: debug: ROOTFSCACHE_VERSION not set, getting remotely [ Github API and armbian/mirror  ] 
[👉]    --> 🌱  468: 1 - 1 - 1: ehBE: info: Checking local cache [ bullseye-cli-arm64.1f6...3e1.tar.zst ] 
[👉]    --> 🌿  468: 1 - 1 - 1: ehBE: other: searching on servers [  ] 
 👉]    HTTP/2 302 
 👉]    server: nginx/1.18.0
 👉]    date: Sun, 04 Sep 2022 12:12:51 GMT
 👉]    location: https://stpete-mirror.armbian.com/cache/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst
 👉]    x-geo-distance: 1726033.199933
 👉]    
[👉]    --> 🐸  470: 1 - 1 - 1: ehBE: command: Command debug [ /bin/bash -e -o pipefail -c aria2c --download-result=hide --disable-ipv6=true --summary-interval=0 --console-log-level=error --auto-file-renaming=false --continue=false --allow-overwrite=true --dir=/root/armbian/cache/rootfs https://cache.armbian.com/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst.asc https://fi.mirror.armbian.de/dl// https://es-mirror.bret.dk/armbian/dl// https://mirrors.netix.net/armbian/dl// https://armbian.astra.in.ua// https://mirrors.dotsrc.org/armbian-dl// https://armbian.hosthatch.com/dl// https://stpete-mirror.armbian.com/dl// https://xogium.performanceservers.nl/dl// https://armbian.site-meganet.com/dl// -o bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst.asc ] 
[👉]    09/04 12:12:54 [ERROR] CUID#7 - Download aborted. URI=https://cache.armbian.com/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst.asc 
[👉]    Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://stpete-mirror.armbian.com/cache/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst.asc 
[👉]      -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found 
[👉]    -->--> command run successfully after 1 seconds 
[👉]    --> 🌿  471: 1 - 1 - 1: ehBE: other: downloading from cache.armbian.com using http(s) network [ bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst ] 
[👉]    --> 🐸  472: 1 - 1 - 1: ehBE: command: Command debug [ /bin/bash -e -o pipefail -c aria2c --allow-overwrite=true --download-result=hide --rpc-save-upload-metadata=false --console-log-level=error --dht-file-path=/root/armbian/cache/.aria2/dht.dat --disable-ipv6=true --summary-interval=0 --auto-file-renaming=false --dir=/root/armbian/cache/rootfs https://cache.armbian.com/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst https://fi.mirror.armbian.de/dl// https://es-mirror.bret.dk/armbian/dl// https://mirrors.netix.net/armbian/dl// https://armbian.astra.in.ua// https://mirrors.dotsrc.org/armbian-dl// https://armbian.hosthatch.com/dl// https://stpete-mirror.armbian.com/dl// https://xogium.performanceservers.nl/dl// https://armbian.site-meganet.com/dl// -o bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst ] 
[👉]    09/04 12:12:56 [ERROR] CUID#7 - Download aborted. URI=https://cache.armbian.com/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst 
[👉]    Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://fi.mirror.armbian.de/cache/rootfs/0064/bullseye-cli-arm64.1f64541fbb6f34ea104a483b58f233e1.tar.zst 
[👉]      -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found 
[👉]    -->--> command run successfully after 1 seconds 
[👉]    --> 💥  473: 1 - 1 - 1: ehBE: err: error: verification failed [  in download_and_verify_internal() at /root/armbian/lib/functions/general/downloads.sh:187 ] 
[👉]    --> 💥  473: 1 - 1 - 1: ehBE: err: Build terminating... wait for cleanups... [  ] 
[👉]    --> 🧽  473: 1 - 1 - 1: ehBE: trap: main_trap_handler [ EXIT and 43 trap_manager_error_handled: ] 
[👉]    [💥]     👆👆👆 Showing logfile above 👆👆👆 [ /root/armbian/.tmp/logs-3a7e51e4-7c1a-4029-ab44-108abf573c8f/021.000.get_or_create_rootfs_cache_chroot_sdcard.log ]
[💥] Exit with error detected [ 43 at /root/armbian/lib/functions/logging/traps.sh:1 -
             exit_with_error() --> ./lib/functions/logging/traps.sh:1
download_and_verify_internal() --> ./lib/functions/general/downloads.sh:187
         download_and_verify() --> ./lib/functions/general/downloads.sh:44
get_or_create_rootfs_cache_chroot_sdcard() --> ./lib/functions/rootfs/create-cache.sh:49
             do_with_logging() --> ./lib/functions/logging/logging.sh:104
      build_rootfs_and_image() --> ./lib/functions/main/rootfs-image.sh:51
   main_default_build_single() --> ./lib/functions/main/default-build.sh:169
              cli_entrypoint() --> ./lib/functions/cli/cli-entrypoint.sh:141
                        main() --> ./compile.sh:52
 ]
[🌿] Unmounting recursively [ /root/armbian/.tmp/rootfs-3a7e51e4-7c1a-4029-ab44-108abf573c8f/ ]
[🌱] Unmounted OK after 1 attempt(s) [ /root/armbian/.tmp/rootfs-3a7e51e4-7c1a-4029-ab44-108abf573c8f/ ]
[🌿] Unmounting recursively [ /root/armbian/.tmp/mount-3a7e51e4-7c1a-4029-ab44-108abf573c8f/ ]
[🌱] Unmounted OK after 1 attempt(s) [ /root/armbian/.tmp/mount-3a7e51e4-7c1a-4029-ab44-108abf573c8f/ ]
[🌱] Cleaning up [ extension manager ]
[🌿] Built ANSI log file [ /root/armbian/output/logs/armbian-logs-ansi-3a7e51e4-7c1a-4029-ab44-108abf573c8f.txt.log ]
[💥]     👇👇👇 Showing logfile below 👇👇👇 [ /home/christoph/workdir_local/armbian/build/.tmp/logs-3bc19d8d-c4f8-4173-b501-a303f6cb98fc/002.000.prepare_host_basic.log ]
[👉]    --> 🦋    0: 566616 - 566616 - 566616: ehBE: group:  [ <prepare_host_basic> ]
[👉]    --> 🦋    0: 566616 - 566616 - 566616: ehBE: group:  [ </prepare_host_basic> in 0s ]
[👉]    --> 🌱    0: 566616 - 566616 - 566616: ehBE: info: Using config file [ /home/christoph/workdir_local/armbian/build/userpatches/config-docker.conf ]
[👉]    --> 🌿    0: 566616 - 566616 - 566616: ehBE: other: Using existed a armbian Docker container [  ]
[👉]    --> 🌱    1: 566616 - 566616 - 566616: ehBE: info: Running the container [  ]
[👉]    --> 🧽  476: 566616 - 566616 - 566616: ehBE: trap: main_trap_handler [ ERR and 43 trap_manager_error_handled: ]
[💥]     👆👆👆 Showing logfile above 👆👆👆 [ /home/christoph/workdir_local/armbian/build/.tmp/logs-3bc19d8d-c4f8-4173-b501-a303f6cb98fc/002.000.prepare_host_basic.log ]
[💥] Error occurred [ code 43 at /home/christoph/workdir_local/armbian/build/userpatches/config-docker.conf:118
                      source() --> ./userpatches/config-docker.conf:118
              cli_entrypoint() --> ./lib/functions/cli/cli-entrypoint.sh:108
                        main() --> ./compile.sh:52
 ]
/home/christoph/workdir_local/armbian/build/lib/functions/logging/logging.sh: line 409: zstdmt: command not found
[🌿] Built ANSI log file [ /home/christoph/workdir_local/armbian/build/output/logs/armbian-logs-ansi-3bc19d8d-c4f8-4173-b501-a303f6cb98fc.txt.log ]
[💥] Error occurred [ code 43 at /home/christoph/workdir_local/armbian/build/lib/functions/cli/cli-entrypoint.sh:20
              cli_entrypoint() --> ./lib/functions/cli/cli-entrypoint.sh:20
                        main() --> ./compile.sh:52
 ]

igorpecovnik, 2022-09-04:
Please use build script from a stable branch.

closed 2022-09-04 by igorpecovnik

ccbur, 2022-09-04:
You're right. Sorry, I forgot to mention that I'm on armbian-next. But as my goal was a debian / linux 5.18.19 based build for Odroid M1, I'll have no luck on the stable branches for now.
Nevertheless I conviced the build framework to debootstrap a bullseye rootfs for me (ROOT_FS_CREATE_ONLY=yes) after removing **nala** from _config/cli/bullseye/main/packages_. I needed to change back to ROOT_FS_CREATE_ONLY=no in another build because of some sdcard unmount thingy. No idea what this was. At the end I got some errors from a hack in _odroidm1.wip_. So yeah, still some WIP ;-)

Anyway... if someone is interested: I now have a nvme bootable bullseye image based on armbian/tobetter kernel. Thanks! :+1:

igorpecovnik, 2022-09-04:
> Nevertheless I conviced the build framework to debootstrap

Good! I always encurage people to try, before asking for assistance. We can't provide real-time support anyway. Not even for something that we are working daily with - master branch.

> Sorry, I forgot to mention

Its evident ;)

We have a rule to not accept issues on unfinished work. Without that rule, most of us would already be hospitalised in mental institutions :)

We are planning to merge this, but just preparing documentation, getting familiar with changes ... is weeks of work. And we haven't really started yet.

> I'll have no luck on the stable branches for now.

Perhaps try to cherry pick this board into master branch?

ccbur, 2022-09-05:
> We have a rule to not accept issues on unfinished work. Without that rule, most of us would already be hospitalised in mental institutions :)

That makes a lot of sense! 

> We are planning to merge this, but just preparing documentation, getting familiar with changes ... is weeks of work. And we haven't really started yet.

I digged into the changelog of armbian-next and yes, looks like a lot of work to be done ;-)

> Perhaps try to cherry pick this board into master branch?

Thanks for the suggestion. Will try!