armbian/build #4178: Kernel can't be build error

easy-and-simple, 2022-09-13:
I tried to build image using armbian build framework, but I got following error. What is the problem? Why kernel can;t be build?

Cyclomatic Complexity 4 drivers/gpu/arm/bifrost_for_linux/ipa/mali_kbase_ipa.c:get_static_power_locked
Cyclomatic Complexity 1 drivers/gpu/arm/bifrost_for_linux/ipa/mali_kbase_ipa.c:kbase_get_static_power
Cyclomatic Complexity 5 drivers/gpu/arm/bifrost_for_linux/ipa/mali_kbase_ipa.c:kbase_get_real_power
cc1: some warnings being treated as errors
scripts/extract-cert.c: In function ‘display_openssl_errors’:
scripts/extract-cert.c:46:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   46 |         while ((e = ERR_get_error_line(&file, &line))) {
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/extract-cert.c: In function ‘drain_openssl_errors’:
scripts/extract-cert.c:59:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   59 |         while (ERR_get_error_line(&file, &line)) {}
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
make[4]: *** [scripts/Makefile.build:334: drivers/gpu/arm/bifrost_for_linux/ipa/mali_kbase_ipa.o] Error 1
make[3]: *** [scripts/Makefile.build:637: drivers/gpu/arm/bifrost_for_linux] Error 2
make[2]: *** [scripts/Makefile.build:637: drivers/gpu/arm] Error 2
make[1]: *** [scripts/Makefile.build:637: drivers/gpu] Error 2
make: *** [Makefile:1195: drivers] Error 2
[ error ] ERROR in function compile_kernel [ main.sh:588 -> main.sh:489 -> compilation.sh:509 -> general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

igorpecovnik, 2022-09-13:
which kernel, which build script version, did you make any changes?

easy-and-simple, 2022-09-17:
Hello guys,
Thank you for your replays.
It looks like error is caused by Cyclomatic Complexity compiler option I turned on, because I was able to compile few images with this option turned off.
I don't know what this option means, but turned if on because had no idea how to set it.
It may be something else because I was changed a lot of kernel options, but because with it turned of I had not problems to build images, I suppose build error was cause by it.

hzyitc, 2022-09-19:
It seem like that you have enabled `Compile the kernel with warnings as errors`(`-Werror`). So the `OpenSSL 3.0`'s warnings ares treated as errors.

closed 2022-11-25 by igorpecovnik