armbian/build #4223: Failed to build preempt_rt kernel

roshanprince402, 2022-09-23:
**When i'm building preempt rt kernel for rockpi-4b then every time getting this error**

scripts/sign-file.c:89:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   89 |         while ((e = ERR_get_error_line(&file, &line))) {
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/sign-file.c: In function ‘drain_openssl_errors’:
scripts/sign-file.c:102:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  102 |         while (ERR_get_error_line(&file, &line)) {}
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/extract-cert.c: In function ‘display_openssl_errors’:
scripts/extract-cert.c:46:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   46 |         while ((e = ERR_get_error_line(&file, &line))) {
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/extract-cert.c: In function ‘drain_openssl_errors’:
scripts/extract-cert.c:59:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   59 |         while (ERR_get_error_line(&file, &line)) {}
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:22.11-27.6: Warning (spi_bus_reg): /fragment@1/__overlay__/spidev: SPI bus unit address format                              error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:36.11-41.6: Warning (spi_bus_reg): /fragment@2/__overlay__/spidev: SPI bus unit address format                              error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:50.11-55.6: Warning (spi_bus_reg): /fragment@3/__overlay__/spidev: SPI bus unit address format                              error, expected "0"
arch/arm64/boot/dts/rockchip/overlay/rockchip-spi-spidev.dts:64.11-69.6: Warning (spi_bus_reg): /fragment@4/__overlay__/spidev: SPI bus unit address format                              error, expected "0"
drivers/video/fbdev/core/fbcon.c:1919:25: error: implicit declaration of function ‘fb_scrollmode’ [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:289: drivers/video/fbdev/core/fbcon.o] Error 1
make[3]: *** [scripts/Makefile.build:552: drivers/video/fbdev/core] Error 2
make[2]: *** [scripts/Makefile.build:552: drivers/video/fbdev] Error 2
make[1]: *** [scripts/Makefile.build:552: drivers/video] Error 2
make: *** [Makefile:1898: drivers] Error 2
fs/aufs/i_op.c:639:52: error: ‘struct rw_semaphore’ has no member named ‘owner’
make[2]: *** [scripts/Makefile.build:289: fs/aufs/i_op.o] Error 1
make[1]: *** [scripts/Makefile.build:552: fs/aufs] Error 2
make: *** [Makefile:1898: fs] Error 2
[ error ] ERROR in function compile_kernel [ main.sh:588 -> main.sh:489 -> compilation.sh:508 -> general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

igorpecovnik, 2022-09-23:
Check build logs to see where is the actual problem.

The-going, 2022-09-23:
@roshanprince402 What do you want from this patched kernel?
If it starts working, what do you want to implement on this platform?

roshanprince402, 2022-09-24:
> 

Working on Low latency audio system, currently with low latency kernel I'm getting higher jitter as compare to Realtime kernel.
Issue is when I'm installing OS into onboard emmc(Board - RockPi 4B plus) using nand-sata-install tool then OS is booting in read only file system, with SD Card everything works fine. also without preempt_rt kernel everything works fine.

The-going, 2022-09-24:
> drivers/video/fbdev/core/fbcon.c:1919:25: error: implicit declaration of function ‘fb_scrollmode’ [-Werror=implicit-function-declaration]
> cc1: some warnings being treated as errors

`-Werror` This compiler option can be disabled, but it's better to fix it.

> make[4]: *** [scripts/Makefile.build:289: drivers/video/fbdev/core/fbcon.o] Error 1
> make[3]: *** [scripts/Makefile.build:552: drivers/video/fbdev/core] Error 2
> make[2]: *** [...]

If you took the patch from this source,
https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.19/
look carefully at which files the changes are applied to.

You will have to work hard so that the pre-emption technology can start working on this platform.
At a minimum, `gpio` and `pinctrl` subsystems require attention.

igorpecovnik, 2022-11-25:
If fbdev subsystem is having troubles, disabling boot logo might help SKIP_BOOTSPLASH=yes ... anyway closing as its not Armbian issue.

closed 2022-11-25 by igorpecovnik