armbian/build #4306: CONFIG_RANDOMIZE_BASE support?

Manouchehri, 2022-10-20:
I noticed that we aren't passing in a `kaslr-seed` in any of the DTBs, which breaks `CONFIG_RANDOMIZE_BASE` (kASLR).

https://github.com/torvalds/linux/blob/e022620b5d056e822e42eb9bc0f24fcb97389d86/arch/arm64/kernel/pi/kaslr_early.c#L79

```
[    0.052888] KASLR disabled due to lack of seed
```

```
root@nanopi-r4s:~# ls /proc/device-tree/chosen/
bootargs	  linux,initrd-start  stdout-path
linux,initrd-end  name		      u-boot,version
```

Not entirely sure what we need to add for `kaslr-seed` to show up.

EvilOlaf, 2022-10-20:
This sounds like an issue with *running* Armbian rather than *building* Armbian.
Take note that the issues on Github are only for issues with the build framework and the building process. Feel free to try forums for solution.

Manouchehri, 2022-10-20:
I mean.. we're not _building_ with `/chosen/kaslr-seed`, so _running_ Armbian with kASLR doesn't work. This isn't something that can be configured at runtime, only during the u-boot build as far as I know.

Manouchehri, 2022-10-21:
We need to add `kaslrseed` to the various `*.cmd` files.

https://github.com/armbian/build/tree/master/config/bootscripts

It'll only work if an rng is present I think, but doesn't seem to do anything bad if it's not there.

PR #4352 (2022-10-26): Add kaslrseed to Rockchip's u-boot
# Description

Partly resolves #4306.

# How Has This Been Tested?

If the `kaslrseed` command hasn't been compiled in to u-boot, it gracefully skips generating the kASLR 

- [X] Booted on a NanoPi R4S

# Checklist:

- [X] My code follows the style guidelines of this project
- [X] I have performed a self-review of my own code
- [ ] I have commented my code, particularly in hard-to-understand areas
- [ ] I have made corresponding changes to the documentation
- [X] My changes generate no new warnings
- [X] Any dependent changes have been merged and published in [... cut]
closed 2022-10-26 by igorpecovnik