armbian/build #4315: OFFLINE_WORK is fragile and untested

Manouchehri, 2022-10-22:
e.g. on a NanoPi R4S image build, we attempt to install ethtool off the internet. That obviously doesn't work.

While I can fix that one specific issue with ethtool's package, I think fundamentally we would have to have some CI/CD testing of `OFFLINE_WORK`.

Open question: should we remove this flag completely for now, add a warning, or wait until it's fixed?

eerison, 2022-11-27:
Maybe we could enable the discussion tab for topics related with questions?

The-going, 2022-11-27:
> Open question: should we remove this flag completely for now, add a warning, or wait until it's fixed?

The correct answer is to try to fix it and make a pull request.

This key is designed to work for packages build. When you have updated all local sources and are building only one or several packages, it is obvious that this key will be useful to you. I don't recommend using it to build an image.