armbian/build #4316: Add READONLY_MODE

Manouchehri, 2022-10-22:
In some situations, it's quite nice to have a SBC with a stateless image.

- [ ] Automate https://www.dzombak.com/blog/2021/11/Reducing-SD-Card-Wear-on-a-Raspberry-Pi-or-Armbian-Device.html
- [ ] Add https://github.com/BertoldVdb/sdtool to build
- [ ] Automate enabling temporary write protection with sdtool

leggewie, 2025-08-19:
removing "can be closed" label.

I quite like the idea

tabrisnet, 2025-09-07:
hmm. BliKVM H616 appears to be an Armbian fork, and they have a mechanism for mounting `/` as `ro` and back to `rw`.
May be worth investigating if its worth using.

igorpecovnik, 2025-09-07:
R/O mode is supported with `overlayrootfs`. https://docs.armbian.com/User-Guide_Armbian-Config/System/#read-only-fs