armbian/build #4322: Problem found in ...

toneywoo, 2022-10-23:
**Describe the bug**
build error
use 
sudo ./compile.sh  BOARD=odroidc4 BRANCH=current RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_ONLY=no KERNEL_CONFIGURE=prebuilt DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base DESKTOP_APPGROUPS_SELECTED="desktop_tools remote_desktop" COMPRESS_OUTPUTIMAGE=sha,gpg,gz EXTERNAL=yes  USE_GITHUB_UBOOT_MIRROR=yes

logs:
E: Unable to locate package system-monitoring-center
[ error ] ERROR in function create_rootfs_cache [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:222 -> functions/logging/traps.sh:0 ]
[ error ] Installation of Armbian desktop packages for current odroidc4 jammy desktop_tools remote_desktop xfce no failed 
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:222 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 ]
[ o.k. ] Unmounting [ /home/runner/work/orangepi-action-build/orangepi-action-build/build/.tmp/rootfs-604e35a7-b9e4-48f4-94e0-39b15728d9be/ ]
[ error ] ERROR in function unmount_on_exit [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:222 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:26 -> functions/logging/traps.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
Error: Process completed with exit code 255.

igorpecovnik, 2022-10-23:
This must be some type of network issue (without seeing the logs). Try again, try with additional parameter NO_APT_CACHER="yes"

The-going, 2022-10-25:
> logs:
> E: Unable to locate package system-monitoring-center
> [ error ] ERROR in function create_rootfs_cache [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> [...]

The same error message is printed three times. But there is no message about what caused this error.
@rpardini  Why?

There may be two reasons for this error.
1) One or more packages cannot be installed.
2) Not enough space.

igorpecovnik, 2022-10-25:
This error is almost certainly related to repository index out of sync.

The-going, 2022-10-25:
> This error is almost certainly related to repository index out of sync.

Maybe yes! But I came across another case.
When we give a task for apt to install a large number of packages,
its dependency resolution logic may not cope.
This can be seen if you set the be verbose flag.

In any case, we don't see the reason in the logs.

igorpecovnik, 2022-10-25:
> In any case, we don't see the reason in the logs.

That is correct. Logging is not in the best shape yet.

The-going, 2022-10-25:
The algorithm can be as follows:
```mermaid
flowchart TD
	A(Create a temporary file) --> B[Run apt in verbose mode and\nsend the output to a temporary file]
	B --> C{"$? > 0"\nCheck the completion code or/and\nfor errors in the temporary file}
	C -->|True| D[Print the temporary file as error.log]
	C -->|False| E[Delete the temporary file and continue]
	D -->F[Terminate by mistake and print the trace once]
```

rpardini, 2022-10-25:
In `armbian-next`, this is handled in two ways:

- Logging in general. `set -e` is in effect. When error occurs in command execution, the whole log section is shown on the screen. Which would indicate this problem clearly.
- General "apt" retry and `--download-only`. `a-n` has special handler functions for use with apt, and all installs are split between "download only" which is retried 5 times, and "actually install pre-downloaded packages".

rpardini, 2022-10-25:
With `set -e` in effect, `$?` is a meaningless construct; all errors abort execution, unless you're in a shortcircuit or in an `if` clause.

The-going, 2022-10-25:
> When error occurs in command execution, the whole log section is shown on the screen.

A good option would be to send the output of the command to a separate error.log file,
and display a short message on the screen.

> and all installs are split between "download only" which is retried 5 times, and "actually install pre-downloaded packages".

When `apt` coped with the resolution of dependencies, this logic will help, you're right.
And when he gets confused and walks in circles? After a few cycles, it will fail with an error.

toneywoo, 2022-10-26:
> This must be some type of network issue (without seeing the logs). Try again, try with additional parameter NO_APT_CACHER="yes"

build ok

closed 2022-11-24 by igorpecovnik