armbian/build #4328: Cannot build armbian using rootless podman

forty, 2022-10-24:
**Describe the bug**
It is not possible to build Armbian using rootless podman. It's not supported so it makes sense.
I'm mostly opening this bug to document what works and what doesn't work, in case someone try to do the same.

**To Reproduce**

Prerequisites:
- Make sure you have a symlink `docker` pointing to `podman` in your PATH (or similar stub of the docker command, available in the ` podman-docker` packages in several distributions)
- Make sure basic dependencies are already available on the host (https://github.com/armbian/build/blob/037248761dc78601015bf0f4a4e779de53384d7d/lib/functions/host/basic-deps.sh#L10). Else the build script will try to try to install them with root
- Comment this line https://github.com/armbian/build/blob/master/config/templates/config-docker.conf#L48 (parameter is unsupported by podman)
- Comment the root check https://github.com/armbian/build/blob/master/lib/functions/cli/cli-entrypoint.sh#L13-L21

We will build using the following command as a test:
```
./compile.sh docker BOARD=odroidc2 KERNEL_ONLY=no KERNEL_CONFIGURE=no BUILD_MINIMAL=yes BUILD_DESKTOP=no BRANCH=current RELEASE=bullseye
```

Problem: binfmt_misc is not namespaced and requires host root. This causes "Exec format error" as the emulator cannot be registred in binfmt_misc.
Workaround: using https://github.com/multiarch/qemu-user-static (requires root on the host... but once it's run, you can continue trying to build rootless)
```
sudo podman run --rm --privileged docker.io/multiarch/qemu-user-static --reset -p yes
```

Then everything builds fine, up until sdcard creation:

Problem: loop devices are not namespaced and requires host root. This causes "losetup: cannot find an unused loop device: Permission denied"
Solution: switch image creation to root less tools such as https://github.com/pengutronix/genimage (Like buildroot is doing for example)

This is far more involved I was going to be with this little experiment so I stopped here ^^ There are probably other issues after this one.

forty, 2022-10-24:
Building rootless or with podman is not supported.

closed 2022-10-24 by forty

rpardini, 2022-10-24:
Hey @forty  take a look at the [`armbian-next`](https://github.com/armbian/build/commits/armbian-next) branch where I've started a rewrite of all `cli` and Docker stuff, with more control of the Dockerfile generation etc. 
I've already handled special cases for Docker Desktop on Mac, both Intel and M1, also for Rancher Desktop.
D.Desktop on Windows WSL2 works but needs a lot of special handling.
I think if we twist it a bit more we can add
- nerdctl + buildkit
- podman + buildah

rpardini, 2022-10-24:
But yeah -- maybe `/dev/loop` stuff won't ever work rootless.

igorpecovnik, 2022-10-24:
> But yeah -- maybe /dev/loop stuff won't ever work rootless.

Yes. This remain open wound ... But in our defence we can say - we are trying to address it for some time. Perhaps proposed idea might work for us too?

rpardini, 2022-10-24:
Some kind of userspace image building might or not work for all our purposes. `genimage` for example uses `genext2fs` underneath. "emulating" ext4 (for instance) code in userspace. 

It's worth investigating, but I imagine a huge effort.

I'd love to build without root, but not that urgent; people can get VMs / Docker etc.

forty, 2022-10-24:
Yes, Buildroot builds everything using userspace tools (they don't even need qemu). It's beautiful and fascinating, but it does make things much more complex (especially the no emulator part, which allows them to avoid messing with binfmt_misc - but is probably not doable when dealing with deboostrap for example).

People can get VMs, but they'd be doing qemu in a VM, which I feel is too many layers to be efficient, but I don't know much and maybe VMs and emulations are much more efficient than what I imagine these days.

@rpardini I have tried `armbian-next`, no more luck with _rootless_ podman (unsurprisingly, stuck with binfmt_misc mount), but I'm able to generate an image with _rootfull_ podman. I had to comment a bunch of docker options, have a few errors in the logs (git repository cross file system related), but I do end up with an image - which is better than `master` where I was stuck with losetup errors even with root I think. I haven't tried the image yet though, I'll let you know when I do, but at least it's promising :)