armbian/build #4360: Problem found in ...

codingCoffee, 2022-10-28:
**Describe the bug**
Unable to build Armbian for `XFCE` desktop environment for `BananaPi M2 Zero`. It is working for the `Mate` desktop environment (saying in case it helps debug the issue). In the below command, replacing `xfce` with `mate` works.

**To Reproduce**
```sh
./compile.sh docker BOARD=bananapim2zero BRANCH=current RELEASE=buster BUILD_MINIMAL=no BUILD_DESKTOP=yes KERNEL_ONLY=no KERNEL_CONFIGURE=no DESKTOP_ENVIRONMENT=xfce DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base DESKTOP_APPGROUPS_SELECTED="3dsupport internet remote_desktop" COMPRESS_OUTPUTIMAGE=sha,gpg,img EXPERT=yes
```

**Provide logs:**
- [debootstrap-list.log](http://0x0.st/o3di.log)
- [output.log](http://0x0.st/o3d-.log)
- [hostdeps.log](http://0x0.st/o3do.log)
- [compilation.log](http://0x0.st/o3dH.log)

TL;DR
```
...
aspell-autobuildhash: processing: en [en_US-w_accents-only].
aspell-autobuildhash: processing: en [en_US-wo_accents-only].
Processing triggers for libc-bin (2.28-10+deb10u2) ...
Processing triggers for systemd (241-7~deb10u8) ...
Processing triggers for dbus (1.12.24-0+deb10u1) ...
[ o.k. ] Checking MD5 sum of installed packages [ debsums ]
[ error ] ERROR in function create_rootfs_cache [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:229 -> functions/logging/traps.sh:0 ]
[ error ] MD5 sums check of installed packages failed
[ o.k. ] Process terminated
[ error ] unmount_on_exit() called! [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:229 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 ]
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-f939d9ac-ede9-4476-ae1a-1c144ea6db73/ ]
[ error ] ERROR in function unmount_on_exit [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:229 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:26 -> functions/logging/traps.sh:0 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

Please tell me if I need to share any other logs. I don't mind debugging the issue, if someone could point me in the right direction to get started where the error lies or how to run things manually. The error seems to be in `create_rootfs_cache`, but not sure which package is failing `MD5 sums check` or how to find it.

igorpecovnik, 2022-10-28:
Try something else then Buster as we don't maintain it anymore.

codingCoffee, 2022-11-01:
Not opening the issue, since Buster is not supported.

Just logging additional stuff here, since there is an additional logline after the fix 4363.

```
...
aspell-autobuildhash: processing: en [en_US-w_accents-only].
aspell-autobuildhash: processing: en [en_US-wo_accents-only].
Processing triggers for libc-bin (2.28-10+deb10u2) ...
Processing triggers for systemd (241-7~deb10u8) ...
Processing triggers for dbus (1.12.24-0+deb10u1) ...
[ o.k. ] Checking MD5 sum of installed packages [ debsums ]
debsums: changed file /usr/share/icons/Numix-Circle/icon-theme.cache (from numix-icon-theme-circle package)
[ error ] ERROR in function create_rootfs_cache [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:66 -> functions/rootfs/create-cache.sh:228 -> functions/logging/traps.sh:0 ]
[ error ] MD5 sums check of installed packages failed
[ o.k. ] Process terminated
[ error ] unmount_on_exit() called! [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:66 -> functions/rootfs/create-cache.sh:228 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 ]
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-23c83cbd-2009-4b56-adbc-2a00aea55611/ ]
[ error ] ERROR in function unmount_on_exit [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:66 -> functions/rootfs/create-cache.sh:228 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:26 -> functions/logging/traps.sh:0 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```

igorpecovnik, 2022-11-02:
Was this done after "debsums fix" was merged or before? IIRC this icon cache is recreated once package is installed ... strange thing is that its a part of the md5 sum ... perhaps this is a package problem?

codingCoffee, 2022-11-02:
This was done **after** the "debsums fix" was merged. 

I don't believe it has impacted this issue. I see `-s` flag was added to the `debsums` command in the #4363 PR, which is causing the exact error to be printed out. 

```
debsums: changed file /usr/share/icons/Numix-Circle/icon-theme.cache (from numix-icon-theme-circle package)
```

hzyitc, 2022-11-03:
Clean the caches of `apt-cacher-ng` and try again? Or try to use another ubuntu mirror?

hzyitc, 2022-11-03:
If you readlly need this. Comment this line to ignore it.

https://github.com/armbian/build/blob/d101f2a69c63b5b487585deae5e951214ae05948/lib/functions/rootfs/create-cache.sh#L228