armbian/build #4370: [orangepi4-lts] rockchip64_common.inc: line 191: qemu-x86_64: command not found

Turbine1991, 2022-10-30:
**Describe the bug**
Both the 5.15 and 6 kernel output this error. I could compile fine a couple of weeks ago. I've recloned the repo just to ensure it was clean. qemu-system is installed.

**To Reproduce**
Run ./compile.sh
or
./compile.sh BOARD=orangepi4-lts BRANCH=edge KERNEL_ONLY=yes RELEASE=kinetic KERNEL_CONFIGURE=yes KERNEL_KEEP_CONFIG=yes USE_CCACHE=yes 

**Provide logs:**
```
Image Type:   Rockchip RK33 (SD/MMC) boot image
Init Data Size: 131072 bytes
qemu-x86_64: --pack: invalid option
qemu-x86_64: --replace: invalid option
[ error ] ERROR in function compile_uboot [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:55 -> functions/compilation/uboot.sh:145 -> functions/logging/traps.sh:0 ]
[ error ] U-boot file not found [ uboot.img ]
[ o.k. ] Process terminated
```

igorpecovnik, 2022-10-30:
> qemu-system is installed.

Is this perhaps aarch64 build host? Then yes, this is expected. u-boot build is not 100% covered due to tools used in the process. Otherwise - in case this is not aarch64 host, we need to see logs.

c-ridgway, 2022-10-30:
> > qemu-system is installed.
> 
> Is this perhaps aarch64 build host? u-boot build is not 100% covered due to tools used in the process. Otherwise - in case this is not aarch64 host, we need to see logs.

As I'm compiling on the pi itself, it would be aarch64. Where abouts would the log be for the build system? It's weird that it worked just a few weeks ago.

igorpecovnik, 2022-10-30:
> It's weird that it worked just a few weeks ago.

This means we could mess it up somehow ... 

> Where abouts would the log be for the build system?

output/debug