armbian/build #4385: Doesn't work anymore

easy-and-simple, 2022-11-05:
I tried different build configurations and none of them works
Compiling return errors on bifrost video driver building or in other place
Using precompiled image gives me the following example errors:

[ o.k. ] Checking cache [ arm64-jammy-gnome-8c83dc03-0079.tar.zst ]
[ .... ] Downloading from servers 

11/05 15:22:59 [ERROR] CUID#6 - Download aborted. URI=https://github.com/armbian/cache/releases/download/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://github.com/armbian/cache/releases/download/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/05 15:22:59 [ERROR] CUID#7 - Download aborted. URI=https://fi.mirror.armbian.de/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://fi.mirror.armbian.de/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/05 15:22:59 [ERROR] CUID#8 - Download aborted. URI=https://imola.armbian.com/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://imola.armbian.com/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/05 15:22:59 [ERROR] CUID#9 - Download aborted. URI=https://stpete-mirror.armbian.com/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://stpete-mirror.armbian.com/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#7da51f 0B/0B CN:1 DL:0B]                                                     
11/05 15:23:00 [ERROR] CUID#10 - Download aborted. URI=https://mirror-eu-de1.armbian.airframes.io/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://mirror-eu-de1.armbian.airframes.io/cache//rootfs/0079/arm64-jammy-gnome-8c83dc03-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

aria2 will resume download if the transfer is restarted.
If there are any errors, then see the log file. See '-l' option in help/man page for details.
[ o.k. ] Creating new rootfs cache for [ jammy ]

####################################################################################
it continues and after a while
####################################################################################

[ o.k. ] Applying common tweaks 
[ .... ] Cleaning [ package lists ]
[ .... ] Updating [ package lists ]
[ .... ] Temporarily disabling [ initramfs-tools hook for kernel ]
[ .... ] Installing PACKAGE_LIST_FAMILY packages [ ethtool ]
[ .... ] Installing from repository [ linux-u-boot-rock-3a-legacy ]
[ .... ] Installing from repository [ linux-image-legacy-rk35xx ]
[ .... ] Installing from repository [ linux-dtb-legacy-rk35xx ]
[ .... ] Installing from repository [ armbian-bsp-cli-rock-3a ]
[ .... ] Installing from repository [ armbian-jammy-desktop-gnome ]
[ .... ] Installing from repository [ armbian-firmware ]
[ error ] ERROR in function install_deb_chroot [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:76 -> functions/rootfs/distro-agnostic.sh:352 -> functions/rootfs/apt-install.sh:21 -> functions/logging/traps.sh:0 ]
[ error ] Installation of armbian-firmware failed [ rock-3a jammy yes rk35xx ]
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:76 -> functions/rootfs/distro-agnostic.sh:352 -> functions/rootfs/apt-install.sh:21 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 ]
[ o.k. ] Unmounting [ /home/user1/armbian/build/.tmp/rootfs-2fb276d7-f482-4f05-9642-2450d8a2050c/ ]
[ error ] ERROR in function unmount_on_exit [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:76 -> functions/rootfs/distro-agnostic.sh:352 -> functions/rootfs/apt-install.sh:21 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:26 -> functions/logging/traps.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated

igorpecovnik, 2022-11-05:
Can you please provide build logs (output/debug) and what (if) you changed to the source? Is this master branch?

easy-and-simple, 2022-11-08:
Hi
Here is a link to last logs: https://github.com/easy-and-simple/dbglogs/blob/main/debug.zip
Files that armbian compile script can't download can be downloaded with web browser, so problem is not in servers.
As I said nothing works, as using pre-built images aswell  compiling locally. Less than month ago is worked fine.
I run armbian compile script on ubuntu 22.04 virtual machine

The-going, 2022-11-08:
> [ error ] ERROR in function install_deb_chroot

I confirm. This function started working with errors. And I can't figure out why.

igorpecovnik, 2022-11-08:
Its APT cache issue. Try with: 
./compile.sh EXPERT=yes NO_APT_CACHER="yes"

easy-and-simple, 2022-11-08:
Same thing. It reports errors and errors and errors and images can't be build. Sometimes it scratches to the disk to the end but there is no image in images folder
How to get back 1 month, when it worked fine?

igorpecovnik, 2022-11-09:
> How to get back 1 month, when it worked fine?

! easy-and-simple, another problems are expected. When you go outside the main branch, you are on your own.

Can you provide exact set of command how you build this - so I can try to recreate.

./comile.sh .... or if you go interactively - what you choose and if you are changing anything in the config

easy-and-simple, 2022-11-09:
No matter what I choose, it always gives me errors and no image is created. I configure compilation interactively
One example configuration is something like:
1  ./compile.sh EXPERT="yes" NO_APT_CACHER="yes"
2. Full OS image
3.  Use precompiled packages
4.  choose any board (I tried odroid n2 and rock 3a)
5.  edge or current (no matter)
6. jammy (I tried only with ununtu jammy)
7. desktop or server (no matter)
8. desktop -cinnamon
9. base config (the only option)
10.  3d and remote desktop

When I compile locally I choose same options and then errors are other, but in any case image build is unsuccessful
Server images also return errors and no image is build

igorpecovnik, 2022-11-09:
"3. Use precompiled packages"

Don't use that.

closed 2022-11-09 by igorpecovnik

easy-and-simple, 2022-11-10:
I dont think it is network issue, because as I said earlier, if I interrupt execution of the compile script and paste failed link in browser is starts to download it immediately.
Probably problem is in the compile script itself. Maybe it uses inappropriate way to detect resource availability.
Now both options with compiling the kernel are working, so main issues are resolved and this issue can be closed.
I would not tried to build image by myself if video driver worked.
I will close the thread later

igorpecovnik, 2022-11-10:
> paste failed link in browser is starts to download it immediately.

Link to routed address or link to our re-director apt/dl.armbian.com ? This can give different servers at different moments and if this is the case, we would need to closely inspect our infrastructure. Sometimes troubles are hard to find ...

> Probably problem is in the compile script itself. Maybe it uses inappropriate way to detect resource availability.

Compile system certainly can have troubles as it is a complex system, but as i said, automation, which mainly starts from a clean image, produce images daily / hourly. Docker images are updated once per week. If we have a trouble on that level, its also discovered quickly. I would very much like to help, but running low on ideas where to look.

Rearmit, 2022-11-11:
I have similar issue

11/11 07:12:36 [ERROR] CUID#6 - Download aborted. URI=https://mirror-eu-de1.armbian.airframes.io/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://mirror-eu-de1.armbian.airframes.io/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/11 07:12:36 [ERROR] CUID#7 - Download aborted. URI=https://imola.armbian.com/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://imola.armbian.com/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/11 07:12:36 [ERROR] CUID#8 - Download aborted. URI=https://stpete-mirror.armbian.com/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://stpete-mirror.armbian.com/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#9b6459 0B/0B CN:1 DL:0B]                                                                                                                                                                                                                                                       
11/11 07:12:36 [ERROR] CUID#9 - Download aborted. URI=https://fi.mirror.armbian.de/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://fi.mirror.armbian.de/cache//rootfs/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/11 07:12:37 [ERROR] CUID#10 - Download aborted. URI=https://github.com/armbian/cache/releases/download/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://github.com/armbian/cache/releases/download/0080/armhf-bullseye-cli-0635a7bc-0080.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

aria2 will resume download if the transfer is restarted.
If there are any errors, then see the log file. See '-l' option in help/man page for details.
[ o.k. ] Checking cache [ armhf-bullseye-cli-0635a7bc-0079.tar.zst ]
[ .... ] Downloading from servers 

11/11 07:12:38 [ERROR] CUID#6 - Download aborted. URI=https://mirror-eu-de1.armbian.airframes.io/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://mirror-eu-de1.armbian.airframes.io/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/11 07:12:38 [ERROR] CUID#7 - Download aborted. URI=https://imola.armbian.com/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://imola.armbian.com/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/11 07:12:39 [ERROR] CUID#8 - Download aborted. URI=https://stpete-mirror.armbian.com/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://stpete-mirror.armbian.com/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/11 07:12:39 [ERROR] CUID#9 - Download aborted. URI=https://fi.mirror.armbian.de/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://fi.mirror.armbian.de/cache//rootfs/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#2b309b 0B/0B CN:1 DL:0B]                                                                                                                                                                                                                                                       
11/11 07:12:39 [ERROR] CUID#10 - Download aborted. URI=https://github.com/armbian/cache/releases/download/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://github.com/armbian/cache/releases/download/0079/armhf-bullseye-cli-0635a7bc-0079.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

igorpecovnik, 2022-11-11:
This looks like cache production issue. If cache is missing, you should see those errors, yes, but build process should still work. Does it?

Rearmit, 2022-11-11:
It dosen't before this morning.
But now it work because https://stpete-mirror.armbian.com/cache//rootfs/0080/armhf-bullseye-cli-baaba586-0080.tar.zst now work
But armhf-bullseye-cli-baaba586-0081.tar.zst.asc return 404

Before this morning it try to download armhf-bullseye-cli-baaba586-0080.tar.zst and return 404 after that it try to download
armhf-bullseye-cli-baaba586-0079.tar.zst that also return 404

igorpecovnik, 2022-11-11:
Cache has been recreated today https://github.com/armbian/cache/releases/tag/0081 successfully. "Resource not found" could pop up since not all servers has the content atm, but it should be synced in 6-12h.

Test:

[ o.k. ] Runtime [ 6 min ]
[ o.k. ] Repeat Build Options [ ./compile.sh  BOARD=odroidn2 BRANCH=current RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=prebuilt COMPRESS_OUTPUTIMAGE=sha,gpg,img  ]