armbian/build #4388: line 44: cli_entrypoint: command not found

Konjaks, 2022-11-06:
I had the problem to even run the compile.sh file, i needed to convert it to unix-format. 

Fortunately, i got after that some errors. So it is seenable that there is something wrong.

ubuntu-****@Konjaks:/mnt/c/Users/Konjak/Desktop/build$ sudo ./_compile.sh
[sudo] password for ubuntu-****: 
/mnt/c/Users/Konjak/Desktop/build/lib/import-functions.sh: line 2: $'\r': command not found
/mnt/c/Users/Konjak/Desktop/build/lib/import-functions.sh: line 6: syntax error near unexpected token `done'
'mnt/c/Users/Konjak/Desktop/build/lib/import-functions.sh: line 6: `done <<< "$(find "${SRC}/lib/functions" -name "*.sh")"
./_compile.sh: line 44: cli_entrypoint: command not found

EvilOlaf, 2022-11-07:
> /mnt/c/Users/
Seems like WSL which is not supported. Try proper VM or native.

Konjaks, 2022-11-07:
Yeah it's WSL, i will try it nativ