armbian/build #4413: Problem found in ...

AaronNGray, 2022-11-10:
**Describe the bug**
Getting the following error :-
```
  LD [M]  fs/xfs/xfs.o
  CC [M]  kernel/kheaders.o
[ error ] ERROR in function compile_kernel [ main.sh:589 -> main.sh:490 -> compilation.sh:506 -> general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```
./compile.sh docker
```

**To Reproduce**
Using WSL to run docker within Windows

**Provide logs:**
Dunno where I get the logs from Docker ?

igorpecovnik, 2022-11-10:
> Using WSL to run docker within Windows

https://github.com/armbian/build#basic-requirements

WSL is officially not supported. WSL - Docker - Qemu ... now go and debut ;)

AaronNGray, 2022-11-10:
mn may use Ubuntu VM then : [

igorpecovnik, 2022-11-10:
Sorry, I think its now the only / best way, but someday WSL will probably also be supported. You can try development version https://github.com/armbian/build/tree/armbian-next but there other problems are expected.

closed 2022-11-10 by igorpecovnik

AaronNGray, 2022-11-10:
Same issue on a different test build
```
CC [M]  fs/binfmt_misc.o
  AR      fs/built-in.a
[ error ] ERROR in function compile_kernel [ main.sh:589 -> main.sh:490 -> compilation.sh:506 -> general.sh:0 ]
```

igorpecovnik, 2022-11-10:
output/logs/*

Ahaa, I see. upstream has changed https://www.kernel.org/ , and possible troubles came ... running all kernel rebuild to see where is the issue.

AaronNGray, 2022-11-10:
I am just trying armbian-next build on WSL, also while downloading Ubuntu 22.04.1 LTS (Jammy Jellyfish) too in parallel :)

igorpecovnik, 2022-11-10:
last bootloader bump to 2022.07 made troubles ... but this is parhaps not related. Dunno.

Make a PR here https://github.com/armbian/documentation/blob/master/docs/Release_Board-Maintainers.md and leave some contacts https://www.armbian.com/update-data/

@EvilOlaf

AaronNGray, 2022-11-10:
@igorpecovnik - I was fine using Armbian for about 3 years, starting with Jessie, then updated to Stretch, doing 'apt upgrades' all fine nice reliable server with uptimes of years, then recently I updated to Buster, it all seemed okay for months, but there were no restarts, I think I did an upgrade on Buster and things went pair shaped. I have just tried an old Buster copied image and updated it to Bulseye using /etc/apt/sources.list manual update and standard apt proceedures, and I am getting exactly the same bad boot behaviour. 
Do I file this on GitHub Issues (if so where) or Jira (which I don't know yet) ?

igorpecovnik, 2022-11-10:
> Do I file this on GitHub Issues 

Github issues are for build system only.

Jira. I'll sent you invite once i have your email.

AaronNGray, 2022-11-10:
aaronngray@gmail.com

AaronNGray, 2022-11-10:
Okay done Documentation PR and and updated Armbian update-date.

igorpecovnik, 2022-11-10:
If you run docker then check if 
DOCKER_FLAGS+=(--privileged)
is uncommented in userpatches/config-docker.conf

igorpecovnik, 2022-11-10:
and:
- https://docs.armbian.com/Development-Code_Review_Procedures_and_Guidelines/
- read forums
- see logs

AaronNGray, 2022-11-10:
Okay I am on UK BST so will pick up with it again tomorrow afternoon now and will post logs, and do the M3 Jira issue, hopefully I should be able to catch the M3 Linux logs using a second M3 and USB SDCard reader.

AaronNGray, 2022-11-11:
Moving over to a Hyper-V Ubuntu 22.04.1 LTS (Jammy Jellyfish) for now.

igorpecovnik, 2022-11-11:
> zstdmt: command not found

This means you are not using latest DockerFile in userpatches/ or you have some old docker image. Remove it and make it again. Make sure to have latest DockerFile https://github.com/armbian/build/blob/master/config/templates/Dockerfile#L120

AaronNGray, 2022-11-11:
I had cleared the Docker image and did a git pull

igorpecovnik, 2022-11-11:
>  This is armbian-next branch.

This should be just for reference. Here I can't help as its in development, expected to break.

AaronNGray, 2022-11-11:
Ah it was `master` branch afterall, my mistake. 

Okay I will use Hyper-V Ubuntu 22.04.1 LTS (Jammy Jellyfish) for now.

AaronNGray, 2022-11-11:
Hyper-V is a lot slower than Docker !

AaronNGray, 2022-11-12:
Okay I got a Buster image using Hyper-V Ubuntu 22.04.1 LTS (Jammy Jellyfish).

Ubuntu came up with dialog warning some process had been terminate prematurely, sorry I cannot remember which for certain it was last thing last night. I will see if this happens again.

The BananaPi M3 Buster image runs okay on cold boots, warm boots fail. I am just saving the uSDCard to an image so will have access to to logs. I will put them online for access if there is nothing I can see that is obvious.

I am going to build an old Jessie or Stretch image to test cold boots on and get the logs. I have four BananaPi M3's hence my interest in getting them working correctly with latest Debian.