armbian/build #4472: Unable to build Buster

AaronNGray, 2022-11-20:
**Describe the bug**
I am unable to build Buster I keep getting the following error :-1: 
```
[ o.k. ] Checking MD5 sum of installed packages [ debsums ]
debsums: changed file /usr/share/icons/Numix-Circle/icon-theme.cache (from numix-icon-theme-circle package)
[ error ] ERROR in function create_rootfs_cache [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:228 -> functions/logging/traps.sh:0 ]
[ error ] MD5 sums check of installed packages failed 
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:228 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 ]
[ o.k. ] Unmounting [ /home/aaron/GitHub/build/.tmp/rootfs-1ec9780e-690d-423b-a374-526b894bd1b3/ ]
[ error ] ERROR in function unmount_on_exit [ functions/cli/cli-entrypoint.sh:108 -> functions/main/default-build.sh:125 -> functions/main/rootfs-image.sh:65 -> functions/rootfs/create-cache.sh:228 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:26 -> functions/logging/traps.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 

```
**To Reproduce**
git clone https://github.com/armbian/build.git
cd build
sudo ./compile.sh BOARD=bananapim3 release=buster EXPERT=yes

**Provide logs:**
[compilation.log](https://github.com/armbian/build/files/10050053/compilation.log)
[compiler.log](https://github.com/armbian/build/files/10050054/compiler.log)
[debootstrap-list.log](https://github.com/armbian/build/files/10050055/debootstrap-list.log)
[hostdeps.log](https://github.com/armbian/build/files/10050056/hostdeps.log)
[install.log](https://github.com/armbian/build/files/10050057/install.log)

AaronNGray, 2022-11-20:
There does not seem to be a lot to go on in the logs :(

EvilOlaf, 2022-11-20:
[Buster is no longer supported. Only Bullseye and Jammy are supported userspace environments.](https://github.com/armbian/build/blob/e799495de999922adeebb873ea4c62119ff4d723/lib/functions/configuration/menu.sh#L57-L68)

closed 2022-11-20 by EvilOlaf

AaronNGray, 2022-11-20:
@EvilOlaf I am trying to get a working system on BananaPi M3 and an updated Jessie to Buster was the last one I had working system with so am trying to recreate a build for Buster. 

I suspect the Kernel 5.15.79 but have to go with what I can get working from what I have.

I have a minimal Buster image buliding with :=
```
sudo ./compile.sh BOARD=bananapim3 release=buster EXPERT=yes
```
but am getting both soft and hard intermittent reboot issues/

And am trying to debug this issue so I would appreciate it if you could please reopen this issue for it to be visible for others to possibly help.

igorpecovnik, 2022-11-20:
> debsums: changed file /usr/share/icons/Numix-Circle/icon-theme.cache

Problem is broken upstream package. I think we should add exception for this (and probably more) as we are not going to fix 3rd party packages.

> but am getting both soft and hard intermittent reboot issues.
> if you could please reopen this issue for it to be visible for others to possibly help.

This is not a problem if Armbian build system 
https://www.armbian.com/bugs/ -> https://forum.armbian.com/forum/204-banana-pi-m3/