armbian/build #4491: Determining correct userpatches `family` attribute

ashthespy, 2022-11-26:
Attempting to streamline some scripts, and would like to clarify the correct way to determine the `<family>` 
in [advanced_patch()](https://github.com/armbian/build/blob/925c344615b2cb02e76e34602f18a06c28012837/lib/functions/compilation/patch/patching.sh#L11) 

Take for example build configuration 
```
Board: orangepizero
Branch: edge
```
Is there a way to determine  `$USERPATCHES_PATH/<dest>/<family>/target_<target>` without parsing the board/source conf?

```
# log 
[ o.k. ] Started patching process for [ kernel sunxi-edge ]
[ o.k. ] Looking for user patches in [ userpatches/kernel/archive/sunxi-6.0 ]
```
The family seems to be set as `archive/sunxi-6.0` - i.e [`${KERNELPATCHDIR}`](https://github.com/armbian/build/blame/925c344615b2cb02e76e34602f18a06c28012837/lib/functions/compilation/kernel.sh). 

Is this correct? Previously it used to be something along the lines of `sunxi-{legacy,current,next}`
If it is correct, is there a simpler way to determine the appropriate "family" without parsing?

The-going, 2022-11-27:
> Previously it used to be something along the lines of [`sunxi-{legacy,current,next}`](https://docs.armbian.com/Developer-Guide_User-Configurations/#user-provided-patches)

These are symbolic links to the corresponding directories:
```
armbian-build> ls -al patch/kernel/sunxi-*
... patch/kernel/sunxi-current -> archive/sunxi-5.15
... patch/kernel/sunxi-edge -> archive/sunxi-6.0
... patch/kernel/sunxi-legacy -> archive/sunxi-5.10
```
The choice is here:
https://github.com/armbian/build/blob/26a00ebcd295db5634dfc30a8136f574e616de8d/config/sources/families/include/sunxi_common.inc#L13-L28

And further:
https://github.com/armbian/build/blob/26a00ebcd295db5634dfc30a8136f574e616de8d/config/sources/families/include/sunxi_common.inc#L44-L46

allows you to select another existing version for the edge. This ignores symbolic links.
This is correct!

The-going, 2022-11-27:
> Is there a way to determine `$USERPATCHES_PATH/<dest>/<family>/target_<target>` without parsing the board/source conf?

https://github.com/armbian/build/blob/26a00ebcd295db5634dfc30a8136f574e616de8d/lib/functions/compilation/kernel.sh#L28-L40
First of all, patches from the series are applied.
The `advanced_patch` function works last and the remaining patches are applied.

For it to work correctly, you must make symbolic links in the folder:
```
userpatches/kernel/sunxi-current -> archive/sunxi-5.15
userpatches/kernel/sunxi-edge -> archive/sunxi-6.0
userpatches/kernel/sunxi-legacy -> archive/sunxi-5.10
```

ashthespy, 2022-11-27:
>These are symbolic links to the corresponding directories: 

So this is not created by the build scripts correct? 

i.e I should parse each board.conf/source.conf to determine  `<family>-{current,edge,legacy}` -> `archive/<family>-${KERNEL_VERSION_LEVEL}`

The-going, 2022-11-27:
> So this is not created by the build scripts correct?

Kernel patches can be taken from two places:
`patch/kernel/` - here symbolic links are made by the maintainer manually.
`userpatches/kernel/` - you're the boss here.
 And only the user can change anything locally. The userpatches folder is ignored by git.

The-going, 2022-11-27:
I don't use the userpatches folder and this is what the `tree` command shows:
```bash
 tree userpatches/kernel/
userpatches/kernel/
├── archive
└── rockchip-rk3588-legacy
```
This is something that was created automatically.
For the required family, you need to create a directory structure
similar to in `patch/kernel/archive/` to `userpatches/kernel/archive/`.

closed 2023-04-29 by rpardini