armbian/build #4502: Some resources not found in building armbian for rockpi4b+

swordow, 2022-11-28:
**Describe the bug**
Some resources not found in building armbian for rockpi4b+

**To Reproduce**
./compile.sh
and choose rockpi4b+ and select desktop environment with xfce.

**Provide logs:**
11/28 04:06:14 [ERROR] CUID#6 - Download aborted. URI=https://mirror-us-phx1.armbian.airframes.io/cache//rootfs/0084/arm64-bullseye-xfce-5aba252a-0084.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://mirror-us-phx1.armbian.airframes.io/cache//rootfs/0084/arm64-bullseye-xfce-5aba252a-0084.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#d14dcc 0B/0B CN:1 DL:0B]
11/28 04:06:26 [ERROR] CUID#7 - Download aborted. URI=https://github.com/armbian/cache/releases/download/0084/arm64-bullseye-xfce-5aba252a-0084.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://github.com/armbian/cache/releases/download/0084/arm64-bullseye-xfce-5aba252a-0084.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#d14dcc 0B/0B CN:1 DL:0B]
11/28 04:06:27 [ERROR] CUID#8 - Download aborted. URI=https://kr.mirrors.naho.moe/armbian/cache//rootfs/0084/arm64-bullseye-xfce-5aba252a-0084.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://kr.mirrors.naho.moe/armbian/cache//rootfs/0084/arm64-bullseye-xfce-5aba252a-0084.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

aria2 will resume download if the transfer is restarted.
If there are any errors, then see the log file. See '-l' option in help/man page for details.
[ o.k. ] Checking cache [ arm64-bullseye-xfce-5aba252a-0083.tar.zst ]
[ .... ] Downloading from servers

11/28 04:06:31 [ERROR] CUID#6 - Download aborted. URI=https://mirror-us-phx1.armbian.airframes.io/cache//rootfs/0083/arm64-bullseye-xfce-5aba252a-0083.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://mirror-us-phx1.armbian.airframes.io/cache//rootfs/0083/arm64-bullseye-xfce-5aba252a-0083.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#e3406c 0B/0B CN:1 DL:0B]
11/28 04:06:52 [ERROR] CUID#7 - Download aborted. URI=https://github.com/armbian/cache/releases/download/0083/arm64-bullseye-xfce-5aba252a-0083.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://github.com/armbian/cache/releases/download/0083/arm64-bullseye-xfce-5aba252a-0083.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

11/28 04:06:53 [ERROR] CUID#8 - Download aborted. URI=https://kr.mirrors.naho.moe/armbian/cache//rootfs/0083/arm64-bullseye-xfce-5aba252a-0083.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://kr.mirrors.naho.moe/armbian/cache//rootfs/0083/arm64-bullseye-xfce-5aba252a-0083.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

aria2 will resume download if the transfer is restarted.
If there are any errors, then see the log file. See '-l' option in help/man page for details.
[ o.k. ] Checking cache [ arm64-bullseye-xfce-5aba252a-0082.tar.zst ]
[ .... ] Downloading from servers

11/28 04:06:55 [ERROR] CUID#6 - Download aborted. URI=https://mirror-us-phx1.armbian.airframes.io/cache//rootfs/0082/arm64-bullseye-xfce-5aba252a-0082.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://mirror-us-phx1.armbian.airframes.io/cache//rootfs/0082/arm64-bullseye-xfce-5aba252a-0082.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#ffaafd 0B/0B CN:1 DL:0B]
11/28 04:07:06 [ERROR] CUID#7 - Download aborted. URI=https://github.com/armbian/cache/releases/download/0082/arm64-bullseye-xfce-5aba252a-0082.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://github.com/armbian/cache/releases/download/0082/arm64-bullseye-xfce-5aba252a-0082.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found
[#ffaafd 0B/0B CN:1 DL:0B]
11/28 04:07:07 [ERROR] CUID#8 - Download aborted. URI=https://kr.mirrors.naho.moe/armbian/cache//rootfs/0082/arm64-bullseye-xfce-5aba252a-0082.tar.zst.asc
Exception: [AbstractCommand.cc:351] errorCode=3 URI=https://kr.mirrors.naho.moe/armbian/cache//rootfs/0082/arm64-bullseye-xfce-5aba252a-0082.tar.zst.asc
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

If the above resources are not supported, so is there any guide/method to build them and upload to servers?

igorpecovnik, 2022-11-30:
We have several automated builds in place and we haven't notice any troubles of this kind.

closed 2022-11-30 by igorpecovnik

swordow, 2022-11-30:
> We have several automated builds in place and we haven't notice any troubles of this kind.

The download urls are not setted in any script in armbin/build repo and they are recorded in some servers i think .
If you click into any one of the urls above, the response will be 404.

igorpecovnik, 2022-11-30:
> If you click into any one of the urls above, the response will be 404.

That is expected behaviour. We only keep two cache https://imola.armbian.com/cache/rootfs/ banks. If files are changed or when new cache is generated, older (then current -1) become obsolete.

If you are building from a build script that is not master, it could always be displaying this as cache files does not exists ...

This output we need to deal with once, but otherwise, issues does not exists.

swordow, 2022-11-30:
> > If you click into any one of the urls above, the response will be 404.
> 
> That is expected behaviour. We only keep two cache https://imola.armbian.com/cache/rootfs/ banks. If files are changed or when new cache is generated, older (then current -1) become obsolete.
> 
> If you are building from a build script that is not master (like in this case i would suspect), it could always be displaying this as cache files does not exists ...
> 
> This output we need to deal with once, but otherwise, issues does not exists.
> 
> Is this perhaps Radxa fork? Then its normal that it doesn't work as its 6 months + old.

No, it is the armbian/build master branch which outputs the above log.
I compiled using rockpi4bplus(csc) version and selected the xfce desktop envirnment.

swordow, 2022-11-30:
> That is expected behaviour. We only keep two cache https://imola.armbian.com/cache/rootfs/ banks. If files are changed or when new cache is generated, older (then current -1) become obsolete.
>
> If [...]
And how could i update the url links and then i could make a test.

igorpecovnik, 2022-11-30:
> I compiled using rockpi4bplus(csc) version and selected the xfce desktop envirnment.

Then you have select combination that is not having a cache files. We only generate most commonly used combinations:
https://github.com/armbian/scripts/blob/master/.github/workflows/update-cache.yml#L221-L234

swordow, 2022-11-30:
> > I compiled using rockpi4bplus(csc) version and selected the xfce desktop envirnment.
> 
> Then you have select combination that is not having cache files. We only generate most commonly used combinations: https://github.com/armbian/scripts/blob/master/.github/workflows/update-cache.yml#L221-L234

Got it.