armbian/build #4567: "not have a Release file" error when building according to CONTRIBUTING.md

avtoneru, 2022-12-14:
I did everything according to the instructions on the CONTRIBUTING.md page. After 
"sudo apt-add-repository https://cli.github.com/packages" 
it displays the following:

**Repository: 'deb https://cli.github.com/packages jammy main'
Description:
Archive for codename: jammy components: main
More info: https://cli.github.com/packages
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Found existing deb entry in /etc/apt/sources.list.d/archive_uri-https_cli_github_com_packages-jammy.list
Adding deb entry to /etc/apt/sources.list.d/archive_uri-https_cli_github_com_packages-jammy.list
Found existing deb-src entry in /etc/apt/sources.list.d/archive_uri-https_cli_github_com_packages-jammy.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/archive_uri-https_cli_github_com_packages-jammy.list
Ign:1 https://cli.github.com/packages jammy InRelease
Hit:2 http://ru.archive.ubuntu.com/ubuntu jammy InRelease
Err:3 https://cli.github.com/packages jammy Release
  404  Not Found [IP: 185.199.109.153 443]
Hit:4 http://ru.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:5 http://ru.archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:6 http://ru.archive.ubuntu.com/ubuntu jammy-security InRelease
Reading package lists... Done
E: The repository 'https://cli.github.com/packages jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.**

Please, help.

igorpecovnik, 2022-12-14:
> Please, help.

Try installing GH tool this way https://github.com/cli/cli/blob/trunk/docs/install_linux.md

avtoneru, 2022-12-14:
If you do as described in https://github.com/cli/cli/blob/trunk/docs/install_linux.md it still gives an error...

**4+1 records in
4+1 records out
2270 bytes (2.3 kB, 2.2 KiB) copied, 0.00206541 s, 1.1 MB/s
Hit:1 http://ru.archive.ubuntu.com/ubuntu jammy InRelease
Ign:2 https://cli.github.com/packages jammy InRelease
Hit:3 http://ru.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:4 https://cli.github.com/packages stable InRelease
Hit:5 http://ru.archive.ubuntu.com/ubuntu jammy-backports InRelease
Err:6 https://cli.github.com/packages jammy Release
  404  Not Found [IP: 185.199.108.153 443]
Hit:7 http://ru.archive.ubuntu.com/ubuntu jammy-security InRelease
Reading package lists... Done
E: The repository 'https://cli.github.com/packages jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.**

avtoneru, 2022-12-16:
The script is running. Thank you.

closed 2022-12-16 by avtoneru