armbian/build #4599: rockchip64_common.inc contain some jetson-nano code

xlazom00, 2022-12-21:
rockchip64_common.inc contain some jetson-nano code

https://github.com/armbian/build/blob/master/config/sources/families/include/rockchip64_common.inc#L331

xlazom00, 2022-12-21:
in dtb-6.1.0-rockchip64/nvidia
tegra132-norrin.dtb
tegra186-p2771-0000.dtb
tegra186-p3509-0000+p3636-0001.dtb
tegra194-p2972-0000.dtb
tegra194-p3509-0000+p3668-0000.dtb
tegra194-p3509-0000+p3668-0001.dtb
tegra210-p2371-0000.dtb
tegra210-p2371-2180.dtb
tegra210-p2571.dtb
tegra210-p2894-0050-a08.dtb
tegra210-p3450-0000.dtb
tegra210-smaug.dtb
tegra234-p3737-0000+p3701-0000.dtb
tegra234-sim-vdk.dtb

igorpecovnik, 2023-01-04:
Which HW is this? Is this perhaps media kernel?

xlazom00, 2023-01-04:
@igorpecovnik
This code
https://github.com/armbian/build/blob/master/config/sources/families/include/rockchip64_common.inc#L338

And also kernel config has tegra platfom enebled
It is simmilar as my rk3588 mainline patch. I think we should clean up kernel configs

igorpecovnik, 2023-01-04:
>  I think we should clean up kernel configs

media kernel covers more sub-architectures and some Rockchip boards (Nanopc T4 for example) are supported with that kernel. 

But yes, I agree, we should clean kernel configs and have all enabled only in UEFI kernel https://github.com/armbian/build/blob/master/config/kernel/linux-uefi-arm64-edge.config#L298-L341

xlazom00, 2023-01-04:
@igorpecovnik that is fine
I am only talking about kernel configs for specific platforms.
UEFI ARM64 is something different
https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-edge.config#L328

igorpecovnik, 2023-01-05:
> I am talking about for example rockchip64 kernel config

Ahaa. Yes, that needs cleaning.

xlazom00, 2023-01-05:
@igorpecovnik and this
https://github.com/armbian/build/blob/master/config/sources/families/include/rockchip64_common.inc#L338
why rockchip64_common.inc contains code for tegra? :)

igorpecovnik, 2023-01-05:
I have absolutely no idea. Git blame shows @150balbes

150balbes, 2023-01-06:
> why rockchip64_common.inc contains code for tegra? :)

This code should be.

xlazom00, 2023-01-06:
@150balbes And media group also use rockchip64 kernel config?

150balbes, 2023-01-06:
> And media group also use rockchip64 kernel config?

No, it has its own kernel configuration (common to all devices of the media Rockchip and Jetson Nano group).

rpardini, 2023-01-06:
let me speculate.

`media` includes `rockchip64_common.inc` to be able to use `prepare_boot_configuration()` and others

since there can be only one `family_tweaks()` defined then the tweaks for `media` boards ends up in rockchip64.

Tonymac32, 2023-10-03:
@igorpecovnik is this still valid in the "main" branch?  I don't see the above reference to the nano, can it be closed?

igorpecovnik, 2023-11-15:
No, seems not relevant in "main" anymore, closing.

closed 2023-11-15 by igorpecovnik