armbian/build #4690: readme.md

150balbes, 2023-01-12:
### Which feature would you like to have?

I suggest adding the options --depth=1 --branch=master to the description of the cloning steps

git clone --depth=1 --branch=master https://github.com/armbian/build

This will significantly speed up code loading for ordinary users. For experienced users, it will not be difficult to get the code in any form.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

rpardini, 2023-01-12:
I've no problem with this. GitHub.com server is highly optimized for shallow fetches, since they deal a lot with that, they even  default to a depth of 1 in the standard GHA `actions/checkout` Action. So it should be ok.

EvilOlaf, 2023-02-05:
I pushed the necessary PRs

closed 2023-02-05 by EvilOlaf