armbian/build #4708: Problem found ...

piet-vr, 2023-01-17:
### What happened?

Hi,

I am trying to build real time kernel image for BananaPi M5. I have tried several combinations in Build for days now, but get errors. 

Could you please assist. I added output.log for my last attempt.

I am not sure if the BPi M5 is even capable of handling real time and is experimenting as I can not find a RT pre build image for the M5 to see if LinuxCNC will work on M5.

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Bullseye

### Relevant log output

```shell
Displaying message: Extension manager processed 2 Extension Methods calls and 2 Extension Method implementations info
Checking if /home/craftsmancnc/build/config/desktop/bullseye/environments/cinnamon is available for arm64 in /home/craftsmancnc/build/config/desktop/bullseye/environments/cinnamon/only_for
Checking if /home/craftsmancnc/build/config/desktop/bullseye/environments/gnome is available for arm64 in /home/craftsmancnc/build/config/desktop/bullseye/environments/gnome/only_for
Checking if /home/craftsmancnc/build/config/desktop/bullseye/environments/xfce is available for arm64 in /home/craftsmancnc/build/config/desktop/bullseye/environments/xfce/only_for
Checking if /home/craftsmancnc/build/config/desktop/bullseye/environments/gnome is available for arm64 in /home/craftsmancnc/build/config/desktop/bullseye/environments/gnome/only_for

## BUILD SCRIPT ENVIRONMENT

Repository: 
Version: 

Host OS: buster
Host arch: amd64
Host system: Linux CraftsmanCNC 4.19.0-17-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux
Virtualization type: oracle

## Build script directories
Build directory is located on:
TARGET SOURCE    FSTYPE AVAIL
/      /dev/sda1 ext4   20.7G

Build directory permissions:
# file: /home/craftsmancnc/build
# owner: craftsmancnc
# group: craftsmancnc
user::rwx
group::r-x
other::r-x

Temp directory permissions:
# file: /home/craftsmancnc/build/.tmp
# owner: root
# group: root
user::rwx
group::rwx
other::r-x

## BUILD CONFIGURATION

Build target:
Board: bananapim5
Branch: current
Minimal: no
Desktop: yes
Desktop Environment: gnome
Software groups: 

Kernel configuration:
Repository: git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
Branch: branch:linux-6.1.y
Config file: linux-meson64-current

U-boot configuration:
Repository: https://source.denx.de/u-boot/u-boot.git
Branch: tag:v2022.10
Config file: bananapi-m5_defconfig

Partitioning configuration: msdos offset: 4
Boot partition type: (none) 
Root partition type: ext4 

CPU configuration: 667000 - 2100000 with ondemand
Displaying message: Preparing host info
Displaying message: Build host OS release buster info
Displaying message: Installing build dependencies
Displaying message: Syncing clock host info
Displaying message: Checking for external GCC compilers  info
Displaying message: Checking git sources u-boot v2022.10 info
Displaying message: Up to date
Displaying message: Checking git sources linux-mainline linux-6.1.y info
Displaying message: Up to date
Displaying message: Checking git sources odroidc2-blobs master info
Displaying message: Up to date
Displaying message: Checking git sources amlogic-boot-fip master info
Displaying message: Up to date
Displaying message: Building deb armbian-plymouth-theme info
Displaying message: Checking for rootfs cache current bananapim5 bullseye gnome no info
Displaying message: Checking cache arm64-bullseye-gnome-45d82679-0118.tar.zst info
Displaying message: Extracting arm64-bullseye-gnome-45d82679-0118.tar.zst 0 days old info
Displaying message: Adding Armbian repository and authentication key /etc/apt/sources.list.d/armbian.list info
Displaying message: Applying distribution specific tweaks for bullseye info
Displaying message: Configuring locales en_US.UTF-8 info
Displaying message: Applying common tweaks  info
Displaying message: Cleaning package lists
Displaying message: Updating package lists
Displaying message: Temporarily disabling initramfs-tools hook for kernel
Displaying message: Installing linux-u-boot-current-bananapim5_23.02.0-trunk_arm64.deb
Displaying message: Installing linux-image-current-meson64_23.02.0-trunk_arm64.deb
Displaying message: Installing linux-dtb-current-meson64_23.02.0-trunk_arm64.deb
Displaying message: Installing armbian-bsp-cli-bananapim5_23.02.0-trunk_arm64.deb
Displaying message: Installing armbian-bullseye-desktop-gnome_23.02.0-trunk_all.deb
Displaying message: Installing armbian-bsp-desktop-bananapim5_23.02.0-trunk_arm64.deb
Displaying message: Installing armbian-firmware_23.02.0-trunk_all.deb
Displaying message: Installing armbian-config_23.02.0-trunk_all.deb
Displaying message: Installing armbian-zsh_23.02.0-trunk_all.deb
Displaying message: ERROR in function install_deb_chroot functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:285 -> functions/main/build-tasks.sh:208 -> functions/main/rootfs-image.sh:77 -> functions/rootfs/distro-agnostic.sh:377 -> functions/rootfs/apt-install.sh:22 -> functions/logging/traps.sh:0 err
Displaying message: Installation of /root/armbian-zsh_23.02.0-trunk_all.deb failed bananapim5 bullseye yes meson64 err
Displaying message: Process terminated  info
Displaying message: unmount_on_exit() called! functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:285 -> functions/main/build-tasks.sh:208 -> functions/main/rootfs-image.sh:77 -> functions/rootfs/distro-agnostic.sh:377 -> functions/rootfs/apt-install.sh:22 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:0 err
Displaying message: Unmounting /home/craftsmancnc/build/.tmp/rootfs-2cc28e57-7086-46c2-85fb-f601294fa3be/ info
Displaying message: ERROR in function unmount_on_exit functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:285 -> functions/main/build-tasks.sh:208 -> functions/main/rootfs-image.sh:77 -> functions/rootfs/distro-agnostic.sh:377 -> functions/rootfs/apt-install.sh:22 -> functions/logging/traps.sh:1 -> functions/main/rootfs-image.sh:27 -> functions/logging/traps.sh:0 err
Displaying message: debootstrap-ng was interrupted  err
Displaying message: Process terminated  info
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hzyitc, 2023-01-18:
> Installation of /root/armbian-zsh_23.02.0-trunk_all.deb failed bananapim5 bullseye yes meson64 err

Please check `output/debug/install.log` to find out why it failed.

piet-vr, 2023-01-18:
Hi, thanks for the reply. Constantly an issue with qemu.

[install.log](https://github.com/armbian/build/files/10443834/install.log)

igorpecovnik, 2023-01-18:
>  Constantly an issue with qemu.

Sticking to officially recommended host is at most important to avoid unstable Qemu binaries. Kernel compilation usually don't have issues, while generating images does. https://github.com/armbian/build#basic-requirements Build [with Docker](https://docs.armbian.com/Developer-Guide_Building-with-Docker/) will also help.

This seems to be the problem:

>Host OS: buster
>Host arch: amd64
>Host system: Linux CraftsmanCNC 4.19.0-17-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux
>Virtualization type: oracle

piet-vr, 2023-01-18:
Thank you Igor, I get it. It can't build on amd machine. I will close issue. Thanks for your help.

closed 2023-01-18 by piet-vr

igorpecovnik, 2023-01-18:
You can, just use Docker or some more modern host system. Buster for host OS is very old and was never supported due old Qemu packages, probably something else too. With upcoming [improved build system](https://forum.armbian.com/topic/20337-armbian-next-development/?do=getNewComment), we will support more as its done better.

Also this board is probably possible to build natively on arm64 host just fine - in case you have some other arm64 hardware running Jammy (or at least Bullseye).

igorpecovnik, 2023-01-18:
> It can't build on amd machine.

Just a side note. 80% of packages and images we are providing, are build on AMD machines.