armbian/build #4761: free() invalid pointer

blmhemu, 2023-01-28:
### What happened?

[image]

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 2 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/472 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 44574 files and directories currently installed.)
Preparing to unpack .../python3-pkg-resources_59.6.0-1.2ubuntu0.22.04.1_all.deb ...
double free or corruption (out)
Aborted
dpkg: warning: old python3-pkg-resources package pre-removal script subprocess returned error exit status 134
dpkg: trying script from the new package instead ...
dpkg: ... it looks like that went OK
Unpacking python3-pkg-resources (59.6.0-1.2ubuntu0.22.04.1) over (59.6.0-1.2ubuntu0.22.04.1) ...
Preparing to unpack .../python3-setuptools_59.6.0-1.2ubuntu0.22.04.1_all.deb ...
Unpacking python3-setuptools (59.6.0-1.2ubuntu0.22.04.1) over (59.6.0-1.2ubuntu0.22.04.1) ...
Setting up python3-pkg-resources (59.6.0-1.2ubuntu0.22.04.1) ...
Setting up python3-setuptools (59.6.0-1.2ubuntu0.22.04.1) ...
free(): invalid pointer
Aborted
dpkg: error processing package python3-setuptools (--configure):
 installed python3-setuptools package post-installation script subprocess returned error exit status 134
Errors were encountered while processing:
 python3-setuptools
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-01-28:
Unmaintained hardware
Not a building issue. Please use forums for issues when *running* Armbian.

closed 2023-01-28 by EvilOlaf

rpardini, 2023-01-28:
Helios64 suffers (at least) from an overwritten DTS during rockchip64 patching, and is in dire need of a maintainer to clean it up. My efforts to find one have failed: people either have an old kernel running in a "production" helios64, or not willing to put in the effort.

blmhemu, 2023-01-28:
Have some questions. Would be great if you can help me understand the following
* What would it take to maintain it ?
  * Does it need intimate code knowledge or is it mostly rewriting existing patches ?
  * Do you have some pointers / prs on fixing this DTS overwrite issues ?
  * Should we constantly flash the device and lose persistent settings / config ?

igorpecovnik, 2023-01-28:
General instructions:
https://docs.armbian.com/Board_Maintainers_Procedures_and_Guidelines/

> Do you have some pointers / code references / prs on fixing this DTS overwrite issues ?

Check upstream code and what our patches does. Perhaps they are not needed anymore (maintainer - you - have to know that) at all or we have some features enabled that are not upstream or vice versa. Resolving that patch diffs.

> Should we constantly flash the device and lose "production" persistent settings / config ?

Only prior to upcoming release or by request (rarely). But checking functionality by booting from SD card is already a lot better then no checking at all. No need to mess up your existing setup.

blmhemu, 2023-01-28:
> Check upstream code and what our patches does. Perhaps they are not needed anymore (maintainer - you - have to know that) at all or we have some features enabled that are not upstream or vice versa. Resolving that patch diffs.

I just checked upstream code dts - it was unchanged from jan 2022. Not sure if upstream change is the reason.

Tonymac32, 2023-01-28:
For the DTS overwrite, basically we had a DTS patched into Armbian thanks to the Kobol guys before it was available in mainline.  Then it became available in mainline and the Armbian patch never got revised/removed.  All the helios64 specific patches need rebased on the mainline device tree.

blmhemu, 2023-01-28:
> For the DTS overwrite, basically we had a DTS patched into Armbian thanks to the Kobol guys before it was available in mainline. Then it became available in mainline and the Armbian patch never got revised/removed. All the helios64 specific patches need rebased on the mainline device tree.

Will try to do it and report if it worked or not.

rpardini, 2023-01-28:
> basically we had a DTS patched into Armbian thanks to the Kobol guys before it was available in mainline

Yes. The patch added a new file back then. Now, that file already exists in mainline, but is completely removed by the bash patching done in the `master` branch, thus becoming this monstrosity: https://github.com/rpardini/linux/commit/dc718a4d6ef073662c1d3f036136383a8428ff04

The fact it even boots is... surprising.

- A bit outdated patching summary https://rpardini.github.io/linux/armbian-next-rockchip64-6.1-20230126.html - notice ❌`[overwrites]` patches.

blmhemu, 2023-01-29:
So, I tried removing the patch `add-board-kobol-helios64...` in 5.15 and compiled armbian.
It did compile fine and even booted. Two things I observed:
* The fans are running at full speed unlike before.
* There is still a memory double free issue. Still getting the invalid pointer or double free issue. It honestly feels like something wrong with ubuntu.

rpardini, 2023-01-30:
> * The device booted ! Yay !
> * The fans are running at full speed unlike before.

Nice, probably some picking from the previous patch can result in a good-enough new patch.

> * The memory double free issue persists. Still getting the invalid pointer or double free issue. It honestly feels like something wrong with userspace / ubuntu / debian.

Did you try switching userspace? RELEASE=jammy etc?

blmhemu, 2023-01-30:
> Nice, probably some picking from the previous patch can result in a good-enough new patch.

Likely.

> Did you try switching userspace? RELEASE=jammy etc?

I tried with jammy, impish and kinetic. All suffer from this.

prahal, 2023-02-06:
I also have this issue running Ansible to the helios64 python3. I was able to reproduce your py3compile crash. Both py3compile crash and Ansible rule that crash rae random. I sometimes get a kernel oops. Unlikely DTS. Likely a kernel code patch. Might even affect other rockchip64 device because most apps runs fine on the helios64. I always ends up with a kernel memory error (also random) and need to reboot.

blmhemu, 2023-02-06:
> I also have this issue running Ansible to the helios64 python3. I was able to reproduce your py3compile crash. Both py3compile crash and Ansible rule that crash rae random. I sometimes get a kernel [...]

I tried searching for `free()` function calls, but could not find in patches 🤔

prahal, 2023-02-06:
@blmhemu free is a userspace call. There is no such call as free in the kernel. 
All in all, I do not see how a python3 call can end up in kernel memory corruption. But I am unable to reproduce the python3 free invalid pointer issue on another armbian box also running bullseye with linux-image current 22.11.4 kernel 6.1.7-meson64. So userspace python3 is identical on both.
It could still be that the issue is only harder to reproduce on armbian meson, hard to ascertain.

I was able to reproduce on helios64 with linux image edge 22.11.4 kernel 6.1.7-rockchip64 but I was able to do two runs of ansible before the bug "free(): invalid pointer" triggered. I also got "munmap_chunk(): invalid pointer" from running `sudo py3compile -p python3-pkg-resources` in a loop.

About the lack of maintainership, I would like to help with that, but until I find out why my board is unstable I focus on debugging the stability issue. Be it the kernel crash due to memory corruption (and half of the time the board hangs without even rebooting on panic, and most of the time at reboot after panic it hangs early on) or the rk3399 hs400es breakage that I nailed down to a commit in https://forum.armbian.com/topic/18855-upgrading-to-bullseye-troubleshooting-armbian-21081/page/3/#comment-128793.

igorpecovnik, 2023-02-07:
> About the lack of maintainership, I would like to help with that

https://docs.armbian.com/Board_Maintainers_Procedures_and_Guidelines/
tl;dr; = scan the text, send contact details (link is in the text) and beeing around to run some tests and fix (or at least log) problems prior to major releases / when kernel version is changed or similar bigger changes like upcoming build system upgrade.

rpardini, 2023-02-07:
Next suggestion: remove kernel patches, maybe all of them, and try a build with mostly mainline only stuff. Does problem persist? If not (my bet...) bisect the evil patch out of rockchip64... another idea: update u-boot and/or blobs.

prahal, 2023-02-07:
I change vin-supply in pwm-supply in helios64 board dts vdd-log section. vdd-log is known for stability issues if not powered properly and without this fix it was assigned the dummy regulator.

Hard stretched but I am still not set if the helios64 kernel memory errors I have are due to a driver which corrupts the memory or a wrong setting that makes the CPU unstable.

prahal, 2023-02-08:
@blmhemu do you run OMV above helios64?

OMV has a few optimizations (sysctl), tools that stress the memory (folder2ram), and probably others that might make the bug from the unknown source more visible, but it could be that the issue is not helios64 specific.

Tonymac32, 2023-02-12:
can anyone reproduce this issue with another RK3399 board?

blmhemu, 2023-02-13:
> @blmhemu do you run OMV above helios64?
No. Plain ubuntu.

prahal, 2023-02-20:
@blmhemu could you paste the output of /proc/buddyinfo when python3 starts to output invalid free? It seems that python3 does not cope well when an allocation fails and tries to free it even if it was not allocated. That may explain our issue.
I had page allocation of order 7 errors in my kernel log (but not always).

What is not clear is why we get these even though all was fine before. The fact is it may be another issue. In the process of debugging this invalid free I turned off my zswap so maybe I produced a page allocation failure with my debug attempts.
Also, it is not that I do not have memory left but that memory is fragmented and there are no higher allocation pages left. 

I tested in a docker container on the helios64 (still with latest master edge kernel) with Debian bookworm python3.11 and was able to reproduce the invalid free.

prahal, 2023-02-20:
@blmhemu sorry to bother you again. Do your tests with different releases (jammy, kinetic, etc) all run with a different "current" armbian kernel, or did the build run with the latest kernel (6.1 ?)?
The issue is python3.9 on my board was not upgraded for months and it was working (a long time ago but I cannot tell when it broke. Still, I doubt it was broken a year ago. So this leaves the libraries below python3.9 or the kernel. Though if you already tested a lot of stable kernels this makes the kernel an unlikely target and can save time in debugging.
I tried the latest armbian kernel 6.1.12 without armbian patches (but still with helios64 armbian dts, aufs and wifi patches) and I still get the python3 free invalid.

prahal, 2023-02-21:
I made a mistake and thus u-boot booted on my old eMMC install (which I left untouched since at least July 2022).
The issue python3 invalid free is reproducible there ... 5.15.48-rockchip64 #22.05.3. I do not know how I did notice back then this ansible python3 issue (I already had ansible setup for the helios64 but it is true that I had issues with my roles and thus tended to run the playbooks on a targeted system so maybe I did not try helios64 for a long time or at all).

Then I tried on the SD card install (up to date bullseye) with current armbian kernel `Linux helios64 5.15.89-rockchip64 #22.11.4` and the issue is also reproducible.

blmhemu, 2023-02-21:
Hey @prahal ! I am currently a bit busy with life and will likely start testing from next week. In the meanwhile, I did install the latest bullseye from [this](https://imola.armbian.com/archive/helios64/archive/) armbian mirror and it works smoothly with ansible and could not see this free issue.

prahal, 2023-02-22:
@blmhemu I confirm that just installing linux-image-current-rockchip64 package and its ad-hoc linux-dtb current-rockchip64 package at version 21.08.2, which is 5.10.63-rockchip64, fixes this issue. 
I was able to run:
```
for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
```
five times without an issue.

I can even run the test case fine with latest kernels if I disable cpufreq with kernel boot parameter cpufreq.off=1

prahal, 2023-02-27:
I have been able to reproduce the python3 invalid free with linux-image-legacy-rk3399 that is 4.4.213. I believe I did not encounter the issue before because before my ansible setup was using the python2 installed on the helios64, not the python3.

@blmhemu, I need to retry but I believe with cpufreq disabled (`cpufreq.off=1` on linux kernel command line) I was not able to reproduce the python3 invalid free with latest kernel (and probably current too). If you could confirm that would help. You can add `cpufreq.off=1` to /boot/armbianEnv.txt `extraargs=`.

Also I tried latest 6.1.12 with cpufreq enabled and all armbian patches removed except the add helios64 board add-board-helios64.patch, board-helios64-remove-pcie-ep-gpios.patch, my emmc hs400 es patch to read emmc hs400 and rk3399-enable-dwc3-xhci-usb-trb-quirk.patch and I can still reproduce the issue (I also tried with `EXTRAWIFI=no AUFS=no`).

thomas-maurice, 2023-03-27:
Hello !

@prahal I can confirm that the `cpufreq.off=1` worked on the current bulleseye build, I was able to successfully install OMV on my helios64 wihtout any issues

```
root@helios64:~# uname -a
Linux helios64 5.15.104-rockchip64 #3 SMP PREEMPT Wed Mar 22 12:31:37 UTC 2023 aarch64 GNU/Linux
```

I built the image with the `./compile.sh` script of the toolchain without any customisation.

blmhemu, 2023-07-06:
> Also I tried latest 6.1.12 with cpufreq enabled and all armbian patches removed except the add helios64 board add-board-helios64.patch, board-helios64-remove-pcie-ep-gpios.patch, my emmc hs400 es [...]

Does this mean, it could be a bug in upstream kernel ?

prahal, 2023-07-07:
@blmhemu It is a bug that could affect the upstream kernel. But I believe to bug report the upstream kernel, one would have to test with the upstream kernel helios64 dts and I did not try with it. Could be that an HW feature we enable in the armbian helios64 dts is not implemented correctly or has a hardware issue.

Call for help:
I wonder how one could test if the HW is behaving properly. I tried testing voltage on the board while it is running (but the trace are so small I am afraid to short them). Voltages looked fine (12V, 5V). Also, the helios64 chips were delivered during COVID so I am wondering if people having no crash at all are just not stressing their HW enough or not triggering specific software code paths or if only a handful of boards have faulty chips.
Seems borg, urbackup and jellyfin stress the board enough to get it to require hacks like cpufreq.off (but I still get crash from time to time so the issue might not be cpufreq at all, only that cpufreq renders it more frequently... or there are twxo separate issue. I investigate the PCI bug with RK3399 but it seems it is not the same issue I encountered, ie the error log do not match. I have a power domain issue at automatic reboot after a board hang. There was an issue with the U-Boot ATF not resetting the power domains but it should be fixed long ago).

jmue, 2023-07-17:
Hi, I'm also a owner of a helios64 device.

I downloaded https://redirect.armbian.com/helios64/Bookworm_current_minimal and also encountered the same Python crashes.

I took another sd-card and installed https://imola.armbian.com/archive/helios64/archive/Armbian_21.08.2_Helios64_bullseye_current_5.10.63.img.xz on it. There are no crashes, everything is stable. 
So I installed all available updates for user space, kernel and bsp, but still no python errors.
After that I upgraded the image to Bookworm user space as described in https://www.cyberciti.biz/faq/update-upgrade-debian-11-to-debian-12-bookworm/, no Python crashes occur either.
Then I upgraded to bookworm/jammy bsp & kernel and the system is runinng stable on every python test. 
I have also cleared all package differences.

So now I have two different SD cards with the same package base running on the Armbian 23.08 trunk with kernel 6.1.38, one behaves stable, the other one crashes every time on the provided Python test.

Here I'm lost, I cant find the missing piece. I
- ran `diff --recursive --no-dereference /card1/[etc,boot] /card2[etc,boot]`
- compared `tree -agpu /[etc,boot]` of both systems for permission differences 
- compared `debsums -ca` of both systems
- compared `python -m sysconfig`
- cleared the pycache of all python debs vis `py3clean -p`
- did and some other things

There are minor variations but I can't find the decisive one.

Still the upgraded system behaves stable whereas the newly installed one crashes at the python test.

Maybe someone can confirm my findings.

prahal, 2023-07-18:
@jmue u-boot is different on both cards?
Could you share the U-Boot prompt from each of the SD cards?
I would like to try with the u-boot from your good SD card.

Mind I had segfaults with other programs than python3 even though rarely (atop recently).

jmue, 2023-07-18:
So here is the ouput of the two systems of kernel cmdline, zramctl, swapon & the test case.

armbian upgraded from 21.08:
```sh
root@helios64.fritz.box's password: 
 _   _      _ _            __   _  _   
| | | | ___| (_) ___  ___ / /_ | || |  
| |_| |/ _ \ | |/ _ \/ __| '_ \| || |_ 
|  _  |  __/ | | (_) \__ \ (_) |__   _|
|_| |_|\___|_|_|\___/|___/\___/   |_|  
                                       
Welcome to Armbian 23.08.0-trunk Jammy with Linux 6.1.38-rockchip64

No end-user support: untested automated build

System load:   36%           	Up time:       1 min	
Memory usage:  4% of 3.77G  	IP:	       192.168.20.69
CPU temp:      43°C           	Usage of /:    17% of 7.2G   	

Last login: Mon Jul 17 20:46:30 2023 from 192.168.20.64
root@helios64:~# cat /proc/cmdline 
root=UUID=939c3717-edfc-4ff6-b0b5-4b0058bda414 rootwait rootfstype=ext4 console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=8b029f90-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
root@helios64:~# zramctl 
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 lzo-rle       1.9G   4K   74B   12K       6 [SWAP]
root@helios64:~# swapon -s
Filename				Type		Size		Used		Priority
/dev/zram0                              partition	1978500		0		5
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
root@helios64:~# free -mw
               total        used        free      shared     buffers       cache   available
Mem:            3864         219        3542           8          13         171        3645
Swap:           1932           0        1932
root@helios64:~# journalctl -b -k
Journal file /var/log/journal/4cd2c68aaa514c8282d4d1834ecf3199/system@33fa8ccbfc814963af7c8862c925ac54-0000000000003f31-000600ac98b20fd7.journal is truncated, ignoring file.
Jul 18 17:58:46 helios64 systemd-journald[767]: Received client request to relinquish /var/log/journal/4cd2c68aaa514c8282d4d1834ecf3199 acc>
Jul 18 17:58:46 helios64 systemd-journald[767]: Received client request to relinquish /var/log/journal/4cd2c68aaa514c8282d4d1834ecf3199 acc>
root@helios64:~# 

```

armbian new install
```sh
root@helios64.fritz.box's password: 
 _   _      _ _            __   _  _   
| | | | ___| (_) ___  ___ / /_ | || |  
| |_| |/ _ \ | |/ _ \/ __| '_ \| || |_ 
|  _  |  __/ | | (_) \__ \ (_) |__   _|
|_| |_|\___|_|_|\___/|___/\___/   |_|  
                                       
Welcome to Armbian 23.08.0-trunk Bookworm with Linux 6.1.38-rockchip64

No end-user support: community creations

System load:   18%           	Up time:       1 min	
Memory usage:  4% of 3.77G  	IP:	       192.168.20.70
CPU temp:      38°C           	Usage of /:    68% of 1.8G   	

Last login: Mon Jul 17 20:28:20 2023 from 192.168.20.64
root@helios64:~# cat /proc/cmdline
root=UUID=9cc1f6a2-d1b6-46c1-ade3-4d66bbd1d54d rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=0bcbf642-01 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
root@helios64:~# zramctl 
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 lzo-rle       1.9G   4K   74B   12K       6 [SWAP]
root@helios64:~# swapon -s
Filename				Type		Size		Used		Priority
/dev/zram0                              partition	1978500		0		5
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
free(): invalid pointer
Aborted
root@helios64:~# free -mw
               total        used        free      shared     buffers       cache   available
Mem:            3864         224        3555           9           9         157        3640
Swap:           1932           0        1932
root@helios64:~# journalctl -b -k
Journal file /var/log/journal/cbf35eccb86d41699897e19be784fffa/system@53ab717da4a3495dbb21e03036011538-00000000000037af-000600b2ecfa6d29.journal is truncated, ignoring file.
Jul 18 17:51:01 helios64 systemd-journald[768]: Received client request to relinquish /var/log/journal/cbf35eccb86d41699897e19be784fffa acc>
Jul 18 17:51:01 helios64 systemd-journald[768]: Received client request to relinquish /var/log/journal/cbf35eccb86d41699897e19be784fffa acc>
root@helios64:~# 
```

jmue, 2023-07-18:
Here are the diffs taken with user rights for privacy reasons:

[diff-boot.txt](https://github.com/armbian/build/files/12088407/diff-boot.txt)
[diff-initrd.txt](https://github.com/armbian/build/files/12088409/diff-initrd.txt)
[diff-etc.txt](https://github.com/armbian/build/files/12088411/diff-etc.txt)

jmue, 2023-07-19:
@blmhemu Basically you change all the references of `bullseye` to `jammy` in `/etc/apt/sources.list.d/armbian.list`, also change the server url from `http://apt.armbian.com` to `http://beta.armbian.com`. Why jammy and not bookworm? Well `armbian-bsp-cli-helios64-current` depends on `base-files` version `23.08.0-trunk`, which is not available in armbian bookworm.
Then you run `apt update && apt upgrade`.

jmue, 2023-07-20:
A hint could be the different detected version of psci. However, I do not understand how this happens, since I use the same kernel and the same dtb in both images.

As psci is the interface for power management, this could also point to the cpufreq direction. More context: [An overview of PSCI](https://www.thegoodpenguin.co.uk/blog/an-overview-of-psci/)

armbian upgraded:
```sh
root@helios64:~# journalctl -b | grep -e 'psci'
Jul 20 21:43:22 helios64 kernel: psci: probing for conduit method from DT.
Jul 20 21:43:22 helios64 kernel: psci: PSCIv1.0 detected in firmware.
Jul 20 21:43:22 helios64 kernel: psci: Using standard PSCI v0.2 function IDs
Jul 20 21:43:22 helios64 kernel: psci: MIGRATE_INFO_TYPE not supported.
Jul 20 21:43:22 helios64 kernel: psci: SMC Calling Convention v1.0
root@helios64:~# apt list *current-rockchip64
Listing... Done
linux-dtb-current-rockchip64/jammy,now 23.08.0-trunk--6.1.39-Sa456-D710f-P5d5f-C2dc9Hfe66-HK01ba-Vc222-B3b41 arm64 [installed]
linux-headers-current-rockchip64/jammy 23.08.0-trunk--6.1.39-Sa456-D710f-P5d5f-C2dc9Hfe66-HK01ba-Vc222-B3b41 arm64
linux-image-current-rockchip64/jammy,now 23.08.0-trunk--6.1.39-Sa456-D710f-P5d5f-C2dc9Hfe66-HK01ba-Vc222-B3b41 arm64 [installed]
root@helios64:~# sha256sum /boot/dtb-6.1.39-rockchip64/rockchip/rk3399-kobol-helios64.dtb 
aa677e9d64998e535ea29ebeb4073954bf7dac43de3ec3bccf2e7affd3db85f7  /boot/dtb-6.1.39-rockchip64/rockchip/rk3399-kobol-helios64.dtb
```

armbian newly installed:
```sh
root@helios64:~# journalctl -b | grep -e 'psci'
Jul 20 21:06:14 helios64 kernel: psci: probing for conduit method from DT.
Jul 20 21:06:14 helios64 kernel: psci: PSCIv1.1 detected in firmware.
Jul 20 21:06:14 helios64 kernel: psci: Using standard PSCI v0.2 function IDs
Jul 20 21:06:14 helios64 kernel: psci: MIGRATE_INFO_TYPE not supported.
Jul 20 21:06:14 helios64 kernel: psci: SMC Calling Convention v1.2
root@helios64:~# apt list *current-rockchip64
Listing... Done
linux-dtb-current-rockchip64/jammy,now 23.08.0-trunk--6.1.39-Sa456-D710f-P5d5f-C2dc9Hfe66-HK01ba-Vc222-B3b41 arm64 [installed]
linux-headers-current-rockchip64/jammy 23.08.0-trunk--6.1.39-Sa456-D710f-P5d5f-C2dc9Hfe66-HK01ba-Vc222-B3b41 arm64
linux-image-current-rockchip64/jammy,now 23.08.0-trunk--6.1.39-Sa456-D710f-P5d5f-C2dc9Hfe66-HK01ba-Vc222-B3b41 arm64 [installed]
root@helios64:~# sha256sum /boot/dtb-6.1.39-rockchip64/rockchip/rk3399-kobol-helios64.dtb 
aa677e9d64998e535ea29ebeb4073954bf7dac43de3ec3bccf2e7affd3db85f7  /boot/dtb-6.1.39-rockchip64/rockchip/rk3399-kobol-helios64.dtb
```

prahal, 2023-07-20:
@jmue do you have a serial console to grab U-Boot output?
Else you could try to get the U-Boot strings from the SD, but the serial console output would tell more.

jmue, 2023-07-21:
You are absolutely right, my ATF on the "good" card is:
```sh
NOTICE:  BL31: v1.3(debug):42583b6
NOTICE:  BL31: Built : 07:55:13, Oct 15 2019
NOTICE:  BL31: Rockchip release version: v1.1
```
The frequencies are as you described in your comment above. Where do we go from here? Should we use the mainline ATF or the older rockchip ATF?

Thank you for your advice on where I should look. 

As for the Python crashes, I don't see any kernel logs when it crashed.
```sh
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
free(): invalid size
Aborted
root@helios64:~# journalctl -k
Journal file /var/log/journal/cbf35eccb86d41699897e19be784fffa/system@2186cf7b4d244bb091d40cbe1bb232c4-0000000000007d0a-000600efd7a8b2b7.journal is truncated, ignoring file.
Jul 21 21:28:20 helios64 systemd-journald[765]: Received client request to reli>
Jul 21 21:28:20 helios64 systemd-journald[765]: Received client request to reli>
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
munmap_chunk(): invalid pointer
Aborted
root@helios64:~# journalctl -k
Journal file /var/log/journal/cbf35eccb86d41699897e19be784fffa/system@2186cf7b4d244bb091d40cbe1bb232c4-0000000000007d0a-000600efd7a8b2b7.journal is truncated, ignoring file.
Jul 21 21:28:20 helios64 systemd-journald[765]: Received client request to reli>
Jul 21 21:28:20 helios64 systemd-journald[765]: Received client request to reli>
root@helios64:~# for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
Segmentation fault
root@helios64:~# journalctl -k
Journal file /var/log/journal/cbf35eccb86d41699897e19be784fffa/system@2186cf7b4d244bb091d40cbe1bb232c4-0000000000007d0a-000600efd7a8b2b7.journal is truncated, ignoring file.
Jul 21 21:28:20 helios64 systemd-journald[765]: Received client request to reli>
Jul 21 21:28:20 helios64 systemd-journald[765]: Received client request to reli>
```
and my buddyinfo
```sh
root@helios64:~# cat /proc/buddyinfo 
Node 0, zone      DMA    571    480    186     87     42     10      3      3      3      4    880 
```

prahal, 2023-07-21:
@jmue ATF is fine at least I was unable to reproduce the issue at stake with current ATF but using rockchip ddrbin instead of u-boot version of it. The only way out that I know of as of yet is:
ass `BOOT_SCENARIO=tpl-blob-atf-mainline` in `config/board/helios64.csc` and:
```
./compile.sh uboot BOARD=helios64 BRANCH=edge RELEASE=bookworm KERNEL_CONFIGURE=no BUILD_KSRC=no
```
then install the package and apply the firmware to the emmc drive as I told you above.

I am currently trying to upgrade to u-boot v2022.10 as there were a few rockchip-related ram commits around September 2022.

prahal, 2023-07-24:
@blmhemu could you give your serial console U-Boot output so we can check you do not have the supposedly working ddrbin binary from rockchip loaded by U-Boot?
Then you could compile and install U-Boot as I explained on this page (writing it to the emmc else it will not apply) to check it "workaround" the issue (in that well it works but it would be better to fix the U-Boot ram initialization code than reintroduce a closed binary blob).

blmhemu, 2023-07-25:
@prahal  here is the boot serial console - https://pastebin.pl/view/bd5f5d71
I have followed the upgrade path from bullseye (stable 21.08.2) to bookworm.

prahal, 2023-07-25:
@blmhemu intersting. You already use the ddrbin DDR initialization in the bootloader.

Could you try with the latest u-boot, atf, and ddrbin ? instructions at https://github.com/armbian/build/issues/4761#issuecomment-1646211095

If older ddrbin  (the latest with the above u-boot build instruction is "DDR Version 1.25 20210517" while yours is "Version 1.24 20191016") was not working it might make sense out of the fact I had various crashes even with ddrbin. Or it could be the issue was fixed by a mix of ATF, U-Boot fixes above the latest rockchip ddrbin.

You can test if the bootloader upgrade helped by running this command a few times (six times should be enough to be confident the issue is gone):
```
for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done
```

igorpecovnik, 2023-07-26:
uninstall installed u-boot package ...

blmhemu, 2023-07-26:
Tried - says something like `Cannot uninstall virtual packages`

igorpecovnik, 2023-07-26:
correct, uninstall real package linux-u-boot-helios64-current

blmhemu, 2023-07-26:
Here is the new serial console output - https://pastebin.mozilla.org/YjfnuShC

igorpecovnik, 2023-07-26:
installing packages does't flash u-boot. this is done via `armbian-install` utility.

prahal, 2023-07-27:
> Uninstalled the above package and installed using dpkg -i
> Here is the new serial console output - https://pastebin.mozilla.org/YjfnuShC
> It seems like ddrbin did not update ?
Indeed as Igor hinted the package only put the u-boot files in a folder.
You could do the u-boot upgrade via armbian-install`  then choose `5 'Install/Update the bootloader on SD/eMMC'`.

But please keep in mind that if you do not run your OS from emmc and want to install to EMMC this will not do the job.
That is armbian-install write the bootloader to:
`lsblk -ndo pkname $root_partition "$(sudo blkid | tr -d '":' | grep "$(sed -e 's/^.*root=//' -e 's/ .*$//' < /proc/cmdline)" | awk '{print $1}')"`
which points to the SD if you boot to the SD, not the emmc (and the helios64 looks for u-boot on the emmc first, except if you put a jumper to disable the emmc on the helios64 board).
You can run this command to find which it will install to, or read the "root=" value in /proc/cmdline and compare it to the `blkid` output.

To manually install to the emmc when you boot on the SD.
The linux-u-boot-helios64-edge install a script for armbian-install to know how to install  u-boot on the helios64 `/usr/lib/u-boot/platform_install.sh` and the u-boot files to write to the board at `/usr/lib/linux-u-boot-edge-helios64_23.08.0-trunk_arm64` .

For the case where this folder contains u-boot.itb the platform_install.sh script tells to execute:
```
dd if=$1/idbloader.img of=$2 seek=64 conv=notrunc status=none;
dd if=$1/u-boot.itb of=$2 seek=16384 conv=notrunc status=none;
```
where $1 is the folder containing the bootloader binaries, here `/usr/lib/linux-u-boot-edge-helios64_23.08.0-trunk_arm64`
and $2 is the device file for the storage you want to write u-boot to, here on linux 6.3 emmc is `/dev/mmcblk1` (you can check with `lsblk -f`).

For one if your u-boot upgrade leaves you with a broken u-boot on the emmc, then you add the jumper on the board, halt u-boot, remove the jumper, rescan the mmc device with `mmc rescan`, then enter the boot command in u-boot and finally from the linux install on the SD, write manually the u-boot to the emmc.

> Ran `for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done` -> could not repro issue.

Could you try running this command at least six times before upgrading the u-boot?
This is to sort out if your old rockchip64 DDR blob is really also broken with regards to setting up the DDR parameters. That is if you cannot reproduce then there might be other issues (though unlikely).
Though now that we know which version of this blob you had on could reproduce it by tweaking armbian build to install it with this blob.
Mind I will be away a few weeks very soon, so I won't be able to test such an install for quite a while.

blmhemu, 2023-07-27:
Updated the uboot (ddrbin now shows 1.25) - see https://pastebin.mozilla.org/M1XXJnLn
Could not repro the error ! (Ran 6 times)

prahal, 2023-07-28:
@blmhemu then I believe when you did the latest bullseye install somehow you modified the installed bootloader.
I believe you do not have a log of the u-boot output from when you had the invalid free issue, else could you give it?

Probably the 21st of February 2023 when you told us you got the issue fixed (sorry I forgot you already had a fixed setup, I though you were still suffering this issue):
> In the meanwhile, I did install the latest bullseye from [this](https://imola.armbian.com/archive/helios64/archive/) armbian mirror and it works smoothly with ansible and could not see the free issue.

So likely the rockchip DDR blob 1.24 is fine too.

blmhemu, 2023-07-28:
> I believe you do not have a log of the u-boot output from when you had the invalid free issue, else could you give it?

Unfortunately, I do not have those logs :(

blmhemu, 2023-07-29:
@prahal 
Update: I was able to flash the recently compiled build (unstable with free issue) - Here are the logs you asked for https://pastebin.com/zGjvrvux

blmhemu, 2023-07-29:
UPDATE (Again):
> Setting BOOT_SCENARIO=tpl-blob-atf-mainline in config/board/helios64.csc and:

I have compiled armbian with the above option and flashed it.
I could NOT reproduce the issue now. I see also see the ddrbin logs in the serial console.

May be we found the root cause ?
Pasting the link -

blmhemu, 2023-07-30:
UPDATE 3:
I was able to boot fedora !!! Using the above u-boot and idbloader and following the steps at https://fedoraproject.org/wiki/Architectures/ARM/Installation

Ran the python loop `for i in $(seq 1 100);do python3 -c "import pkg_resources" || break;done` and no free error.

prahal, 2023-08-02:
@blmhemu about the DDR frequencies, I added the ddrbin freq to blob less u-boot (keeping all other ddr parameters the same which is probably not fine) and forced them. Still the same issue (though I should post the hack for this issue to be reproduced by others but for one I am away for a few weeks).
Changing the frequencies is not enough. At least one have to tweak the DDR parameters in u-boot lpddr4 inc files. But those are pretty cryptic to me.
And we don't not have these parameters from rockchip it seems (I believe they are on th DDR binary blob). Or maybe it is just that the DDR blob does two training, one before setting the freq to 416MHz and one after with an added advanced training afterwards.

Mind v2023.04 has a fix to do the training at 400MHz instead of 50MHz bit this did not help with our issue.

About he SATA/nvme, maybe look on the kobold wiki, probably in the comments I am confident this was answered. (I also made an u-boot 2023.04 build that seems to have a pretty good support for SATA, but it requires to migrate to new apis (bootlow, bootdev, boothmeth). I want to spend time sharing this hack of a build but it turned out it did not help with the DDR stability issue so it became lower priority. Though I believe instruction to achieve SATA boot are already available in the kobold wiki. If not tell me I will try to share my u-boot v2023.04 for Helios64 build. Mind this build had an issue that it can boot loop in u-boot (I manage to stop the loop but did not investigate the cause yet). So pretty experimental. And somehow u-boot v2022.10 I believe was the version that was not buidlable as it partially migrate to bin man binary build while still being half makefile based. So I was not able to build both the idbloader.imh and u-boot.itb binaries. All in all I attempted those to try the new DDR related fixes in these version which ended up not being related to this invalid free bug.
Either way you have to keep u-boot on emmc and set it up to boot the kernel from the SATA (mind the m.2 slot on the Helios64 is SATA not pcie).

About the eth0 error "Net: dw_dm_mdio_init" I always though it had always been so. I will take a look if I can get this working but not asap I believe ( out of that being an easy catch).
Do you know with which u-boot was it working?

blmhemu, 2023-08-03:
> Do you know with which u-boot was it working?

2020.10 - https://pastebin.mozilla.org/YjfnuShC

blmhemu, 2023-08-03:
```
dw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busNo ethernet found.
```

I have upgraded the system (apt update && apt upgrade) and could not boot now.

prahal, 2023-08-03:
> ```
> dw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio busdw_dm_mdio_init: mdio node is missing, registering legacy mdio [...]

Do you mean u-boot load the kernel then nothing or an error on the serial console?

By the way this looks like another issue ds to avoid this thread becoming unreadable I guess this requires a thread of it's own on the armbian forum. Feel free to tag me in your forum thread so I get a notice by email.

prahal, 2023-08-19:
> > I have upgraded the system (apt update && apt upgrade) and could not boot now.
> 
> If anyone encounters this - this is due to the armbain provided linux-libc-dev (use the debian one instead by giving a lower priority to the armabian repo)

Thanks for the follow-up and workaround. Feel free to open another bug report to track this issue!
It might even be an armbian rockchip64 family issue recently introduced.

I am currently trying a few ideas as I was able to reproduce the raid10 resync always crashing the kernel on helios64 I randomly have since I received the unit. I will try to sort out which of the ideas are useless against this issue (I even had hints that it could be related to the HDDs firmwares above the SATA/pci bridge (rk3399 pcie is known to have bugs, but I suspect at the very least it is not the known issue which affects pcie devices being slow to enumerate).  Or it could be another memory ddr corruption that the mdadm raid10 resync stresses and is the only test case to reliably reproduce its crashes.
At least my old issue predates our current one which requires the ddr rockchip blob to avoid memory corruption from python3 as the initial u-boot form kobol had already this ddr rockchip blob.
I believe I could workaround this crasher but I would really like to sort the cause of this issue (even if hardware related). In the meantime, I cannot boot my helios64.

snakekick, 2023-08-21:
Hello,
i have the same issue, it´s possible to share the new, working armbian-u-boot dpkg?
thank you

jmue, 2023-08-21:
@prahal : Is there anything wrong with opening a pull request until a better solution is found?

[9 more replies]