armbian/build #4765: Moving development related discussion to Github

EvilOlaf, 2023-01-29:
### What happened?

In one of the last weekly developer meeting an interesting idea came up which was to move board development related discussions from forums to Github.

The reasons @SteeManMI, @igorpecovnik  and @rpardini pointed out are that development itself happens on Github anyways, developers most likely have a Github account but not a forum account and also may not want to create one. Also linking issues and proper formatting is present in Github while not in forums.

Therefore the suggestions is to enable the Github discussions feature (https://docs.github.com/de/discussions) for the build repository and set the current "board bring up" forums to read-only with a hint about this change.

What structure should Github discussion have?
As for myself I think a single "forum" is enough to replace current one in Community forums.
Also there was the question about the necessity of moderation then where I think this won't be an issue.

Suggestions or objections?

Corresponding meeting: https://forum.armbian.com/topic/25730-armbian-developers-meeting-1112023/#comment-157588

### Branch

other

### On which host OS are you observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lanefu, 2023-01-29:
@EvilOlaf good idea.

Hopefully helps decouple what's community open source development priorities from whatever initiatives of Armbian that are commercially driven.

schwar3kat, 2023-01-30:
I like this idea.  Why don't we start with a single forum and see how it goes?

EvilOlaf, 2023-02-05:
https://armbian.atlassian.net/browse/AR-1529

igorpecovnik, 2023-03-06:
> for the build repository and set the current "board bring up" forums to read-only with a hint about this change.

We would like to avoid users opening user land and kernel bugs as issues to the build framework. Yes, we direct them into forums now ... but some people will always ignore rules and recommendations.

Board bring up is one thing and this is actually the least problematic things. A PR was opened and within that and we had discussions there ...

We have more problems with users opening bugs whenever they found a submit button. To have bug tracker clean, we close wrong requests, but we could also move them elsewhere.

Transferring an issue from one repository / project to another is simple. Possible destinations: 

- https://github.com/armbian/os 
- https://github.com/armbian/community

This IMO should be solved before discussions itself.

EvilOlaf, 2023-03-06:
This cannot be solved entirely. Some cleanup will always be necessary. Also often people simply don't know due to lack of technical understanding (a discussion if a SBC is the correct toy for those folks is a different story). 
I would not open issues on another repository since this increases fragmentation even more and renders forums to certain extend useless/not needed anymore.

leggewie, 2025-04-27:
I can see @igorpecovnik point, but then again. Noise is going to be noise and as @EvilOlaf has pointed out will need to be dealt with one way or the other.

But lowering the barrier to entry for devs by not requiring yet another subscription (I know that I hate those) as pointed in the original proposal should not be ignored.  Is this going to magically fix everything? No.  Is it still a good idea?  Certainly.