armbian/build #4770: U-boot compilation of Orange Pi 5 failed

bimizx, 2023-01-31:
### What happened?

Failed to compile U-boot, using the default configuration without any modification, running make in the U-boot source directory can compile normally, I also use Orangepi official compilation script to compile without any problems, so it is impossible for me to rely on this machine The reason, and I see that the U-boot compilation in the Build train project in Armbian's Actions is also wrong.

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
LD      spl/drivers/mtd/nand/built-in.o
  LD      spl/drivers/mtd/built-in.o
  LD      spl/drivers/built-in.o
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  CAT     spl/u-boot-spl-dtb.bin
  COPY    spl/u-boot-spl.bin
[ error ] ERROR in function compile_uboot [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:278 -> functions/main/build-tasks.sh:145 -> functions/compilation/uboot.sh:131 -> functions/logging/traps.sh:0 ]
[ error ] U-boot compilation failed
[ o.k. ] Process terminated
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2023-01-31:
Tried to build against `armbian-next` branch? u-boot compiles for me there.

igorpecovnik, 2023-01-31:
https://github.com/armbian/build/actions/runs/4042987680
It compiles on master too, but sometimes not. Since logging in master is not so good, I wasn't been able to find out what is causing this.

bimizx, 2023-01-31:
> Tried to build against `armbian-next` branch? u-boot compiles for me there.

I tried armbian-next, also reported an error, but the error was different.

bimizx, 2023-01-31:
> Tried to build against `armbian-next` branch? u-boot compiles for me there.

What device are you.

EvilOlaf, 2023-01-31:
I'm a human, not a device. But I think what you mean is a qemu VM with Jammy.
https://paste.next.armbian.com/eyoyajizah (rendering will take a few seconds, log is hugh!)

I suggest to start over with both clean environment and build framework.

igorpecovnik, 2023-01-31:
I also tried on my (x64) workstation and it builds just fine.

bimizx, 2023-01-31:
> I'm a human, not a device. But I think what you mean is a qemu VM with Jammy.
> https://paste.next.armbian.com/eyoyajizah (rendering will take a few seconds, log is hugh!)
> 
> I suggest to start over with both clean environment and build framework.

I don't mean that you are a machine, I am asking what kind of development board you are, I am indeed compiling armbian-next on an Aarch64 virtual machine, because I am an Apple Silicon Mac, I will try again later, but master There should be a problem with the source code. I see that the U-boot compilation in the GitHub action has the same error log as mine.

EvilOlaf, 2023-01-31:
> an Aarch64 virtual machine,

Ahaa. That you are using native compiling is an important information you did not give us beforehand. My try was on x86.

igorpecovnik, 2023-01-31:
> I see that the U-boot compilation in the GitHub action has the same error log as mine.

Probably when hitting aarch64 runners.

> There should be a problem with the source code

Base of the problem is dirty code that is delivered by Rockchip so this is not exactly Armbian build engine problem. I am sure code can be fixed, but we have no budget / people that can waste time doing that.

aarch64 support is experimental at best and especially problematic for compiling is this old code that is found in boot loaders. In some cases x86 binaries are needed for compilation. Dunno if this is the problem here?

bimizx, 2023-01-31:
> https://github.com/armbian/build/actions/runs/4042987680
> It compiles on master too, but sometimes not. Since logging in master is not so good, I wasn't been able to find out what is causing this.

[https://github.com/armbian/build/actions/runs/4028999823]
Look at this action, I have the same error as me, I don’t understand that some actions can be successfully compiled.

bimizx, 2023-01-31:
> https://github.com/armbian/build/actions/runs/4042987680
> It compiles on master too, but sometimes not. Since logging in master is not so good, I wasn't been able to find out what is causing this.

https://github.com/armbian/build/actions/runs/4028999823
Look at this action, I have the same error as me, I don’t understand that some actions can be successfully compiled.

bimizx, 2023-01-31:
> https://github.com/armbian/build/actions/runs/4042987680
> It compiles on master too, but sometimes not. Since logging in master is not so good, I wasn't been able to find out what is causing this.

https://github.com/armbian/build/actions/runs/4028999823
Look at this action, and report the same error as me. I don’t understand that some actions can be compiled successfully.

bimizx, 2023-01-31:
> > an Aarch64 virtual machine,
> 
> Ahaa. That you are using native compiling is an important information you did not give us beforehand. My try was on x86.

I also tried x86 in the master branch.

igorpecovnik, 2023-01-31:
I can't reproduce it. Tried on several machines, on the same one that was reporting error (master branch in all cases).

bimizx, 2023-01-31:
> Probably when hitting aarch64 runners.
>
>
> Base of the problem is dirty code that is delivered by Rockchip so this is not exactly Armbian build engine problem. I am sure code can be fixed, but we have [...]

I also suspected the problem of aarch64 in advance, but after I tried it on x86, it was the same, so it was not the problem of aarch64

EvilOlaf, 2023-01-31:
Have you checked output/debug folder for further clues why it might fail?

Since cross-compilation has been confirmed working I close this one as not reproducible but leave the discussion open.

closed 2023-01-31 by EvilOlaf

bimizx, 2023-01-31:
> I can't reproduce it. Tried on several machines, on the same one that was reporting error (master branch in all cases).
> 
I'll try again later

igorpecovnik, 2023-01-31:
> so it was not the problem of aarch64

Ofc not. If code is clean, it compiles on any arch. We will pay attention on this ...

bimizx, 2023-01-31:
@efectn

bimizx, 2023-01-31:
> > so it was not the problem of aarch64
> 
> Ofc not. If code is clean, it compiles on any arch. We will pay attention on this ...

The U-boot of my board is Rockchip's bsp, which is really messy.
But with the same U-boot source code, I can successfully compile it on the orangepi compilation script.

igorpecovnik, 2023-01-31:
> The U-boot of my board is Rockchip's bsp, which is really messy.

Yes, the same code is compiled here as there is no clean boot loader yet and there won't be any alternative for many months. This code is messy, yes, and like I said, we will monitor. Better response then "we will monitor" we cannot provide. I can take a look into this in a couple of weeks/months fastest, while board maintainer is amateur who has his own life. He will also look "when possible". Thank you for your patience. I acknowledge the problem, but it happens rare so its not urgent not critical issue. Its not simple to reproduce, we are not magician and its almost certainly in the part of the code that has nothing to do with our work.

> I can successfully compile it on the orangepi compilation script.

Armbian script compiles 100 boards boot loaders with 100% success and fails Opi 5 in lets say 5%. It makes me believe our code good while I leave things open.

efectn, 2023-01-31:
I think we should provide logs as action artifact to see why compilation failed. Otherwise, it's very difficult to refactor it.

rpardini, 2023-01-31:
`orangepi5` u-boot build, on arm64, using armbian-next, successful every time I tried it...
https://github.com/rpardini/armbian-release/actions/runs/4049971900/jobs/6966912363#step:6:134 (inside "Build board..." and then the `compile_uboot` subgroup)

bimizx, 2023-01-31:
Thank you, I successfully compiled with armbian-next.