armbian/build #4797: ERROR in function compile_kernel - Orange Pi 5

cantalupo555, 2023-02-02:
### What happened?

Hello, I can compile other images normally like Orange Pi 4 LTS.

But after this update in the compiler list, I can no longer create Orange Pi 5 image.

[ error ] ERROR in function compile_kernel [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:280 -> functions/main/build-tasks.sh:155 -> functions/compilation/kernel.sh:143 -> functions/logging/traps.sh:0 ]

### How to reproduce?

  ./compile.sh EXPERT="yes" ROOTFS_TYPE=btrfs BOARD=orangepi5 BRANCH=legacy RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=yes BUILD_ONLY=default KERNEL_CONFIGURE=no DESKTOP_ENVIRONMENT=gnome DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base COMPRESS_OUTPUTIMAGE=sha,gpg,img

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_txgapk_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_iqk_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_reg_cfg_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_dack_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_dpk_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_set_pwr_table_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_efuse_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_tssi_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8853be/phl/hal_g6/phy/rf/halrf_8852b/halrf_kfree_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/phl/hal_g6/phy/rf/halrf_8852b/halrf_psd_8852b.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/core/rtw_prealloc.o
  CC [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/core/rtw_mem.o
  LD [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/rtkm.o
  LD [M]  drivers/net/wireless/rockchip_wlan/rtl8852be/8852be.o
scripts/extract-cert.c: In function ‘display_openssl_errors’:
scripts/extract-cert.c:46:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   46 |         while ((e = ERR_get_error_line(&file, &line))) {
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
scripts/extract-cert.c: In function ‘drain_openssl_errors’:
scripts/extract-cert.c:60:9: warning: ‘ERR_get_error_line’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   60 |         while (ERR_get_error_line(&file, &line)) {}
  411 | unsigned long ERR_get_error_line(const char **file, int *line);
make[4]: *** [scripts/Makefile.build:273: drivers/net/wireless/rtl8188eu/core/rtw_mlme.o] Error 1
make[3]: *** [scripts/Makefile.build:516: drivers/net/wireless/rtl8188eu] Error 2
make[2]: *** [scripts/Makefile.build:516: drivers/net/wireless] Error 2
make[1]: *** [scripts/Makefile.build:516: drivers/net] Error 2
make: *** [Makefile:1920: drivers] Error 2
[ error ] ERROR in function compile_kernel [ functions/cli/cli-entrypoint.sh:109 -> functions/main/build-tasks.sh:280 -> functions/main/build-tasks.sh:155 -> functions/compilation/kernel.sh:143 -> functions/logging/traps.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-02-02:
Could not recreated. Are you sure you are using latest sources?

cantalupo555, 2023-02-02:
> Could not recreated. Are you sure you are using latest sources?

I just updated my post.
Yes, I have updated:
`git clone https://github.com/armbian/build.git`

igorpecovnik, 2023-02-02:
Try to start in a fresh folder.

cantalupo555, 2023-02-02:
> Try to start in a fresh folder.

I did that, I even used another machine.

igorpecovnik, 2023-02-02:
ok, i'll try from a clean start ...

cantalupo555, 2023-02-02:
> 

Thanks, i am waiting.

igorpecovnik, 2023-02-02:
Compilation was successful. In this case, only host can be broken. Start from clean or try Docker.

cantalupo555, 2023-02-02:
How can I do it via docker?
Wouldn't the performance be limited?

igorpecovnik, 2023-02-02:
./compile.sh docker EXPERT="yes" ...

no

cantalupo555, 2023-02-02:
> ./compile.sh docker EXPERT="yes" ...
> 
> no

You might to reconsider, the error continues even with docker...
I now decided to try it on an Oracle Cloud ARM instance, had the same problem.

[image]
[image]

rpardini, 2023-02-02:
@cantalupo555 you might wanna try `armbian-next` branch. use a fresh clone in a clean machine.

cantalupo555, 2023-02-02:
> @cantalupo555 you might wanna try `armbian-next` branch. use a fresh clone in a clean machine.

Hello @rpardini 
Would that be it?
`git clone -b armbian-next https://github.com/armbian/build.git`

SteeManMI, 2023-02-02:
I also can't reproduce.  The build works for me.  There must be something about your environment that is different.  You said under the host OS question that you are building on jammy, but what hardware?  My build on x86 with jammy was fine.

...
[ o.k. ] Runtime [ 28:31 min ]
[ o.k. ] Repeat Build Options [ ./compile.sh  BOARD=orangepi5 BRANCH=legacy RELEASE=jammy BUILD_MINIMAL=no BUILD_DESKTOP=yes BUILD_ONLY=default KERNEL_CONFIGURE=no DESKTOP_ENVIRONMENT=gnome DESKTOP_ENVIRONMENT_CONFIG_NAME=config_base COMPRESS_OUTPUTIMAGE=sha,gpg,img  ]

cantalupo555, 2023-02-02:
@SteeManMI My environment is ARM Linux Ubuntu 22.04.
I was able to compile Orange Pi 5 image normally, but after this update I can't anymore.

igorpecovnik, 2023-02-02:
>  ARM Linux Ubuntu 22.04

This is game changer. ARM is supported experimental ... Here host compilers are used, which we don't control and this frankenstein legacy code needs exact version.

Probably this is the problem. armbian-next could work ...

cantalupo555, 2023-02-02:
@rpardini @igorpecovnik 
Thank you very much, that solved the problem.
Using the armbian-next branch presents no problems in the ARM64 environment.

` git clone -b armbian-next https://github.com/armbian/build.git`

rpardini, 2023-02-02:
I've just pushed to `armbian-next` with #4775 de-revert/cherry-pick which should now include a few more drivers for rk3588 legacy.

cantalupo555, 2023-02-03:
It works great, I just tested it on an Oracle Cloud ARM instance.

Logs: https://paste.next.armbian.com/digedofuni
[image]

[image]