armbian/build #4804: ROOTFS name error during arbitrary creation.Problem found ...

150balbes, 2023-02-03:
### What happened?

ROOTFS name error during arbitrary creation. If you create an arbitrary ROOTFS, with a non-standard set of groups \ packages, which is not available on the server for downloading, the name of the received ZST archive is formed using the HASHE-DATE_TIME.zst at the end of the name. But with the subsequent reuse of this set, the system searches for an archive either with HASHE-HASH.set in the name, or with HASHE-NOMER_REPO.zst (0121 0120 0119 0118, etc.) and does not find an archive named HASHE-DATE_TIME.zst.

### How to reproduce?

Command you gave to the framework!

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-02-24:
Is this still the problem in main?

rpardini, 2023-02-25:
Yes, this needs attention on `main` too -- will try to reproduce

rpardini, 2023-02-25:
Checked. Does not apply to `main`. This originated (back in old master) with inconsistent usage of `ROOT_FS_CREATE_VERSION` vs `ROOTFSCACHE_VERSION`. Those are no longer used in `main` branch since the introduction of the rootfs artifact, which should enforce the consistent versioning and naming...

I'll close this, let me know if I missed something

closed 2023-02-25 by rpardini