armbian/build #4809: Unable to make a new forum account (Kreyren)

Kreyren, 2023-02-03:
### What happened?

[image]

* Submitted here based on the bug thing telling me things

### How to reproduce?

Command you gave to the framework!

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2023-02-03:
I don't moderate forums.

closed 2023-02-03 by igorpecovnik

EvilOlaf, 2023-02-04:
You were using some sort of proxy which lead the system into flagging you as spammer.
Check your emails.