armbian/build #4816: drivers/bluetooth/hci_ldisc.c: In function ‘hci_uart_register_dev’:

AGM1968, 2023-02-05:
### What happened?

‘HCI_QUIRK_BROKEN_PARK_LINK_STATUS’ undeclared (first use in this function)

### How to reproduce?

custom configuration file: config-h616.conf parameters
CLEAN_LEVEL="make,debs,cache,oldcache,images,patch" #note patch does not clean
INSTALL_HEADERS=""									# install kernel headers package
LIB_TAG="master"									# change to "branchname" to use any branch currently available.
KERNEL_VERSION_LEVEL=5.19
KERNELSWITCHOBJ='tag:v5.19.17'
EXPERT=yes
BRANCH=edge
CREATE_PATCHES=yes 
BOARD=orangepizero2 
RELEASE=bullseye 
BUILD_MINIMAL=yes 
BUILD_DESKTOP=no 
BUILD_ONLY="" 
KERNEL_CONFIGURE=yes 
COMPRESS_OUTPUTIMAGE=sha,gpg,img

### Branch

master (main development branch)

### On which host OS are you observing this problem?

Jammy

### Relevant log output

```shell
drivers/bluetooth/hci_ldisc.c: In function ‘hci_uart_register_dev’:
drivers/bluetooth/hci_ldisc.c:665:11: error: ‘HCI_QUIRK_BROKEN_PARK_LINK_STATUS’ undeclared (first use in this function); did you mean ‘HCI_QUIRK_BROKEN_STORED_LINK_KEY’?
   set_bit(HCI_QUIRK_BROKEN_PARK_LINK_STATUS, &hdev->quirks);
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           HCI_QUIRK_BROKEN_STORED_LINK_KEY
drivers/bluetooth/hci_ldisc.c:665:11: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:249: drivers/bluetooth/hci_ldisc.o] Error 1
make[1]: *** [scripts/Makefile.build:466: drivers/bluetooth] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1848: drivers] Error 2
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AGM1968, 2023-02-05:
Here the hci.h rej. not much info about this, seems related to uwe5622 - will have a sit down and look!

--- include/net/bluetooth/hci.h
+++ include/net/bluetooth/hci.h
@@ -262,6 +262,12 @@ enum {
 	 * during the hdev->setup vendor callback.
 	 */
 	HCI_QUIRK_BROKEN_READ_TRANSMIT_POWER,
+
+	/*
+	 * Device declares that support Park link status, but it really
+	 * does not support it and fails to initialize
+	 */
+	HCI_QUIRK_BROKEN_PARK_LINK_STATUS
 };
 
 /* HCI device flags */

Processing file /home/gordon/build/patch/misc/wireless-driver-for-uwe5622-park-link-pre-v6.1.patch
1 out of 1 hunk FAILED -- saving rejects to file include/net/bluetooth/hci.h.rej
1 out of 1 hunk FAILED -- saving rejects to file net/bluetooth/hci_core.c.rej

EvilOlaf, 2023-02-05:
While we do archive patches from superseded kernel versions there is no guarantee that those will build. Upstream fetched code moves on and might break things which Armbian cannot control.

For the record: Building edge (6.1.y atm) from `armbian-next` branch works just fine:  https://paste.next.armbian.com/asabazifoh

AGM1968, 2023-02-05:
Thanks I do realise that.

EvilOlaf, 2023-02-05:
No. Next `current` will most likely be new upstream LTS kernel which 6.1 may become. It is still TBD.

AGM1968, 2023-02-05:
Thanks.. guess ill move myself along then...

EvilOlaf, 2023-02-05:
Well I cannot speak for you how you spend (or waste) your time 😁 but yes, I think closing this is a good idea.

closed 2023-02-05 by EvilOlaf

AGM1968, 2023-02-07:
A very temporary and not recommended workaround for 5.19.17 to skip applying!
```
-		if linux-version compare "${version}" lt 6.1; then
+		if linux-version "${version}" ge != 5.19 && compare "${version}" lt 6.1; then
		process_patch_file "${SRC}/patch/misc/wireless-driver-for-uwe5622-park-link-pre-v6.1.patch" "applying"
```
I did have a quick look at the failed hunks, it seems to be mismatch etc so will probably just do a proper job! Thanks for the Evil advise Olaf .